letscontrolit/ESPEasy #3472: ANA input problem A0

Gigboce, 2021-01-21:
Bonjour, 

J'ai un problème sur l'entre A0 j'ai brancher un capteur de luminosité avec une résistance, mais j'ai la valeur qui est 65535 ne change jamais, j'ai essayer avec deux ESP8266 monté sur un NODEMCU mais cela ne fonctionne pas 

J'utilise la version ESP_Easy_mega_20201227_normal_ESP8266_4M1M_VCC.bin

Hello,

I have a problem on input A0 I connect a light sensor with a resistor, but I have the value which is 65535 never changes, I tried with two ESP8266 on mounted on a NODEMCU but it does not does not work

sorry for my english it is not very good

TD-er, 2021-01-21:
You are using a "VCC" build, which is the one you should **not** use when you want to use the analog input.
The "VCC" builds are made to use the analog input to measure the power voltage of the ESP.

So please try to use a build which does not have "VCC" in the file name.

Gigboce, 2021-01-21:
Oh Yes very thank You :D

TD-er, 2021-01-21:
I'll close this one.
If this wasn't the solution, please let me know here.

closed 2021-01-21 by TD-er