letscontrolit/ESPEasy #3474: ESP.Easy.Flasher.exe doesn't recognize COM port that works using FlashESP8266.exe

yang3535, 2021-01-22:
### Summarize of the problem/feature request

The title says it all.  When using ESP.Easy.Flasher.exe, the COM port for my attached Lolin D1 Mini v3.1.0 does not appear in the drop down list even though I can confirm in Windows Device Manager that it is recognized by the OS.  It makes no difference if the Only Active Ports box is checked on not.

The FlashESP8266.exe program recognizes the COM port and can be used to flash the Lolin D1 Mini board without issues.

### Expected behavior

ESP.Easy.Flasher.exe should be able to  utilize all COM ports that the OS recognizes.

### Actual behavior

ESP.Easy.Flasher.exe does not recognize COM port even though COM port existence is confirmed through Windows Device Manager and through use FlashESP826.exe.

### Steps to reproduce

1. Attach Lolin D1 Mini v3.1.0
2. Verify COM port # through Windows Device Manager
2. Run ESP.Easy.Flasher.exe - note missing COM port
3. Run FlashESP8266.exe - note COM port is available

### System configuration

Hardware:
- Lolin D1 Mini v3.1.0 - https://www.wemos.cc/en/latest/d1/d1_mini.html

ESP Easy version: 
I have verified this issue in ESPEasy_ESP82xx_mega-20210114.zip and ESPEasy_ESP82xx_mega-20201130.zip

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2021-01-22:
This is a known issue.
Some serial to uart chips do install some different driver which does not mark the presence of the COM ports in the same location in the registry as do the other drivers.
I also have some boards which are not recognized in the flasher tool.

Grovkillen, 2021-01-22:
Yep and I'm currently developing a new web based flasher which will not use
virtual com ports. It's gonna be a lot easier to maintain and use. Nothing
need to be installed and you will be using latest version automatically. If
you're a patreon supporter you will be able to take part in the beta
testing.

inzuno, 2021-11-07:
You can merge tis issue with this one: #3172 https://github.com/letscontrolit/ESPEasy/issues/3172
Is it possible to release at least a version where you can enter manually the port number?

tonhuisman, 2023-09-01:
The ESP Easy Flasher has been removed from the repository as there has been no maintenance on it for a long time, so this issue can be closed.

closed 2023-09-01 by TD-er