letscontrolit/ESPEasy #3479: Can ESPeasy work in AP+STA mode?

ProgrammingPlanet, 2021-01-25:
Can I achieved AP+STA mode simultaneously?

I have an esp8266 (esp-12f).

TD-er, 2021-01-25:
Yep, you can.
There is a command to start the AP: `wifiapmode`
However this will have some consequences on how ESPeasy does handle WiFi reconnects as it looks at whether a client is connected to its AP.

I'm currently working on the WiFi implementation to make it more stable and made also a test build for it: https://github.com/letscontrolit/ESPEasy/pull/3473#issuecomment-766346327
Maybe you can also test that build and report there if you find anything not working well.
At least that implementation should work a lot more predictable on WiFi connections (using the strongest signal for example).

I'm not 100% sure if the AP will automatically be disabled if it isn't used after N seconds. I will check and add a configuration flag to tune this if it does disable it when you enabled it via a command.

TD-er, 2021-01-25:
By the way, this is more like a question on how to use ESPeasy and those questions are better asked at [the forum](https://www.letscontrolit.com/forum/index.php)
So I will close this one as it isn't (yet) a feature request or a bug.

closed 2021-01-25 by TD-er