letscontrolit/ESPEasy #3481: INA219 cannot output just 1 value

affinite2, 2021-01-28:
I'm trying to output only the "Current" value from an INA219 Voltage/Current/Power sensor.
Problem is that, in the plugin, when I select "Current" in the Measurement Type box I continue to get Voltage, Current and Power values sent out.
Build ESP_Easy_mega_20201227_normal_ESP8266_4M1M

tonhuisman, 2021-01-28:
From the [Forum](https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=8267)

tonhuisman, 2024-07-17:
@affinite2 I've done a fix for this issue (see link above this post), are you available for testing? Build can be downloaded from this [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/9981658248).

Please report your findings here, TIA.

PR #5089 (2024-07-27): [P027] Add optional Powerdown mode
Resolves #2248 
Resolves #3481 

Features:
- P027 Add Powerdown mode (optional) to reduce the quiescent current when not taking measurements, f.e. during sleep mode in battery operated configurations.
  - This feature is not available in `LIMIT_BUILD_SIZE` builds (Display, Collection, Climate) for bin size reasons.
- Extend/Update documentation.
- P027 Fix sending single values via Controller as reported in #3481

TODO:
- [ ] Testing
- [ ] Update documentation
closed 2024-07-27 by TD-er