letscontrolit/ESPEasy #3485: Esp Easy TemT6000

krais1996, 2021-02-02:
Good afternoon! is it possible to connect the TEMT6000 light sensor in this firmware? If Yes, then what type of device it is necessary to choose

tonhuisman, 2021-02-02:
It seems to have an analog output, so you should be able to connect that to the analog in port of an ESP. You may need to add a voltage divider as not all ESP have that installed by default (standard input range is 0-1Volt).
If you need higher accuracy you could use an ADS1115 converter that offers 16 bit A/D conversion, instead of the 10 bit A/D converter in an ESP8266 or 12 bit A/D converter in an ESP32.

TD-er, 2021-02-02:
Just as extra information.
Most ESP8266 boards like NodeMCU or Wemos D1 mini have resistors to change the range from 0 ... 1V to 0 ... 3.3V.

closed 2023-09-05 by TD-er