letscontrolit/ESPEasy #3489: Round in rules don't work

fly74, 2021-02-05:
I tried round like in https://espeasy.readthedocs.io/en/latest/Rules/Rules.html?#math-functions

with
...
let,1,round([BLK#Temperature])
[7SEG1].7dtext,round([var#2])
...
but the display show always "round 0" not the value.

TD-er, 2021-02-05:
That function should also log something at "info" level.
Can you see what it logs when you try to set the 7dtext ?

fly74, 2021-02-05:
ACT  : let,1,round(-1.39*10)/10
38130 : Debug  : Command: let
38143 : Info   : ACT  : [7SEG1].7dtext,-1.0

TD-er, 2021-02-05:
```
let,1,[BLK#Temperature]*10
let,1,round([var#1])/10
```
The "math" functions cannot be used with parameters that still need to be computed.
This is because the math functions are replaced in the first run and not after the calculate is done.

fly74, 2021-02-05:
ist the same

2423744 : Debug  : Command: timerSet
2423755 : Info   : ACT  : let,1,-0.87*10
2423757 : Debug  : Command: let
2423760 : Info   : ACT  : let,1,round(-8.7)/10
2423761 : Debug  : Command: let
2423773 : Info   : ACT  : [7SEG1].7dtext,-1.0

should 0.9

fly74, 2021-02-05:
works!:

2548023 : Debug  : Command: timerSet
2548034 : Info   : ACT  : let,1,-0.87*10
2548035 : Debug  : Command: let
2548039 : Info   : ACT  : let,1,round(-8.7)
2548040 : Debug  : Command: let
2548043 : Info   : ACT  : let,1,-9/10
2548044 : Debug  : Command: let
2548055 : Info   : ACT  : [7SEG1].7dtext,-0.9,-0.87
...

let,1,[BLK#Temperature]*10
let,1,round([var#1])
let,1,[var#1]/10
[7SEG1].7dtext,[var#1#d2.1]

closed 2021-02-05 by fly74