letscontrolit/ESPEasy #3491: Weird switch state (state value 4 instead of 1 or 0)

qwertangel, 2021-02-06:
Using build ESP_Easy_mega_20210114_normal_ESP8266_4M1M , mega-20210114_cdc8a1a, ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support  on a D1 Mini clone.

i have connected a touch sensor board [TTP223](https://www.ebay.com/itm/10PCS-TTP223-Capacitive-Touch-Switch-Button-Self-Lock-Module-for-Arduino/363237197208?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649) to GPIO 4 (D2). Enabled Internal Pullup resistor in device config.   This GPIO is configured as a switch. It configured to sends a MQTT message (using opehab mqtt controller setting) on state change.

Since it is configured as a switch it should only return 1 or 0. However randomly (not able to explain why and when), it returns payload as 4. 
Why would payload be 4 instead of 1 or 0 ?

TD-er, 2021-02-06:
There are other states possible as well, like long press, double click, etc.
Have to check the source to get a list for your.

TD-er, 2021-02-06:
I think this is the correct explanation of your return values:
https://github.com/letscontrolit/ESPEasy/blob/e25f0d9994677bccb04e704f9ab10aefb567dfa9/src/_P001_Switch.ino#L395-L408

Maybe @giig1967g can shed some more light on this (<hint> documentation ;) </hint>)

qwertangel, 2021-02-06:
Don't see 4 in the code ?