letscontrolit/ESPEasy #3492: P093 missing in latest test build

TungstenE2, 2021-02-06:
Hi all,

I just updated to latest test build and it appears that P093 is missing.

ESP_Easy_mega_20210114_test_ESP8266_4M1M_VCC.bin

According to the plugin list it should be included.
https://espeasy.readthedocs.io/en/latest/Plugin/P093.html#p093-page

Please check.

[image]

thx

tonhuisman, 2021-02-06:
This plugin is currently excluded because it won't fit in the testing build.

You could make a custom build yourself that does include this plugin, or install a release created before September 20th, 2020, as that's the date this exclusion was merged into the code.

TungstenE2, 2021-02-06:
oh, what a petty.

The plugin list, does not contain such a plugin status. How best to reflect such excluded status and the date of such status change? Also the plugin page should be updated in such cases.

https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html

Is there a tutorial on how to make custom build including a plugin and excluding unneeded plugins?

thx for the support!

tonhuisman, 2021-02-06:
This documentation [for installing PlatformIO](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html) is quite accurate, and TD-er has made a Youtube contribution some time that is quite helpful [here](https://youtu.be/ArqwMcYSMsU) with one addition: You should install Git for Windows first (if not already installed, and on Windows platform 😄).

TungstenE2, 2021-02-06:
thx, will give it a try.

How about updating the plugin list with a status of excluded plugins in order to be in sync with the releases?

TungstenE2, 2021-02-06:
Thx, the youtube video is great, but could you also explain how to build test version exclude or including plugins?

TD-er, 2021-02-06:
In short, you can rename the `Custom_sample.h` to `Custom.h` and uncomment the plugins you like to have.
If that `Custom.h` (capital C) is present, a PlatformIO "Custom" build will use that file.

I guess that's the simplest way to personalize your build.

crnjan, 2021-02-07:
Out of curiosity - are you experiencing any issues with P093?

TD-er, 2021-02-07:
I guess it would also make sense to have it included in the "energy" build, right?

TungstenE2, 2021-02-07:
@TD-er custom build like this?
[image]

TungstenE2, 2021-02-07:
hm, strange.

With custom build the plugin is loading, but something is wrong.

ESP_Easy_mega_20210207_custom_ESP8266_4M1M.bin

[image]

Using ESP_Easy_mega_20200721_test_ESP8266_4M1M_VCC.bin

[image]

Do I need to include other plugins for the serial port?

tonhuisman, 2021-02-07:
Just find that #define in the source and enable it.

TungstenE2, 2021-02-07:
which #define are you referring to?
As you can see I enabled '#define USES_P093'  but I get a problem with that custom build with the serial port.

tonhuisman, 2021-02-07:
Ah, it is only defined depending on a lot of plugins, but not this one 😊 so P093 should probably be added to that list.
Just add `#define PLUGIN_USES_SERIAL` below the USES_P093 in Custom.h and you should be fine.

TungstenE2, 2021-02-07:
thx, now serial is working but others are still missing.

[image]

To be honest I would be great if the plugin could be added again to a real regular build.

TD-er, 2021-02-07:
I will add it to the "energy" build for now.
Problem is we're constantly running out of max. allowed build size, so I really must find time to get the build service running.

<rant>
Tomorrow the schools would finally open again, but I just learned the schools will be kept closed tomorrow due to "snow".
Only problem is... I can't find this snow they are talking about here in Groningen
So again another day of complete non-productivity
</rant>

TungstenE2, 2021-02-08:
fyi, https://www.letscontrolit.com/forum/viewtopic.php?t=8159

TungstenE2, 2021-02-08:
> 
> Problem is we're constantly running out of max. allowed build size, so I really must find time to get the build service running.
> 

this is the downside of the great success of your project.... ;-)

TungstenE2, 2021-03-04:
> I will add it to the "energy" build for now.

@TD-er where to find energy build? In ESPEasy_ESP82xx_mega-20210223 I do not find such a build.

TD-er, 2021-03-04:
Just tried a test build on my PC to see why it was not included and noticed this build error:
```
c:/users/gijsn/.platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio/build/energy_ESP8266_4M1M/ESP_Easy_mega_20210304_energy_ESP8266_4M1M.elf section `.text1' will not fit in region `iram1_0_seg'
```
So apparently running into resource limitations again.... :(

TD-er, 2021-03-04:
I removed the sensor used in the Sonoff POW r1 from the "energy" build in this commit:
https://github.com/letscontrolit/ESPEasy/commit/a43910cde6c0725856a9695e77e558dfc9b09b0c
And added the P093 in this commit: https://github.com/letscontrolit/ESPEasy/commit/aef8f4d9b13de7885a3e01cc394dad3bf817f37d

If you like, I can make a build for you including the rest.

TungstenE2, 2021-03-04:
thx, no need to hurry for me. Waiting for official build. So with next release there will be an energy build including P093, right?

TD-er, 2021-03-04:
You can check this [ESP_Easy_mega_20210304_energy_ESP8266_4M1M.bin](https://www.dropbox.com/s/cww1ohvx84c2y8o/ESP_Easy_mega_20210304_energy_ESP8266_4M1M.bin?dl=0) build if you like :)

TungstenE2, 2021-03-05:
thx, but strange that P093 is not fitting into the energy build, as it was working in testing build not long time ago.

@crnjan can you support @TD-er here to check the size of your plugin?

TD-er, 2021-03-05:
P093 wasn't the problem here, as it wasn't a "bin size" problem.
The problem is with the very limited amount of "IRAM" in the ESP8266.
Every function that can be called from an interrupt should use an attribute to place it in IRAM and only use functions with the same attribute.
The variables used by these functions should also remain in that memory by declaring them `volatile`.

There has been a recent change in the P003_Pulse plugin which needed more variables, so maybe that pushed it just over the limit.

crnjan, 2022-08-23:
Is this still an issue? IMHO now it's [very well documented](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html#list-of-official-plugins) for each plugin (build set), i.e. `Status: ENERGY, COLLECTION D` ...

PR #4647 (2023-05-07): [P093] Add support for retrieving all device values
From a [Forum request](https://www.letscontrolit.com/forum/viewtopic.php?t=7678#p64356)

Features:
- Retrieve all available (and enabled) values from the device, similar to the content of the json that is emitted.
- Updated documentation

Resolves #3492
closed 2023-05-07 by TD-er