letscontrolit/ESPEasy #3494: Getting high negative values on Sensor DHT22 with negative Temperatures

tomcatxx, 2021-02-06:
If you use the DHT22 Plugin you get values like -3276,8 for Temperatures <=0°C

The way I fixed it was taking this example: https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTstable/dht.cpp
But I still got -3276.80 at 0.0 °C. For this I just found a dirty way.
Here the way I fixed the whole problem. Maybe someone with more C++ experiance can fix it cleaner....

_P005_DHT.ino:
```
    case P005_DHT22:
      int16_t temp_temperature;
      temp_temperature   = (dht_dat[2] * 256 + dht_dat[3]);
      temperature = temp_temperature * 0.1;
      if (temperature < -3000) { temperature = 0; } // Dirty way to prevent -3276.80 at 0.0 °C.
      humidity = ((dht_dat[0] * 256) + dht_dat[1])/10; 	  
      break;
```

TD-er, 2021-02-06:
My first impression is that you may need to convert 2-complement values.
Isn't it just something like this? 

```c++
    case P005_DHT22:
      int16_t temp_temperature = ((dht_dat[2] & 0x7f) * 256 + dht_dat[3]);
      if (dht_dat[2] & 0x80) {
        temp_temperature  *= -1;
      }
      temperature = temp_temperature * 0.1f;
      humidity = ((dht_dat[0] * 256) + dht_dat[1])/10; 	  
      break;
```

Otherwise you're throwing away negative readings.

tomcatxx, 2021-02-06:
```
    case P005_DHT22:
      int16_t temp_temperature = (dht_dat[2] & 0x7f) * 256 + dht_dat[3]);
      if (dht_dat[2] & 0x80) {
        temp_temperature  *= -1;
      }
      temperature = temp_temperature * 0.1f;
      humidity = ((dht_dat[0] * 256) + dht_dat[1])/10; 	  
      break;
```
Gives compile error.
```

    case P005_DHT22:
      int16_t temp_temperature;
      temp_temperature = (dht_dat[2] & 0x7f) * 256 + dht_dat[3]);
      if (dht_dat[2] & 0x80) {
        temp_temperature  *= -1;
      }
      temperature = temp_temperature * 0.1f;
      humidity = ((dht_dat[0] * 256) + dht_dat[1])/10; 	  
      break;

```

Is compileable but result in high negative number like without the patch.

Barracuda09, 2021-02-10:
Hi @tomcatxx 

Is dht_dat by any chance an unsigned integer type (uint8_t), then you should cast that first so you don't need the multiplication with -1.

Probably there is no sign extension this way.

closed 2022-08-02 by TD-er