letscontrolit/ESPEasy #3496: Special character {D} is not parsing in the Framed OLED plugin

k-zakhariy, 2021-02-08:
I've updated firmware to **ESP_Easy_mega_20210114_dev_ESP8266_4M1M**, and after that i've tried to configure Display - **OLED SSD1306/SH1106 Framed** with following Line1:

Line 1 : `[HDC1080#Temperature] {D}C`

Also i've tried to add some logs in rules:
```
on HDC1080#Temperature do
LogEntry,'Temperature: %c% - {D}C'
endon
```
and got result: "Temperature: %c% - {D}C"

**ISSUE:**
The degree symbol (and i suppose others) is not parsing on SH1106 display and not showing in Log entries

k-zakhariy, 2021-02-08:
Any idea how i can make it work? Maybe i missed something
Thanks in advance

TD-er, 2021-02-08:
Hmm that sounds like a bug.
What older builds did you try that still worked?

k-zakhariy, 2021-02-08:
Unfortunately i didn't remember the old version, flushed my all ESP's ... But i guess that was pretty old, around mega-20200829

k-zakhariy, 2021-02-08:
ok, so latest firmware was ESP_Easy_mega_20201009_dev_ESP8266_4M1M

TD-er, 2021-02-08:
OK will check later today why it is no longer working.
Maybe I have broken something when adding the mathematical formula functions?

TD-er, 2021-02-08:
Hmm that's old information.

In the past we had it configured as a cron job that started at 4am and if there were changes to the 'mega' branch, it would generate a new build.

However that led to undesirable situations where you would get a new build if only a spelling error was changed, so those things were put in a PR if nothing else was ready to merge.
In reality you would have almost daily a new build, which often led to the situation that people did not want to upgrade daily (understandable) and thus reported issues may have been fixed "yesterday" while they were running a build of only a few days old and thus were told they were running an old build.
This is taking a lot more iterations in the processing of new reported issues.

So what I have been done in the past year (or more...) is that I trigger a build manually when I have big enough changes merged.
Also PRs often have had a number of test builds linked in the related issue. This greatly improves the code quality as a PR is not rushed any more to meet the "nightly build deadline" (believe me, it has happend a lot) and also PRs are now tested by several people before they get merged.

On average I try to keep a pace of 2 - 3 builds a month and if you look at the change logs, those often have between 20 - 100 commits in them.
The last official build has indeed been a while already, which is caused by some reported issues after I merged a pending PR.
So I don't want to make a new build until those issues are fixed.

k-zakhariy, 2021-02-09:
Got it, thanks for answer

TD-er, 2021-02-09:
I tested on my test setup and the degree sign is showing.
Can you try this test build I linked here: https://github.com/letscontrolit/ESPEasy/pull/3487#issuecomment-774656609  ?

It still has other issues but at least it can be used to test the degree sign is working on the Framed OLED plugin.

k-zakhariy, 2021-02-09:
So i've tried this build ESP_Easy_mega_20210207_normal_ESP8266_4M1M.bin - and it works as expected, however HDC1080 is not there, so i flashed ESP_Easy_mega_20210207_dev_ESP8266_4M1M and its still not working there
[image]

TD-er, 2021-02-09:
Ah I get it....
To reduce bin size I disabled the special character parsing on `LIMIT_BUILD_SIZE` builds.
Testing and Dev builds also have that defined.

So back to the drawing board to make it all fit in the builds again.... although some "testing" and "dev" builds already fail again due to build size, even with this special character parsing disabled for them.

k-zakhariy, 2021-02-09:
as workaround i rolled back to build mega-20201022 which is works fine. It seems the issue starts from build mega-20201130

k-zakhariy, 2021-02-09:
btw, i noticed that CPU load also higher on newest build : around **26%** , comparing to mega-20201022 : **4.63%**

**Devices:**
Environment - HDC1080 (I2C) [TESTING]
Gases - CO2 MH-Z19
Display - OLED SSD1306/SH1106 Framed

k-zakhariy, 2021-06-12:
Hey @TD-er , thanks for a new releases, how are we doing with this issue, i'm still facing this problem on latest build:
ESP_Easy_mega_20210503_test_A_ESP8266_4M1M
Thanks

TD-er, 2021-06-12:
What is "this" problem as you mentioned several.
- higher CPU load
- Not replacing `{D}`

k-zakhariy, 2021-06-12:
Second one, but i just found that simply using raw text works fine, ex: `[HDC1080#Temperature]°C` 
So looks like this issue can be closed

closed 2021-06-12 by k-zakhariy