letscontrolit/ESPEasy #3512: enhancement request for rules: "in-memory" rules and "compiled" rules

AloisKlingler, 2021-02-12:
rulefiles and parsing need especially on bigger rulesets quite a lot time.
it would be great to have:

 * the possibility to store small, but often used blocks, in memory, to avoid the "reading" overhead, either by an additional ruleset (like "memoryrules.txt") or by a special "on ... endon" block
 * "precompiled" event-rules to avoid parsing and executing all the time some calculations, which should be on microcontrollers done in microseconds, but consume because of the parsing-overhead much more time.

tonhuisman, 2021-02-12:
You have found #3164 already?

AloisKlingler, 2021-02-12:
nope, thanks, thats exactly what I meant, but without code and tecnically not as descriptive ... ;)
awaiting this feature now hard, if you want me for testing please just write. ;)

closed 2021-02-12 by AloisKlingler