letscontrolit/ESPEasy #3516: P083 Gasses - SGP30 TVOC value rises incredibly high

k-zakhariy, 2021-02-16:
After some period of time values of TVOC sensor rises to some unbelievable values - only turn off power supply helps. I also tried another power supply - same problem, could this be problem of this software ?

I would appreciate any help. Thanks in advance

[image]
[image]

Build:⋄
20111  - Mega
System Libraries:⋄
ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄

Plugin Count:⋄
88 [Normal] [Testing] [Development]
Build Time:⋄
Jan 14 2021 23:21:32
Binary Filename:⋄
ESP_Easy_mega_20210114_dev_ESP8266_4M1M
Build Platform:⋄
Linux-4.19.104-microsoft-standard-x86_64-with-glibc2.29
Git HEAD:⋄
mega-20210114_cdc8a1a

N2k1, 2021-02-20:
Whenever the temperature rises - or the humidity changes significantly - this happens.
The sensor calculates the supposed proportion without measuring it itself.
In the manufacturer's library there is a possibility to include the values for temperature and humidity in the calculation - unfortunately this is missing in this implementation.

k-zakhariy, 2021-07-01:
@N2k1 so it can't calculate value precisely without including temp. parameters from other devices, am i got right your point ?