letscontrolit/ESPEasy #3517: PCONFIG cannot store values >255

AlfredoCubitos, 2021-02-17:
Hi,

I'm trying to develop a plugin.
I found out that PCONFIG can not store values >255
When I'm trying to store 256 PCONFIG it returns 0
Values <256 are stored correctly: Storing 255 it returns 255

The documentation in the example template says:
`//   PCONFIG(x)`
`//   x can be between 1 - 8 and can store values between -32767 - 32768 (16 bit)`

Example Code:
`addFormNumericBox(F("Puls Length"),F("rcs_pl"),pulseLength);`
`PCONFIG(1) = getFormItemInt(F("rcs_pl"));`

my environment is: normal_ESP8266_4M1M

Is this a bug?

TD-er, 2021-02-17:
See:
```c++
#ifndef PCONFIG
  # define PCONFIG(n) (Settings.TaskDevicePluginConfig[event->TaskIndex][(n)])
#endif // ifndef PCONFIG
```

```c++
  int16_t       TaskDevicePluginConfig[N_TASKS][PLUGIN_CONFIGVAR_MAX];
```

```c++
int getFormItemInt(const String& key, int defaultValue);
```

So it should indeed be able to store `int16_t` and the `getFormItemInt` function also is not limited to an `uint8_t` (or `byte` type)

What type is your `pulseLength` ? Is that maybe an `uint8_t` ?

tonhuisman, 2021-02-17:
That reference to PCONFIG is not according to current sources, as it is defined as a macro:
`  # define PCONFIG(n) (Settings.TaskDevicePluginConfig[event->TaskIndex][(n)])`
And that array is defined like this: `int16_t       TaskDevicePluginConfig[N_TASKS][PLUGIN_CONFIGVAR_MAX];`
And no reference of 1 - 8, as it is actually 0 - 7.

Does your plugin have a local definition for PCONFIG ? If so the just disable that, as it's plain wrong.

AlfredoCubitos, 2021-02-17:
@TD-er  and @tonhuisman 
alright guys, thank you for your hints.
My type was _byte_ I changed it to `int16_t` and it works as expected :-)

closed 2021-02-17 by AlfredoCubitos