letscontrolit/ESPEasy #3522: ESP_Easy_mega_20210223_normal_ESP8266_4M1M.bin no WLAN after Update

fly74, 2021-02-23:
New WLAN can not connect in ESP_Easy_mega_20210223_normal_ESP8266_4M1M.bin:

Network is hidden with static IP.
...
126143 : Info   : WIFI  : Scan finished, found: 9
126152 : Info   : IP   : Static IP : 1XX.XXX.XXX GW: 1XX.XXX.XXX SN: 255.255.255.0 DNS: 1XX.XXX.XXX
126153 : Info   : WIFI  : Start network scan
126154 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS ESPeasy internal wifi status: DISCONNECTED
128352 : Info   : WIFI  : Scan finished, found: 8
128361 : Info   : IP   : Static IP : 1XX.XXX.XXX GW: 1XX.XXX.XXX SN: 255.255.255.0 DNS: 1XX.XXX.XXX
128362 : Info   : WIFI  : Start network scan
128363 : Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS ESPeasy internal wifi status: DISCONNECTED
....

fly74, 2021-02-23:
Flash Wlansettings with ESP.Easy.Flasher don't  work too.

fly74, 2021-02-23:
@TD-er It look like this option is not migrating:

[image]

when I unhide the SSID I can connect and set the option and then hide the SSID. Can you have a look at this?

TD-er, 2021-02-23:
It was never stored as a setting, but I guess the default is better set to "checked" for this one.

fly74, 2021-02-23:
@TD-er  Yes, if updating from an earlier release without the setting it should be set as default. thx

closed 2021-02-23 by fly74