letscontrolit/ESPEasy #3523: Issue with P079 since mega-20201227

TungstenE2, 2021-02-24:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [x] Tested on other node to make sure hardware isn't defective.
- [x] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Hi all,

I recently discovered an issue with my chicken door operated using the P079 Wemos motor shield plugin.

https://espeasy.readthedocs.io/en/latest/Plugin/P079.html#p079-page

Installation and rules were running stable for a long time. I recently updated to ESP_Easy_mega_20210114_normal_ESP8266_4M1M.bin
and I randomly discovered issues with my sunrise rule to open the door.

I realized that there where cleanup activities to the plugin with Release mega-20201227.

May be these cleanups are causing the issue?

@TD-er: could you check if the clean up done with mega-20201227 caused an issue?

I just rolled back to ESP_Easy_mega_20201130_normal_ESP8266_4M1M.bin and this seemed to fix the issue.

thx

### Expected behavior

see above

### Actual behavior

see above
### Steps to reproduce

1. see above
2. 
3. 

### System configuration

Hardware: Wemos D1 mini with wemos motor shield

ESP Easy version: mega-20201227

ESP Easy settings/screenshots: 
[image]

### Rules or log data

```

```

TD-er, 2021-02-24:
What kind of issues do you experience?

TungstenE2, 2021-02-24:
Oh sorry, I missed this detail.

The open door command in the rule is not correct executed.  The door remains closed. 
On the other hand I am able to open the door using my little android app or by manual switch pushing which also executes a rule. I did not change any rule for several months. But after I installed the mega-20201227 with the code cleanup I discoverd issues just with the door opening rule in the morning. Everthing else is still working as expected. Just the chickens are locked in... ;-)

TD-er, 2021-02-24:
Ohoh, I'm seeing an angry chick.

TungstenE2, 2021-02-24:
Let me know if you need additional info.

Yesterday I rolled back to  ESP_Easy_mega_20201130_normal_ESP8266_4M1M.bin and this morning everything was working as expected.

TungstenE2, 2021-02-25:
Additional info, the rule for DoorOpen is the one having an issue.

With ESP_Easy_mega_20201130_normal_ESP8266_4M1M.bin this is working fine again the last 2 days.

//Events

On DoorOpen Do
If [Reed1-open#Switch]!=1
	**WemosMotorShieldCMD 0 Backward 50**
	TaskValueSet 7,3,1
endif
EndOn

On DoorClose Do
    If [Reed2-close#Switch]!=1
  	WemosMotorShieldCMD 0 Forward 50
	TaskValueSet 7,3,1
endif
EndOn

On DoorStop Do
	WemosMotorShieldCMD 0 Stop
	TaskValueSet 7,3,0
EndOn

TD-er, 2021-02-25:
I do notice you have a `-` sign in the task name.
Not sure if that will work fine.

TungstenE2, 2021-02-25:
Working fine for about 2 years now.

TD-er, 2021-02-25:
> Working fine for about 2 years now.

Well since last build(s) it isn't and as there has been quite a lot of updates in the rules parsing and not so much in the motorshield code, I wonder if it is the rules I should look into or in the motorshield (or something inbetween)

TungstenE2, 2021-02-25:
so should I change '-' to '_' in the Task/Device Name?

[image]

TungstenE2, 2021-02-25:
still using the old Rules Engine

[image]

TD-er, 2021-02-25:
That "old" engine label shoulf be renamed to something else, as the "new" engine is not going to be used in the future as it is not really usable and there are other ways to speed up rules parsing.
So "old" is good :)

You can try to use an underscore, or use no character between them.
Task names and task variable names are not case sensitive, so you can use camel case if you would like to make it clear.

TungstenE2, 2021-02-25:
Ok, need to find sometime to test. The only problem ist that the issue only occurs for the morning rule to open the door by sunrise.
And it seems like it is only randomly not working. First I discovered that Task 7 was missing dummies 2 and 3 after an update. So I added them again and the next day it was working again. But failed again 1 or 2 days later again. I than rolled back to older version and it was working again directly.

TungstenE2, 2021-03-04:
@TD-er did some testing this morning.

- removed '-' from task/device names
- updated all rules accordingly
- updated FHEM devices and doifs accordinly as well
- verfied every thing still working (incl Telegram messages upon close/open)
- updated from  ESP_Easy_mega_20201130_normal_ESP8266_4M1M.bin to ESP_Easy_mega_20210223_normal_ESP8266_4M1M.bin
- this is when the trouble started...
- got browser message: Low memory. Cannot display webpage :-(
- Load 100.00% (LC=1), usually this was much lower
- Tried to controll the door using my android app.
- Door was working but with big delay
- Door got stuck, as the upper reed sensor did not stop the door due to the delay and system load
- had to crawl with a screw driver into the chicken house and fix the stuck door
- roll back after 2 hrs to ESP_Easy_mega_20201130_normal_ESP8266_4M1M.bin
- system load now 63.46% (LC=1372)
- at least I left the house during corona for a short time, even it was raining... ;-)

please advice

TungstenE2, 2021-03-04:
did another test....

update from ESP_Easy_mega_20201130_normal_ESP8266_4M1M.bin to ESP_Easy_mega_20210114_normal_ESP8266_4M1M.bin

Seems to work and system load is about 65%.

Just need to see if the sunrise rule is working again, as reported in the beginning of this issue.

TD-er, 2021-03-04:
Maybe also good to test a build first before potentially bothering the chickens? ;)
Do you use "hidden SSID" access points?

TungstenE2, 2021-03-04:
no hidden SSID used, but by Unifi APs are using a hidden SSID, not sure why exactly.

As I have no test setup with spare motorshield etc the chicken are my testers. And trust me, they find every bug... ;-)

TD-er, 2021-03-04:
They do sound rather picky ;)

TungstenE2, 2021-03-05:
update: I can confirm that ESP_Easy_mega_20210114_normal_ESP8266_4M1M.bin is working again as expected after removing '-' from task/device names and rules.

But as staded yesterday ESP_Easy_mega_20201130_normal_ESP8266_4M1M.bin seems to have an issue with load and low memory.

TD-er, 2021-03-05:
I'll keep an eye out for other (probably WiFi related) issues with the latest build.
Until now there have been rather surprisingly few reported issues with the latest build so far.

TungstenE2, 2021-03-14:
Hi @TD-er the issue is back. On Sat and Sun I have a differnt configuration during weekdays. On weekdays the sunrise time triggers rule to open the door. On Weekends I set it to 8:30am. Yesterday and today I could verify that the rule starts the motor for 1 sec and than stops again. I checked the Log and it seems that the close reed triggers a DoorStop Event. But I do not understand this. I did not change these rules and it was working fine for a long time. The issue only occured after I started updating. During the week with %sunrise% this is working fine. Also manual opening and closing is working fine, and the same rules for DoorStop are applied.

Any idea?

[image]

[image]

[image]

TD-er, 2021-03-14:
Looks like it may have been a 'bounce' of the reed switch?
Could it be that the switch may have been triggered by some mechanical vibration?

TungstenE2, 2021-03-14:
I had the same idea, but I have some doubts. Why would this only happen on weekends. And today I was watching the door and did not see any unusual movement. I will further observe it without changing the settings. I would bet that Mo-Fri it will work as expected. Will let you know tomorrow morning... ;-)

TD-er, 2021-03-14:
Can you check with the rules using 08:30:00 instead of 08:30 ?

TungstenE2, 2021-03-14:
sure I can test, but I doubt this beeing the reason, as the event DoorOpen is triggered correct, Some how the event DoorStop is triggered to early on Sat/Sun.

tonhuisman, 2021-03-14:
You haven't set a de-bounce value on your reed switch (according to the screenshot above). For any switch I'd use a de-bounce value, especially for an end-switch that stays in a specific state when that position is reached, as there is no expected reason to go to that state again, unless the door is moved in the other direction first.
And we can't see your DoorOpen and DoorStop code, does that reset the timers in some way?

And having several event handlers for %sunrise% and probably also %sunset%, you could put that into 1 event handler (to lower system load a little) that then checks %sysweekday%, f.e. like this:
```
...
on Clock#Time=All,%sunrise% do
  if %sysweekday%=1 or %sysweekday%=7
    LogEntry,"Weekend! %eventvalue1%"
    // Not working day
  else
    LogEntry,"Workday mon-fri: %eventvalue1%"
    TimerSet,1,60 // Working day
  endif
endon
```
The events for Saturday and Sunday 08:30 you should probably keep.

TungstenE2, 2021-03-14:
@tonhuisman thx for your feedback and input.

Here is my DoorOpen, DoorClose and DoorStop code.
[image]

What exactly do you mean by 'de-bounce' value. Can you give an example of your code used?

My setup was workingt fine and stable for a very long time, issues started with mega-20201227 I assume, but hard to say what caused the issues.

tonhuisman, 2021-03-14:
> What exactly do you mean by 'de-bounce' value. Can you give an example of your code used?

De-bouncing is stabilizing the action of a switch, the state of the switch has to be in the 'on' state for the de-bounce time before it is accepted and acknowledged as being changed. This filters out spurious changes and signal-spikes that can be picked up by the wiring from external sources, as long wires act as an antenna.
A switch accepts up to 250 msec as the de-bounce value. For a switch that needs to give a steady but quick response, I'd set it to 50, if stability is more important, and response doesn't really matter much, a value between 100 and 200 msec seems feasible.
As reed switches aren't the most firm construction, but your setup requires a prompt response, I'd set it to 50. If it responds too slow, a value of 20 might be just right.

tonhuisman, 2021-03-14:
> issues started with mega-20201227 I assume, but hard to say what caused the issues.

I think that in that release, many code improvements are included, making ESPEasy more responsive, IMHO, making setups, depending on previous 'delays' or 'slugishness', somewhat less stable, even though the overall stability and speed have improved.
Adding some stabilizing settings, like the de-bounce value for switches (I use that on all switches used with my ESP's, to avoid soldering capacitors on input pins), will improve your projects.

TungstenE2, 2021-03-14:
ok, now I got you. You are referring to the switch advanced event managment value 'De-Bounce'.

To be honest I never realized what this is for. I now set it to 50, as suggested. 
[image]

Thanks a lot for this advice!

giig1967g, 2021-03-14:
Hi all,
I am the one who coded the de-bounce and safe button.
First of all, let's say that switches states are checked every 100ms (10 times per second). So any value below 100ms is useless.

De-bounce:
When you activate a switch, the de-bounce "ignores" every state change within the set interval.
So if switch oscillate between 0 and 1 within the set interval, these state changes do not trigger an event. 
A correct de- bounce value is 100-200ms, as 50ms it's too low as the single cycle is 100ms.

"safe button":
The safe button waits two 100ms cycles before firing the trigger, in other words, the switch must remain in the state position for at least 100ms. It avoids false positive. It's drawback is that you have to keep the button pressed for at least 100ms. A very quick press and release might not fire the event.

If you need more explanations please ask.

TungstenE2, 2021-03-14:
Hi @giig1967g, thx for explanations.

So can I assume this to be the right configuration for my setup?

[image]

TungstenE2, 2021-03-14:
btw: shouldn 'Use Safe Button' be named 'Safe Button', as the check box means 'use'?

giig1967g, 2021-03-15:
@TungstenE2
Yes, that should be fine.

Regarding the label, you are right! Will change it.

TungstenE2, 2021-03-15:
@giig1967g thx.

Regarding the label I would even suggest to name it different. The device is a switch. 
So may be 'Safe Switch', 'Safe State', 'Safe Switch State' or similar would be more intuitive and user friendly.

tonhuisman, 2021-03-15:
> switches states are checked every 100ms (10 times per second). So any value below 100ms is useless.

OK, but looking at the code I'd guestimate that any value > 10 will result in a 100 msec de-bounce. So that's still good enough.

> Regarding the label, you are right! Will change it.

Will that include the (partially missing ReadTheDocs (RTD)) documentation? (As the Wiki documentation is _really_ outdated)

TD-er, 2021-03-15:
Actually the functionality is "Delay Accept".

giig1967g, 2021-03-15:
> OK, but looking at the code I'd guestimate that any value > 10 will result in a 100 msec de-bounce. So that's still good enough.

ok, I have coded it few years ago... I might have forgotten something :(

> Will that include the (partially missing ReadTheDocs (RTD)) documentation? (As the Wiki documentation is _really_ outdated)

Will do my best to update the documentation.

> Actually the functionality is "Delay Accept".

You want me to change the label to "Delay Accept"?

TungstenE2, 2021-03-15:
I am getting confused.... 
How can value 10 result in 100 msec if the vlaue is labled with (mesc)?

tonhuisman, 2021-03-15:
> How can value 10 result in 100 msec if the vlaue is labled with (mesc)?

The check is done every loop (100 msec). The first de-bounce check is done in the first loop, but the time passed since that first press will (most likely) be less than 10 msec, so it will not be accepted. In the next loop, 100 msec later, the state should be accepted, assuming the state hasn't changed, effectively giving ~100 msec de-bounce delay.

TD-er, 2021-03-15:
Maybe we should then set a minimum of 100 in the input field?

TD-er, 2021-03-15:
> You want me to change the label to "Delay Accept"?

Not the best name I guess, so if you have a better suggestion?
The label should describe what it does from the user's viewpoint.
"Safe button" what it is called now, is more a generic intention and not describing what it does.
But since the users of ESPEasy are a bit more tech minded, I think it should describe the actual effect.

So maybe it is more like "Require extra 100 msec state duration", but that's too long for a label. (could be a note)

So we're now at the hardest part of programming... thinking of a good descriptive name for a parameter.

giig1967g, 2021-03-15:
From the user point of view we could describe the effect:
we could call it "Avoid false positive"

TD-er, 2021-03-15:
That's also an intended "side effect" and not describing what it does.

What it does is, it will make sure the state is stable for another 100 msec.
Essentially extending the "debounce time", right?

"Avoid false positive" is some promise you can't keep up.

tonhuisman, 2021-03-15:
If stated like that, even '(Use) Safe Button' is such a promise...

Possibly this setting should be removed, and a clear note added to the De-bounce setting. (And also allow for somewhat longer de-bounce time than 250 msec, possibly up to 500 msec)

TD-er, 2021-03-15:
The debounce period may allow a few triggers inbetween.
The now called "Safe Button" mode requires the state to remain stable for 100 msec.
So it is not really the same.

It is more like "Require 100 msec stable state" :)

TungstenE2, 2021-03-15:
shouldn't the feature be moved below the 'De-Bounce' value, as they are related?

[image]

TD-er, 2021-03-15:
Yep.

TungstenE2, 2021-03-15:
Hi all,

so what are the next steps and who needs to take a decission?

I see several options and different views.

De-bounce:
- set minimum value

Safe-Button:
- renaming to 'De-bounce plus' with detailed description in documentation what 'plus' means
- or may be even removing
- moving the feature below the de-bounce setting

Documentation:
- Updating the documentation with examples on how to use the settings.

giig1967g, 2021-03-15:
@TungstenE2 
first of all, does the de-bounce setting solved your problem?

TungstenE2, 2021-03-15:
@giig1967g in order to answer this I would suggest that I go on with my testing for the next week.
I have 2 reed switches. The issue only occured for the lower reed indication the door is closed and for the DoorOpen event in the morning. Not 100% sure if it only happend on weekends. 

Yesterday I changed the settings as suggested and this morning it was working fine. But last week it was also working fine Mo-Fr. So lets wait for the weekend.

But the de-bounce settings sound like a solution to such a situation with reed switches not beeing 100% reliable.

But independed of my issue the topics I listed should be be solved in general, right?

giig1967g, 2021-03-15:
Please keep us informed if the issue is solved, so we can close this incident.

Regarding your comments, they are absolutely valid. Will make the changes as soon as I have time (and Gijs appreves them).

TungstenE2, 2021-03-16:
bad news, the issue reoccured this morning. The door did not fully open, but stopped after about 1 sec. So I went outside and checked the door and reed sensors. The door is a bit shaky in the close position and more firm in the open position. So the issue is caused as the door is moving a bit and the reed is triggered and since the latest releases the software is more sensitive.

[image]

I played a bit with the settings and de-bounce 200 and safe-button on was not a good combination. Seems like this caused the door even to bypass the close reed, as the reed usually stops the door a bit over the ground.

Possible solutions:
- changing the door sliders so the door can not move any more in the lower position, in order to prevent false reed switch states. But this is a bit hard, as the door is hard to reach without to ball out my right arm...
- solving this by setup configuration, but de-bounce and safe button did not fully do the trick by now
- changing the rules in order to prevent this false reed triggers
- changing the speed of the motor

TD-er, 2021-03-16:
Just for understanding... Were the days when the door switch "bounced" a bit more wet compared to other days?
A wet door may jam slightly and thus causing a more shaky movement.

TungstenE2, 2021-03-16:
no, not really. The door and the sliders are aluminium, so rain/wet should have no impact. the sliders are just not 100% straight, as it was hard to fix the screws. I only could work through a tiny side door.

TungstenE2, 2021-03-20:
small update:

3 days ago I repositioned the right door slider a bit, so the door has less space to move in the close position.
I also set the de-bounce value to 250 (this is max value) and the safe button.
This was working fine for 2 days, Thur, Fri when the door opens by sunrise.

But today (Sat) the door rules are supposed to open the door by 8:30 and again the door open rule started to open the door, but stoped again after 1 sec.

So I started to make up my mind what might be the difference between Fri and Sat. 
Differences are:
- fixed time for Sat and Sun, instead of sunrise
- at 8:30 the chicken are already waiting at the door to open, while at sunrise they are still sitting

So my assumption is, that the chicken waiting at the door makes the door magnet moving slightly and the reed contact is too sensitive and triggers the DoorStop rule, right after the motor was started. Seems like that recent code changes result in too sensitive and fast triggering, which was not the case before for a long time.

So I see following options for me:

- changing the rules for weekends to open also with sunrise (my nighboors will get angry in summertime...)
- changing the rules, that after motors was started, the DoorStop rule should not be triggered for 3 sec. (Not sure how to implement)

Further ideas?

tonhuisman, 2021-03-20:
>   the DoorStop rule should not be triggered for 3 sec. (Not sure how to implement)

I'd code that something like this: (partial code only, replacing some existing code)
```
On Clock#Time=Sat,08:30 do
  Let 100,1 // Disable DoorStop temporarily
  TimerSet 3,3 // Delay for disabled DoorStop to be re-enabled
  TimerSet 1,60
endon

On Clock#Time=Sun,08:30 do
  Let 100,1 // Disable DoorStop temporarily
  TimerSet 7,3 // Delay for disabled DoorStop to be re-enabled
  TimerSet 1,60
Endon

On Reed1open#Switch=1 do
  If %v100%=0
    AsyncEvent,DoorStop // Queue command
    GPIO 15,0
  Endif
Endon

On Reed2close#Switch=1 do
  If %v100%=0
    AsyncEvent,DoorStop // Queue command
    GPIO 15,0
  Endif
Endon

On Rules#Timer=7 do
  Let 100,0 // Allow DoorStop to be activated again
Endon
```

TungstenE2, 2021-04-03:
Hi all, here comes a final update.

I updated the rules as suggested (syntax needed to be adjusted), and chicken door is working as expected again.
Needed to enable a delay, as the new code is to sensitive for short reed switch triggers.

Thx for your support. Ticket can be closed. Please remember the changes discussed.

Rule set 1
```

On System#Boot do
  TaskValueSet 7,3,0 // DC Motor off
  endon

//## Taster Functions

//close door if Taster1 is pressed and door is not closed

On Taster1close#Switch do
	If [Taster1close#Switch]=1 and [Reed2close#Switch]!=1
		Event,DoorClose
	Else
		
endif
endon

//##

//close door if Taster2 is pressed and door is not open

On Taster2open#Switch do
	If [Taster2open#Switch]=1 and [Reed1open#Switch]!=1
		Event,DoorOpen
	Else
		
endif
endon
```

Rule set 2

```

//## Automatic Stop Functions

//stop motor if Reed1 or Reed2 triggered

On Reed1open#Switch=1 do
  If %v100%=0
    AsyncEvent,DoorStop // Queue command
    GPIO 15,0
  Endif
Endon

On Reed2close#Switch=1 do
  If %v100%=0
    AsyncEvent,DoorStop // Queue command
    GPIO 15,0
  Endif
Endon

//##
```

Rule set 3

```
//at sunrise time set timer1 for 1 Sec

// On Clock#Time=All,%sunrise% do 
//  Let,100,1 // Disable DoorStop temporarily using variable v100 set =1
//  TimerSet,1,1 //TimerSet 1 for DoorOpen Event, 1 Sek
//  TimerSet,7,3 // Delay for disabled DoorStop to be re-enabled, Timer 7, 3 Sek
// endon

On Clock#Time=Mon,%sunrise% do 
  Let,100,1 // Disable DoorStop temporarily using variable v100 set =1
  TimerSet,1,1 //TimerSet 1 for DoorOpen Event, 1 Sek
  TimerSet,7,3 // Delay for disabled DoorStop to be re-enabled, Timer 7, 3 Sek
endon

On Clock#Time=Tue,%sunrise% do 
  Let,100,1 // Disable DoorStop temporarily using variable v100 set =1
  TimerSet,1,1 //TimerSet 1 for DoorOpen Event, 1 Sek
  TimerSet,7,3 // Delay for disabled DoorStop to be re-enabled, Timer 7, 3 Sek
endon

On Clock#Time=Wed,%sunrise% do 
  Let,100,1 // Disable DoorStop temporarily using variable v100 set =1
  TimerSet,1,1 //TimerSet 1 for DoorOpen Event, 1 Sek
  TimerSet,7,3 // Delay for disabled DoorStop to be re-enabled, Timer 7, 3 Sek
endon

On Clock#Time=Thu,%sunrise% do 
  Let,100,1 // Disable DoorStop temporarily using variable v100 set =1
  TimerSet,1,1 //TimerSet 1 for DoorOpen Event, 1 Sek
  TimerSet,7,3 // Delay for disabled DoorStop to be re-enabled, Timer 7, 3 Sek
endon

On Clock#Time=Fri,%sunrise% do 
  Let,100,1 // Disable DoorStop temporarily using variable v100 set =1
  TimerSet,1,1 //TimerSet 1 for DoorOpen Event, 1 Sek
  TimerSet,7,3 // Delay for disabled DoorStop to be re-enabled, Timer 7, 3 Sek
endon

On Clock#Time=Sat,07:30 do 
  Let,100,1 // Disable DoorStop temporarily using variable v100 set =1
  TimerSet,1,1 //TimerSet 1 for DoorOpen Event, 1 Sek
  TimerSet,7,3 // Delay for disabled DoorStop to be re-enabled, Timer 7, 3 Sek
endon

On Clock#Time=Sun,07:30 do 
  Let,100,1 // Disable DoorStop temporarily using variable v100 set =1
  TimerSet,1,1 //TimerSet 1 for DoorOpen Event, 1 Sek
  TimerSet,7,3 // Delay for disabled DoorStop to be re-enabled, Timer 7, 3 Sek
endon

// Allow DoorStop to be activated again after Timer 7 has expired, set variable v100 =0

On Rules#Timer=7 do
  Let,100,0 
Endon

 //at sunset set timer2 for 20min

On Clock#Time=All,%sunset% do 
        timerSet,2,1200
endon

//When Timer1 expires, DoorOpen and set Day 1

 On Rules#Timer=1 do  
	If [Reed1open#Switch]!=1
		Event,DoorOpen
		TaskValueSet 7,1,1 
	Else
		Event,DoorStop
endif
endon
 
 
 
 
 //When Timer2 expires, DoorClose and set Day 0

 On Rules#Timer=2 do  
	If [Reed2close#Switch]!=1
		Event,DoorClose
		TaskValueSet 7,1,0
	Else
		Event,DoorStop
endif
endon
```

Rule Set 4

```
//Events

On DoorOpen Do
If [Reed1open#Switch]!=1
	WemosMotorShieldCMD 0 Backward 50
	TaskValueSet 7,3,1
endif
EndOn

On DoorClose Do
    If [Reed2close#Switch]!=1
  	WemosMotorShieldCMD 0 Forward 50
	TaskValueSet 7,3,1
endif
EndOn

On DoorStop Do
	WemosMotorShieldCMD 0 Stop
	TaskValueSet 7,3,0
EndOn
```

TD-er, 2021-04-03:
Just one remark.
The "weekend" could be matched (or at least "should" :) ) with `wkd` and the other days of the week with `wrk`.
That could make your rules slightly shorter.
But please test it as we don't want to make them being locked up for too long.

TungstenE2, 2021-04-03:
nice, thx for the advice.

Where to find documentation on this?

Not mentioned in Rules documenttaion currently.
https://espeasy.readthedocs.io/en/latest/Rules/Rules.html

TD-er, 2021-04-03:
> nice, thx for the advice.
> 
> Where to find documentation on this?
> 
> Not mentioned in Rules documenttaion currently.
> https://espeasy.readthedocs.io/en/latest/Rules/Rules.html

I think the question of "where" should be changed into "when" ;)

[7 more replies]

closed 2023-09-02 by TD-er