letscontrolit/ESPEasy #3524: Feature "Single event with all values" is not working

beeyev, 2021-02-24:
I have a successfully connected device BME280, which produces three different values (Temperature,Humidity,Pressure)
I want these values to be submitted to a remote server using Generic HTTP Controller.
So in order to send these three values ESPEasy  makes three GET requests, but when I use _Single event with all values_ it does not change anything.
Am I missing something? Is it possible to submit all three values in one GET request?

tonhuisman, 2021-02-24:
This is about the events fired for rules, not to a Controller.
It will fire a single event `<taskname>#All=<value1>,<value2>,<value3>`

If you want to send that to a HTTP endpoint with a single call, then you could add a rule like this:
```
on BME280#All do
  SendToHTTP,<host>,80,"/url/etc?value1=%eventvalue1%&value2=%eventvalue2%&value3=%eventvalue3%
endon
```
The Controller can then be disabled (unchecked) from the task configuration.

beeyev, 2021-02-24:
Thank you for the help.

TD-er, 2021-02-24:
So I guess this issue is now asked and answered, so it can be closed.

closed 2021-02-24 by TD-er