letscontrolit/ESPEasy #3526: ESP OTA Update Wroom02

fraeggle, 2021-02-25:
Trying to make a OTA update from ESP_Easy_mega_20210114_normal_WROOM02_2M256.bin with Release mega-20210223 and a WROOM02.
Web Page tells me: Max sketch size 1019 kb (1044464 bytes)
But OTA fails every time with not enoug space.
[image]

Did i forgot something?
Two step OTA with ESPEasy_2step_UploaderMega_1024.bin is not working, because WROOM02 does not boot with this one.

Any suggestions?

Regards Peter

tonhuisman, 2021-02-25:
After installing the ESPEasy_2step_UploaderMega_1024.bin that failed to boot, you will have to do a serial upload I presume, so you could then upload the new bin file directly?

fraeggle, 2021-02-25:
a direct upload via Flasher is always possible without loss of config. but OTA is not possible anymore. Whether I use ESP_Easy_mega_20210114_normal_WROOM02_2M256.bin or ESP_Easy_mega_20210223_normal_WROOM02_2M256.bin.
If i use f.ex. mega-20201227 i can make a OTA to mega_20210114 or mega-20210223. But once of of these two is active i can not make a OTA to a other firmware..
Alwasy get not enough space

fraeggle, 2021-02-25:
[Wroom02_OTA_2Step.zip](https://github.com/letscontrolit/ESPEasy/files/6042994/Wroom02_OTA_2Step.zip)
Made a "own" 2Step WROOM02-256 OTA with 3 embedded plugins like switch,MQTT, Sysinfo. Just for testing. With this one i can update from ESP_Easy_mega_20210114_normal_WROOM02_2M256 to ESP_Easy_mega_20210223_normal_WROOM02_2M256
So i does look like Filesize is really too big.

tonhuisman, 2021-02-25:
Size has been an issue recently (again), there are a couple of builds no longer fitting in the max bin size for their predetermined configurations (or the other way around 😆)

TD-er, 2021-02-25:
The message may be a bit confusing I guess (and actually giving you wrong advice).

What is happening here:
2M flash layout differs from 1M flash layout, so the 2-step OTA image cannot be used here as it cannot read your settings.
The check for 2-step OTA does check if the amount of free space for OTA is less than the current sketch size.
So what you need to do here is flash an image which does fit in the OTA space.
On 2M256 flash layout, that's roughly 750k if you have the largest possible binary loaded.  (256k file system, largest binary ~1M)

There is however a nice work-around for 2M flash users.
Since core 2.7.x it is possible to perform an OTA via a .bin.gz file.
For example, if your sketch is 1M in size, you only have roughly 700 - 750k of free space to upload an OTA image.
If (you need to check yourself!!!) your new .bin.gz file is smaller than the amount of free OTA space, you can flash a .bin.gz image, which will then be extracted to the beginning of the flash at the first boot.

fraeggle, 2021-02-25:
Hi TD-er:
this is what I almost thought. Therefore it will work with my "slimmed down" file, right? Tomorrow I will try your described way with the .gz file.
I will let you know tomorrow.
;-)

TD-er, 2021-02-25:
The .gz file should also be present in the nightly build ZIP files, so maybe that's the easiest way to test?
Make sure to backup your settings to be sure, as there might be a possibility you chose a file which is too large and thus overwrite (parts of) the file system.

fraeggle, 2021-02-25:
I just tried it out. Yes the .gz is in the Builds. The update works fine with the .gz file. If I had known before that it works like this......
Thanks for your help.

Greetings Peter

closed 2021-02-25 by fraeggle