letscontrolit/ESPEasy #3528: ESPEasy wont boot if MQTT connection fails (C002 plugin)

svollebregt, 2021-02-28:
### Checklist
- [Y] The title of this issue is "to the point" and descriptive.
- [Y] This issue describes what is happening.
- [Y] This issue describes what components are affected (e.g. name of plugin/controller)
- [Y] This issue describes how to reproduce it.
- [NA] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [Y] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [NA?] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [Y] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [Y] given a list of active plugins or controllers when applicable.
- [Y] filled out all applicable fields below.

### Steps already tried...
- [NA ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [Not tried, used mega20210223] Tested previous/other build (mention which one already tested)
- [Tried two nodes: exact same behaviour] Tested on other node to make sure hardware isn't defective.
- [Uknown, only use C002 for communication] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

A few days ago I updated my mosquitto broker on my RPi to 2.0.8. As a result, anonymous _and_ non-localhost connections were blocked as the config changed. I noticed that my ESPEasy nodes failed to reboot till I resolved this issue. When the reboot command was used in the web interface they never reconnected to my WiFi. After resolving the mosquitto issue and a hard reboot this issue was resolved

### Expected behavior

If MQTT fails to connect the web interface should still be reachable

### Actual behavior

It seems that the start-up hangs when the MQTT broker cannot be reached. At least for a soft reboot.

### Steps to reproduce

1. Block anonymous traffic in mosquitto and/or block non-localhost (e.g. only listen to 127.0.0.1)
2. Reboot ESPEasy using web interface

### System configuration

Hardware: Wemos D1 Pro

ESP Easy version: mega20210223_normal_ESP8266_4M1M.bin
Plugins used: C002, P003, P014

ESP Easy settings/screenshots: C002 with anonymous login, no other settings beside default ones

[image]

tonhuisman, 2021-02-28:
Strange that a failing MQTT connection stops your ESP from booting, as that Controller is only started _after_ the ESP is booted (and connected to WiFi).
I suspect there is some other force blocking your configuration, not the MQTT controller. 🤔 
For better debugging, you could start a serial monitor on the USB/Serial port of your ESP to see what actually happening during startup.

TD-er, 2021-02-28:
Only thing I can imagine (without having tested myself) is that ESPEasy may trigger some reboot and quickly reboots.
So I agree with @tonhuisman that a serial log would be useful here.

svollebregt, 2021-04-11:
I am having troubles recreating this behaviour. I guess you are right that something else in my LAN caused this behaviour.

closed 2021-04-11 by svollebregt