letscontrolit/ESPEasy #3529: DHT11 not shown in ESPEasy Devices.

b0nete, 2021-02-28:
Hi, i'm reading docs and guides for connect my DHT11 to ESP8266. 
I connected it but when i try add my DHT11 in Devices tab i cant see "DHT Humidity Sensor" like see in guides.

I'm reading a lot but not find anything related; i dont know if have to add any module or plugin.

Thanks!, sorry for my stupid question.

tonhuisman, 2021-03-01:
What .bin file did you upload to your ESP? The DHT11 sensor is available in the builds with 'normal' or 'test' in the name, it looks like you have used a 'custom' or 'dev' build.
What ESP board are you using?

TD-er, 2021-03-01:
It is has the "AcuDC24x in it, which is currently only present in the custom build.
So I suggest to go for a "normal" build
Please pay attention to use the build for the right flash size you need (e.g. with "4M1M" in the name for 4M flash)

tonhuisman, 2022-04-18:
This seems to be resolved, so can be closed.

closed 2022-04-18 by TD-er