letscontrolit/ESPEasy #3531: P093 - please update documentation

TungstenE2, 2021-03-02:
@crnjan thanks for the great plugin P093! 

But it would also be great if you could keep the documentation updated.

https://espeasy.readthedocs.io/en/latest/Plugin/P093.html

Please see for example the findings by Eisbahn on settings using MQTT.

https://www.letscontrolit.com/forum/viewtopic.php?f=2&t=7678&p=50932#p45694

thx a lot!

crnjan, 2021-03-02:
I guess this is duplicate of https://github.com/letscontrolit/ESPEasy/issues/3158.

But wondering - which part of documentation is not up to date? Please note MQTT specifics are not defined by plugin, but rather how EeasyESP works ... There is also a link to a fully working example within the [doc](https://community.openhab.org/t/mitsubishi-heat-pump/91765) including MQTT topics and payloads

```
Type number:acRoomTemperature [ stateTopic="FF_Nik_Room/ac/settings", transformationPattern="JSONPATH:$.roomTemperature" ]
      Type number:acTemperature [ stateTopic="FF_Nik_Room/ac/settings", transformationPattern="JSONPATH:$.temperature", commandTopic="FF_Nik_Room/ac/settings/cmd", formatBeforePublish="MitsubishiHP,temperature,%d" ]
      Type string:acPower [ stateTopic="FF_Nik_Room/ac/settings", transformationPattern="JSONPATH:$.power", commandTopic="FF_Nik_Room/ac/settings/cmd", formatBeforePublish="MitsubishiHP,power,%s" ]
      Type string:acMode [ stateTopic="FF_Nik_Room/ac/settings", transformationPattern="JSONPATH:$.mode", commandTopic="FF_Nik_Room/ac/settings/cmd", formatBeforePublish="MitsubishiHP,mode,%s" ]
      Type string:acFan [ stateTopic="FF_Nik_Room/ac/settings", transformationPattern="JSONPATH:$.fan", commandTopic="FF_Nik_Room/ac/settings/cmd", formatBeforePublish="MitsubishiHP,fan,%s" ]
      Type string:acVane [ stateTopic="FF_Nik_Room/ac/settings", transformationPattern="JSONPATH:$.vane", commandTopic="FF_Nik_Room/ac/settings/cmd", formatBeforePublish="MitsubishiHP,vane,%s" ]
      Type string:acWideVane [ stateTopic="FF_Nik_Room/ac/settings", transformationPattern="JSONPATH:$.wideVane", commandTopic="FF_Nik_Room/ac/settings/cmd", formatBeforePublish="MitsubishiHP,wideVane,%s" ]
```

Or did I missed something?

TungstenE2, 2021-03-02:
@crnjan eg. samples on http comand syntax are missing and would be nice to mention.

commad syntax:
http://<ESP IP address>/control?cmd=MitsubishiHP,power,ON

Referring to external pages for syntax is not very userfriendly.

TungstenE2, 2021-03-02:
also documentation samples on how to send MQTT commands to the plugin would be nice. By now I only managed to subcribe to the MQTT updates from the Plugin, but not to send MQTT commands. Only using http commad. Would be much user friendly to have samples on how to use the plugin in the plugin documentation.

thx a lot

TungstenE2, 2021-03-05:
@crnjan also information is missing that the plugin is currently not fitting into any build and has to be custom build.

tonhuisman, 2021-03-05:
> that the plugin is currently not fitting into any build and has to be custom build

That's not something that should go in the main documentation, as it has nothing to do with the plugin itself, it's just a technical issue.

OT:
I'm working on a build configuration for ESP32 with 16 MB Flash, that should be able to hold (nearly) all plugins at once (currently testing for the most optimal setup). I just defined a sketch size of 4 MB 😄

TD-er, 2021-03-05:
For ESP32 things are indeed different.
Would be nice though if we could have a "single binary" for all ESPEasy plugins and controllers.

@TungstenE2 It isn't as you stated, see the other reply I just made in the other issue.
Still we should move to custom builds ASAP as this keeps biting us in the a**.

TungstenE2, 2021-03-05:
Nice, but in my case I am using P093 on a Wemos D1 mini. This would mean I need to switch to ESP32 in futur?

I have no much experience with custom builds, but last time it took me quite some time to get a working one.
A proper tutorial in this case would be nice and helpfull. 

I see the main reason for the big success of ESPeasy in the easy way to use it. There are a lot of people out there with no developer background but highly interested in IoT, ESPeasy offers a way to build IoT projects without being a developer.

TD-er, 2021-03-05:
I am working on a build service (as in a website where you can tick the boxes, and get a build) but that's not there yet.
There is still too much to do, but it becomes really clear now that the current way of a "specific set of builds" is not the way we can continue.

Right now you don't need to switch to the ESP32, we need to make this build service work.

commit 8ab9865e (2022-05-31): Should resolve #3531 and #3158.