letscontrolit/ESPEasy #3532: ESP32 T-beam Neo-6M GPS not working

henkiejan1, 2021-03-02:
Hi all,

I thought to use my T-Beam for some speed measuring with the build-in gps module. Sadly it will not work. I got continuously ''invalid''. It's the T-Beam V1.0 I tried HWSerial0, HWSerial1, HWSerial2 and the gpio pins 34 and 12 as RX and TX and swapped but nothing works. I can't also never see where i can configure the serial speed. Should be 9600? 

Thank you for any help!

TD-er, 2021-03-02:
The GPS plugin does indeed set it to 9600 bps without an option to set it differently.

Doesn't the v1.0 and later have a power management chip?
I know you need to do stuff with that chip to get the OLED running for example if you use the pins suggested by LilyGo, maybe also for the GPS?
I only tested with the 0.7 revision board

The pins needed for rev 1.0 are:
[image]
So GPIO-34 is connected to the TX of the GPS (there is no data sent back)

henkiejan1, 2021-03-03:
> The GPS plugin does indeed set it to 9600 bps without an option to set it differently.
> 
> Doesn't the v1.0 and later have a power management chip?
> I know you need to do stuff with that chip to get the OLED running for example if you use the pins suggested by LilyGo, maybe also for the GPS?
> I only tested with the 0.7 revision board
> 
> The pins needed for rev 1.0 are:
> [image]
> So GPIO-34 is connected to the TX of the GPS (there is no data sent back)
> 
> Source: https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/tree/master/schematic

It's indeed very good possible that the power management chip need to be set. I tested with the Meshtastic and see that after some time the GPS is powered on and off. With the display i don'[t have any problems. thank you for your help again @TD-er ! :)

henkiejan1, 2021-03-03:
A update. Yes you are totally right! You need to configure the AXP192 on the I2C bus and request IRQ35 ot something. Little bit to difficult for me ;) I plug a stand alone GPS receiver on a ESP32 now. :)

TD-er, 2021-03-03:
So maybe we should turn this issue into a feature request for that power management chip to be supported in a plugin.

PR #4124 (2023-03-23): [P137] Add AXP192 Powermanagement controller
Add AXP192 Power management controller, as used in M5Stack ESP32 units (controls f.e. the display backlight on M5StickC) and LilyGO T-Beam units

Features
- Support for LDO2, LDO3, LDOIO, DCDC2, DCDC3 and GPIO0..GPIO4 pins of AXP192, with preset values
- Configurable values (1..4) can be sent to Controllers and are available as event values, with optional interval
- Only supported on ESP32 (limitation of the used library, and device hasn't been seen on ESP8266 boards)
- All available readable values from AXP192, like VBus, Input voltage & current, i [... cut]
closed 2023-03-23 by TD-er