letscontrolit/ESPEasy #3533: ESP-32 Access Point creation

Shark654, 2021-03-03:
Once i power up the esp32 with usb port(PC) Access point is created,but if power it up by 5V access point vanishes.

Second thing is that once AP is generated through USB powering i remove the USB supply and connect to a 5V supply AP remains as it is..so how to create AP directly through 5v

TD-er, 2021-03-03:
Some ESP32 boards are known to have issues when powered via 5V only (thus not via USB)
It looks like the USB to serial chip does keep the ESP in a not normal state.
Sometimes this can be solved by adding a capacitor between the EN pin and GND as this does delay the startup of the ESP.
You may need to experiment with values for this capacitor.
I think this was also mentioned on the forum, so you may find a proper value for this capacitor there.
I guess it will be something around 10 uF, but I can be off a factor 10 or more.

tonhuisman, 2022-04-18:
No further response, and no action needed, AFAICS, so probably can be closed

closed 2022-04-18 by TD-er