letscontrolit/ESPEasy #3537: P002: Incorrect Calibration Calculation

azsystem, 2021-03-09:
The calculation done in P002 is incorrect.

The function 'P002_applyCalibration' (starting at line 272) should be:

`float P002_applyCalibration(struct EventStruct *event, float float_value) {
  if (P002_CALIBRATION_ENABLED) // Calibration?
  {
    int   adc1 = P002_CALIBRATION_POINT1;
    int   adc2 = P002_CALIBRATION_POINT2;
    float out1 = P002_CALIBRATION_VALUE1;
    float out2 = P002_CALIBRATION_VALUE2;
	float val  = float_value;
	
	// Solve basic lineair equation:
	// y=ax+b
	// a = (out1 - out2) / (adc1 - adc2)
	// b = ((adc2 * out1) - (adc1 * out2)) / (adc2 - adc1)
	
	
    if (adc1 != adc2) AND (out1 != out2)
    {
      val = float_value * ((ou1-out2) / (adc1-adc2)) + (((adc2*out1)-(adc1*out2))/(out2-out1))
    }
  }
  return float_value;
}`

TD-er, 2021-03-09:
I converted your code into a PR, which makes it a bit easier to discuss to actually see the changes.
N.B. I changed your if-statement into valid C++ code (`AND` is not valid C++ and you missed extra braces)

TD-er, 2021-03-09:
I don't think this will even compile, as `float val` is not known when  you return it.
Have you checked this code yourself?
Also this only changes the result when calibration values were the same.
The old behavior is to return 0 when both calibration values are the same (whether that's correct is a good question).
However you can't compare floating point values like this.

I don't think the resulting value is correct as I am not sure if there will be rounding errors due to casts to int inbetween. (the "needs some additional cast functions" remark you made)

azsystem, 2021-03-09:
old approach:
val = (x - adc1) / (adc2 - adc1) * ((out2 - out1) + out1)

new approach:
val = x * ((out1 - out2) / (adc1 - adc2)) + (((adc2 * out1) - (adc1 * out2)) / (adc2 - adc1))

Where X is the measured ADC.

I did not compile the suggested code, but the math is right.
There is also a method to calculate for non-lineair sensors, but that makes the size of the math much bigger.

azsystem, 2021-03-09:
Here you go.
[lineair_calculation.xlsx](https://github.com/letscontrolit/ESPEasy/files/6108779/lineair_calculation.xlsx)

TD-er, 2021-03-09:
There is a typo in your spreadsheet.
You referred to B10, while it must be B8

[image]

azsystem, 2021-03-09:
Also checked, now all good.

closed 2021-03-09 by azsystem