letscontrolit/ESPEasy #3540: Release mega-20210223 on WT32-ETH01 crash / not responsive after a few hours

dx266, 2021-03-10:
Hi,

I am pretty new posting an issue but have been using espeasy successfully for years now. Thanks for the great work behind the project.

I have been trying to use the last two build on a WT32-ETH01 V1.2
The built load sucessfully and both ethernet and wifi are working fine. However after a few hours it crashes. (no ping anymore...)

is there anything known that can be done ?

thanks

TD-er, 2021-03-10:
If it is crashing, what kind of error is reported on the sysinfo page?

dx266, 2021-03-10:
thanks for your input. I will try to give it a try. In the mean time here is the sysinfo: 

Build:⋄
20112  - Mega32
System Libraries:⋄
ESP32 SDK v3.2.3-14-gd3e562907
Git Build:⋄

Plugin Count:⋄
47 [Normal]
Build Time:⋄
Feb 23 2021 12:28:04
Binary Filename:⋄
ESP_Easy_mega_20210223_normal_ESP32_4M316k_ETH
Build Platform:⋄
Linux-5.4.0-54-generic-x86_64-with-glibc2.29
Git HEAD:⋄
mega-20210223_9054eb3
1

TD-er, 2021-03-10:
By the way, is one of the LEDs of the ethernet connector flashing quickly when it is no longer connected?

dx266, 2021-03-10:
yes that is what I noticed as well.

Well I far as I remember the LED was off. I have done 4 tests so far. I will make sure to check next time.

TD-er, 2021-03-10:
I have seen similar behavior in the past with a few nodes where the orange LED started blinking at a seemingly constant rate of like 2 - 4 Hz (not measured, just from memory) and the ethernet was unreachable then.
No idea (yet) how to detect this issue, nor if a re-init of the ethernet device may be a fix for it though.

dx266, 2021-03-11:
hello, it is again unresponsive.  I can confirm the ethernet LEDs are both off. I will try to try your code update.

dx266, 2021-08-19:
Hi Again,
I have been doing several tests but am still having same unresponsiveness issues even with the latest august releases. I figured out that unplugging the ethernet cable while using the WIFI does not change the behavior. However disabling ethernet then using WIFI works fine. This lead me to wonder and check if the chip on the WT32-ETH01 was really a LAN8710. It appears that it is instead a LAN8720A. I am not sure this is changing much but that could be a good reason for the crash. However I am surprised I would be the only one having this issue? I have 3x WT32 ETH01 and all behave similarly. Thanks for the support.

TD-er, 2021-08-19:
I have had some issues with using the LAN interface myself lately, so I would like to ask you to check the following if it happens again.
- Does the ESP mark all elements of a DHCP request? e.g. in my test setup, I noticed it sometimes does not update the DNS server if it receives a new DHCP packet.
- Is the ethernet switching between 10 Mbps and 100 Mbps? This is either an indication of noisy or slightly unstable power supply.

N.B. LAN 7810A and 7820A do not differ that much. Most differences are in the number of pins of the chip and the 7810A does also support an older communication protocol which we don't use as it needs even more pins to the ESP.

dx266, 2021-08-20:
Thanks for the input.

- understand but not that sure how to check /log it. Can you tell me how to
do that?
- Concerning the speed I have not seen mine going at 100Mbps. It remains at
10Mbps although on my router side all is allowed for 100Mbps. Now as far as
the power supply is concerned I was also wondering if that could not be an
issue so I have made one with 3.3V regulator and proper filtering /
decoupling as I was finding the on board regulator maybe a bit hot but
nothing change.

TD-er, 2021-08-20:
Hmm that's strange  that it won't connect to 100 Mbps speed.
I need to have a look at the schematics and maybe also check the board myself.
All other ESP32 + eth boards I have here all work well on 100 Mbps and I only noticed it toggled between 10/100 when using bad wiring for the power supply.
I haven't tried this exact board yet, but I do have it in my drawer.

dx266, 2021-08-21:
Ok please let me know when you will have a chance to check. Also what I
think I noticed is that after flashing the firmware the first crash will
occurs often after one or 2 days ish then after hard reboot it will last
for a couple of hours and then next crashes will be almost immediate.
Could that be a memory issue somewhere?

TD-er, 2021-08-21:
Depends on what you've installed for plugins/controllers
ESPeasy on ESP32 + Ethernet tends to run quite stable in my experience.
Had one running for over 200 days... until a power outage :(

One of those boards I'm using is running a build from: Aug 27 2020 01:06:16

Since you've already got it running (e.g. settings are correct), you could try to flash a build from around that time to check if that's more stable.
If it is, please let me know because then there might have been a change which breaks stability.

One thing that may be an issue on older builds (not sure when I fixed it, could be after  February this year) is that Eth interface could not process an event more than once. (e.g. ethernet disconnect/reconnect)
So if your switch (or ethernet module on the ESP) is turning off an ethernet port, you may be having issues due to this bug.

tonhuisman, 2021-09-26:
This could be retested, now that pr #3775 is merged, fixing dns related issues

dx266, 2021-09-26:
Great thanks I will give it a try and let you know

dx266, 2021-10-11:
Thanks. I have used the latest nighly build. Here are the info

Build:⋄ | 20115 - Mega32
System Libraries:⋄ | ESP32 SDK v4.4-dev-2909-gcdcf92440
Build Time:⋄ | Sep 19 2021 14:14:03

TD-er, 2021-10-11:
That's not the latest nightly build.
I never used SDK 4.4 on a nightly build since that's giving too much issues.
The latest nightly build is from a week ago (not September) and today I merged a commit getting back even one step further in SDK since the latest nightly build has an issue on ESP32 when using a password on the web UI.

dx266, 2021-10-11:
Hum interesting. Then I will wait for your release of tonight and try it
again. Just in case can you please send me the link of where to find it as
I have used the link that was attached to the discussion just  above that
obviously led me to the wrong release or maybe my mistake ?  Thanks

tonhuisman, 2023-09-01:
No response, but is most likely resolved by framework and ESPEasy upgrades, so can be closed.

closed 2023-09-01 by TD-er