letscontrolit/ESPEasy #3551: FHEM: WARNING value name or value is missing

TungstenE2, 2021-03-22:
### Summarize of the problem/feature request

ESP_Easy_mega_20210114_normal_ESP8266_4M1M

Warning in FHEM log:

2021.03.22 13:33:58 2: ESPEasy espBridge_192.168.178.66_61287: WARNING: value name or value is missing (192.168.178.66). Skip processing this value.
2021.03.22 13:33:58 2: ESPEasy espBridge_192.168.178.66_61287: Data: {"module":"ESPEasy","version":"1.04","data":{"ESP":{"name":"ESP_Easy1","unit":0,"version":2,"build":20111,"build_notes":" - Mega","build_git":"","node_type_id":17,"sleep":0,"ip":"192.168.178.66"},"SENSOR":{"0":{"deviceName":"Dummies","valueName":"Day","type":6,"value":"0"},"1":{"deviceName":"Dummies","valueName":"","type":6,"value":"0"},"2":{"deviceName":"Dummies","valueName":"Motor_on","type":6,"value":"0"}}}}

### Expected behavior

no warnings

### Actual behavior

see above

### System configuration

Hardware: Wemos D1 mini

ESP Easy version: ESP_Easy_mega_20210114_normal_ESP8266_4M1M

ESP Easy settings/screenshots: 
[image]

TungstenE2, 2021-03-22:
Does somebody know what causes this warnings in FHEM?

tonhuisman, 2021-03-22:
2 values at Dummy device 'Dummies' don't have a name

closed 2021-03-22 by TungstenE2