letscontrolit/ESPEasy #3554: Can't connect to WIFI using commands

chemmex, 2021-03-23:
Using ESP_Easy_mega_20210223_normal_ESP8266_4M1M_VCC

Following command sequence:
```
>wifissid,Mynet

Wifi SSID:Mynet

Ok
>wifikey,mypassword

Wifi Key:mypassword

Ok
183825 : Info   : WD   : Uptime 3 ConnectFailures 0 FreeMem 21912 WiFiStatus 0 ESPeasy internal wifi status: DISCONNECTED
>save

Ok
>wificonnect

Ok
213831 : Info   : WD   : Uptime 4 ConnectFailures 0 FreeMem 22344 WiFiStatus 0 ESPeasy internal wifi status: DISCONNECTED
```
results in nothing happened.

In previous versions **save** triggered immediate connection to configured network. This applies to both serial and web commands

TD-er, 2021-03-23:
Can you try to call "wifidisconnect" ?

chemmex, 2021-03-23:
Tried. In fact, nothing happened, wifi client remained connected

TD-er, 2021-03-23:
OK, that's for sure not how it is supposed to be.
Any idea what build was working fine?

chemmex, 2021-03-23:
mega-20210114, I beleive. I made several custom builds with it, my setup script worked as usually.

TD-er, 2021-03-25:
Can you test this fix: https://github.com/letscontrolit/ESPEasy/pull/3555 ?

PR #3555 (2021-03-25): [WiFi] Fix unit not reconnecting to WiFi
The initial setup also was next to impossible to do via the web interface as the unit did not attempt to reconnect from the setup page.

Fixes: #3554
closed 2021-03-25 by TD-er