letscontrolit/ESPEasy #3558: Move SCD30 from playground to ESPEasy main repository

TD-er, 2021-03-26:
See [forum request](https://www.letscontrolit.com/forum/viewtopic.php?p=51510#p51510)
Playground P178_SCD30.

fly74, 2021-03-27:
Is this plugin for AZDelivery AZ-Envy Wlan ESP8266-12F Umwelt Entwicklungsboard mit Feuchtigkeits- und Luftqualitätssensor (MQ-2 und SHT30) kompatibel mit Arduino inklusive E-Book! https://www.amazon.de/dp/B08LKVSH2B/ref=cm_sw_r_cp_api_i_G8ECR15WT2554EYGPKGW too?

TD-er, 2021-03-27:
Honestly I have no idea.
There was this request on the forum and I noticed it was essentially a port from a Playground plugin.
That would take more effort than "just create a build for this person" so I made an issue for it to not forget it.
Have not looked at the code at all.

smotek, 2021-04-05:
fly74 no.
A distinction must be made between SCD30 and SHT30.
The board is SHT30, Plugin is SCD30.

PR #3854 (2021-12-12): [P117] Add SCD30 CO2, humidity and temperature sensor
- Add SCD30 CO2, humidity and temperature sensor, migrated from PluginPlayground P178
  - Implement PluginStruct_data for multi-instance support (I2C multiplexer needed if multiple sensors need to go on 1 ESP because of the fixed I2C address `0x61`)
  - Minor changes to WEBFORM_LOAD event
  - Reformatted source using Uncrustify
- Add Frogmore SCD30 library, local copy with a few adjustments:
  - Inhibit I2C bus reset feature (already present in ESPEasy)
  - Fix compilation issues for ESP32 (uint16 was used instead of uint16_t)
  - Reformatt [... cut]
closed 2021-12-12 by TD-er