letscontrolit/ESPEasy #3561: [Rules] Can't save anymore

clumsy-stefan, 2021-03-28:
Not sure if this is an issue particular to my environment, but I just relaized that I can't save rules anymore. If i go to "rules => Rule Set <n>" edit something and then press "save" nothing happens (no pupup and the changes are not saved). Log shows `99737706: Webserver args: 0: 'set' length: 1 1: 'rules' length: 384`.

I can delete the rules file, upload a new one and the changes are accepted, but not via the UI.

Anyone else has this issue?

TD-er, 2021-03-28:
Is this on a node with 1M flash?
Then it is likely the file system is fragmented.

Please also check if you're fetching the JavaScript files (hosted on GitHub CDN servers) if you're running a "testing" or "dev" or "minimal" build.
You can see this on the "inspect" window of your browser.
Either in the console (when JavaScript code is called which is missing) or check the "network" tab when loading the page to see if the .js files are loaded.
N.B. those files served from the CDN servers have a TTL of 1 year, so you may need to try it also in "incognito" mode of your browser, assuming this does not cache those files.

clumsy-stefan, 2021-03-28:
Thanks for the quick answer. This seems to be the correct hint about the JS files. Trying to find out what happens and what I changed that they are obviously not loaded anymore...

I use my own build and it happens on a 4M node as well, so I guess the issue will be the JS missing or similar.

TD-er, 2021-03-28:
"normal" and "custom" builds should have the JS included.
The others link to external CDN urls.
Maybe a browser setting preventing the load of external JS files from a site not using HTTPS?

clumsy-stefan, 2021-03-28:
It did work until some time ago with JS excluded... Probably a define missing somewhere or similar. In the network console I can't see it tryingo to load. Just the JS error.

[image]

TD-er, 2021-03-28:
As a quick work-around, you can try loading a recent nightly build, navigate through all pages and then the files are cached in your browser.

clumsy-stefan, 2021-03-28:
thx.. I tried differnet browsers also..
As it's not relly critical (I can always delete and re-upload the whole file to change it as a workaround) I'll try to find and fix the root cause... I'll report if I find something (which probably could affect also others)...

clumsy-stefan, 2021-03-28:
Hmm... strange, it seems the function is loaded (see screenshot). But there is a JS error `net::ERR_CONTENT_LENGTH_MISMATCH 200 (OK)`. Any hint what this means?

[image]

TD-er, 2021-03-28:
Hmm that sounds like something not going right in saving/loading the rules file.

I do know there have been some issues with respect to CRLF and LF formatting.
The JavaScript does make sure to store it using CRLF.
Maybe editing the file in some other editor and uploading it may have introduced some issues here?
Can you test on an empty rules file, just editing via the web UI to see if it then keeps working?

Do you have some non-ASCII characters (e.g. Unicode) in the rules file?

TD-er, 2021-03-28:
When searching for the error, I also see some threads suggesting the request may be malformed by scanners/firewall etc.
If you have it served by the node itself (you can copy the JS files to SPIFFS on the node) you can exclude external causes.
If it is then still an issue, then your browser or PC may have some scanners or plugins present that try to do something to the JavaScript being served.

clumsy-stefan, 2021-03-28:
No special characters and it also does not work with rules that have not been uploaded and only edited directly on the node.

Tried the following:
1) Deleted the rule.txt from the SPIFFS
2) Added one line on the web-page under "rules"
3) pressed "save"
4) reload the rules again, the line was there (new rules.txt created)
5) added a second line to the same rules on the web page (as in 2)
6) pressed "save"
7) reload the rules again, only the first line was shown.

Interestingly the above error showed as well in 3) as in 6). But in neither of them the "saved" popup came. Even though it obviously wrote it to the SPIFFS in 3)

I don't think it's a JS load error or similar, the files are included in the build (not loaded via JS)... more like overwriting an existing file doesn't work (anymore).

I tried with different browsers and allowed everything on the browser, so I don't think the issue is browser related.

Will also have a look if something related to SPIFFS changed in the esp8266 core. The only breaking thing I found in the last weeks was the BASE64 encoding, but that's nowhere used in the code.

TD-er, 2021-03-28:
Is the error shown when loading the rules or saving?
In other words, what is causing the issue?

Does it also happen on a 'nightly build' (to make sure it isn't something changed in esp8266/Arduino)

clumsy-stefan, 2021-03-28:
Error shows when saving, but independent if it works or not.

I only find the releases (from 20210223), where can I download the nightly?

TD-er, 2021-03-28:
Those are what I meant. I've reduced the number of builds so we get a more steady set of builds used by a lot of people and intermediate test builds.

Just test with the 20210223 build as it is merely a test to see if it is a problem with your browser/system or with recent changes in Arduino code.

clumsy-stefan, 2021-03-28:
Ok, with 20210223 it seems to work fine (at least the rules changing), however, no popup and issues with Wifi (chunked encoding).

So either a change in ESPEasy or the esp8266 core seems to affect it.

Having said this about chunked encoding, could that also be an issue?

[image]

TD-er, 2021-03-28:
What issues about chunked encoding?
We're using that for a long time now (over 2 years) and have not changed anything related to chunked encoding.

clumsy-stefan, 2021-03-28:
See screenshot, the page fails to laod (or takes a very long time) with above error (see screenshot, console lower right):
`GET http://wemos-mini-17.clumsy.ch/rules net::ERR_INVALID_CHUNKED_ENCODING 200 (OK)`

But it's somewaht random and wirks 10times without problems, before it happens. Sometimes it happens a number of times in a row.

TD-er, 2021-03-28:
It just sounds like a local problem on your side.
Are you using a proxy, or maybe have some tool installed that tries to "optimize" stuff?

For example Opera used to route things via some optimizer to reduce traffic.

The URL you posted does have a domainname. Is it just an A-record for a local IP, or do you have things routed via a proxy like Apache/ngix using a proxypass or something similar?

clumsy-stefan, 2021-03-28:
Everything is local and nothing changed in the recent times. I run my own LAN with my own (split-)DNS, DHCP, NTP and other Servers/Services. No Proxies (but quite a stringent FW on the outer side). As I'm a network (and computer) engineer running large networks professionally I'm quite confident, that the network itself runs fine (some 200 IP devices on a local RFC class B subnet, some of them FW'ed and NAT'ed to the outside for public services).

Tried it with Opera, Safari and Chrome. Always direct connection between my Machine and the ESP.

However I agree that the problem is "local" as I seem to be the only one experiencing it.

clumsy-stefan, 2021-03-28:
if you look at the test I did [above](https://github.com/letscontrolit/ESPEasy/issues/3561#issuecomment-808885605). If it's a network issue, why should it work, when there is no rules file on the SPIFFS and not when there is already an existing rule file. Can't currently see a reasoning behind this.

TD-er, 2021-03-28:
OK, just to be sure as I think I make wrong assumptions based on your replies.
As I understood it, you also had chunking errors on the pre-built 20210223 version, but in your last reply you stated it is working with that build.

So first things first; Is chunking an issue in pre-build versions?
I think saving rules is maybe related to these chunking issues as the initial error indicates the nr of bytes received differs from what is expected. Chunking errors are also about what is received and what is expected.

clumsy-stefan, 2021-03-28:
> So first things first; Is chunking an issue in pre-build versions?

It seems (at least I have it on the unit I flashed with the pre-built version now). But it happens when loading the pages.

However in that prebuilt version saving the rules work fine (as it did in my older builds).

The opposite in my new self built I don't see any chunking errors in the build, but can't save the rules (and get a length mismatch error).

So I guess that changes in the core or recent changes in ESPEasy or a combination of both produce this issue. Also compile-enironment (compiler, libraries, etc.) could of course affect it, that's why I agree that this is a (isolated) local issue...

Really strange for me is mainly, that I can create new rule file via Rules WebUI but can't change/save an existing one...

TD-er, 2021-03-28:
> Really strange for me is mainly, that I can create new rule file via Rules WebUI but can't change/save an existing one...

I think it may have something to do with these errors too.
The JavaScript is written with "upload may fail" in mind.
So it fetches the content of the form, encodes it and sends it as a POST request for saving it. (the same call as uploading it)
It does not fetch it again so you won't loose all the just written content when it fails to save. (as did the old version)
The script does fetch the file just the same as when you're using the file browser page and download a file from SPIFFS.

When saving a new version of the file, I guess the upload call may fail for some reason and maybe is aborted by the JavaScript in the browser. Uploading a new file may be handled faster by the ESP, so maybe it will also fail on larger files.

I did try on a few nodes running here and I can't reproduce those chunking errors, but I'm also not running the 'staging' code.

clumsy-stefan, 2021-03-28:
Hmm... The test I did above was on an empty (removed) rules file and added 7chars, after saving it I tried to add again 7 chars, which failed... so you I wouldn't say that's "large"...

On my builds I don't see the chunking errors anymore (prob. all the new WiFi code since the official builds), so only the rules issue. but on your own builds this obviously is not happening. As it happens on different browsers, it would point to changes in the esp core.

TD-er, 2021-03-28:
I will later this evening create a clean build as the more recent builds I tried locally are not based on the 'mega' branch.
So before we draw any conclusions, wait till I tested those.

clumsy-stefan, 2021-03-30:
Could you reproduce it? I tried a number of different things, but can't see what's going on. It just seems that in the new version the JS is not executed or callback is not sent back.

TD-er, 2021-03-31:
Sorry had not tested here.
Will test now.

TD-er, 2021-03-31:
[image]
It does work here, not running the "staging" but just the latest core used in the 'mega' branch.

clumsy-stefan, 2021-03-31:
ok, thanks!... interesting... so I guess it's the latest core commits that break something... just frmo the changes and comments there I can't pinpoint which one t could be...

clumsy-stefan, 2021-04-01:
not sure if commit #656a33e6f82482535782213a6e96c2bd49b22a39 from esp core has to do with it, even though I did a `#define ICACHE_RAM_ATTR IRAM_ATTR` in config.h. Probably it gets defined too late... On the other hand I don't see why this should affect the UI/JS functionality.

TD-er, 2021-04-01:
I don't think https://github.com/esp8266/Arduino/commit/656a33e6f82482535782213a6e96c2bd49b22a39 has anything to do with it.
If it was, then you would see quite a lot of crashes.

clumsy-stefan, 2021-04-05:
I reverted a number of different commits from ESP core without luck... I still don't have any sstarting point where/why it could go wrong :(

clumsy-stefan, 2021-04-13:
What I found is, that calling `http://wemos-mini-2.clumsy.ch/rules1.txt` only downloads the first 64B and restarts downloading again... 

When I try it in a console session:
```wget "http://wemos-mini-2.clumsy.ch/rules1.txt"
--2021-04-13 20:10:53--  http://wemos-mini-2.clumsy.ch/rules1.txt
Resolving wemos-mini-2.clumsy.ch (wemos-mini-2.clumsy.ch)... 10.0.10.102
Connecting to wemos-mini-2.clumsy.ch (wemos-mini-2.clumsy.ch)|10.0.10.102|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 432 [application/octet-stream]
Saving to: ‘rules1.txt.1’

rules1.txt.1                                  14%[============>                                                                                ]      64  --.-KB/s    in 1,9s    

2021-04-13 20:10:55 (33,5 B/s) - Connection closed at byte 64. Retrying.

--2021-04-13 20:10:56--  (try: 2)  http://wemos-mini-2.clumsy.ch/rules1.txt
Connecting to wemos-mini-2.clumsy.ch (wemos-mini-2.clumsy.ch)|10.0.10.102|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 432 [application/octet-stream]
Saving to: ‘rules1.txt.1’

rules1.txt.1                                  14%[============>                                                                                ]      64  --.-KB/s    in 1,9s    

2021-04-13 20:10:58 (33,6 B/s) - Connection closed at byte 64. Retrying.```

Any Idea why?

TD-er, 2021-04-14:
I don't know what is happening at your setup.
But I did have a look at the code and I noticed the function `loadFromFS` does check whether you are logged in.
Do you have a login configured on your nodes?
Can you test to see if it does work with the login cleared?

clumsy-stefan, 2021-04-14:
I checked with and (mostly) without login activated, same results...

It had to be around mid march when the problem occured, as I do regular backup/saves of my configs and rules with GIT and since mid march all files gotten from the nodes are exactly 64Bytes long, security.dat, config.dat and rules[1-4].dat....

When looking what changed in the esp8266 core around that time, I'm wodnering if commit [c720c0d](https://github.com/esp8266/Arduino/commit/c720c0d9e8db2a2911ddd797c0876d675f58ddfb) could be a source of the issue.

I really did a lot of traces, debugs, different versions, but can't find a starting point....

TD-er, 2021-04-14:
Last night I also saw a post on the forum of a new user that also built his own version and showed a screenshot where clearly all CSS was missing.
So I wonder if that's a similar issue here, where files loaded from SPIFFS no longer seem to work in the latest core versions.

clumsy-stefan, 2021-04-14:
> Last night I also saw a post on the forum of a new user that also built his own version and showed a screenshot where clearly all CSS was missing.
> So I wonder if that's a similar issue here, where files loaded from SPIFFS no longer seem to work in the latest core versions.

If the loading of the CSS also only loads the first 64B it would result in such an issue I guess..

TD-er, 2021-04-14:
Could it have something to do with this PR?
https://github.com/letscontrolit/ESPEasy/pull/3576

clumsy-stefan, 2021-04-14:
I can try a build without this PR, but I don'tthink that's the issue as it had to be before mar, 17th.

my commit logs eg. from a config.dat file show the following:
```
commit f5f26299dc1939cd70c87073d1447a5565a9be64 (HEAD -> master)
Date:   Tue Apr 13 15:41:33 2021 +0200
 config.dat | Bin 0 -> 64 bytes
 1 file changed, 0 insertions(+), 0 deletions(-)

commit 473a1bf49497c379692b565c094e358c0855ee32
Date:   Wed Mar 17 11:40:51 2021 +0100
 config.dat | Bin 65536 -> 0 bytes
 1 file changed, 0 insertions(+), 0 deletions(-)

commit 389fd04c986313b50b593606de3e429f0a5a1c0c
Date:   Mon Feb 22 19:51:57 2021 +0100
 config.dat | Bin 65536 -> 65536 bytes
 1 file changed, 0 insertions(+), 0 deletions(-)

commit 8beb1d88f54f2981d99ca5ddf788a636e214f155
Date:   Sat Feb 6 08:47:16 2021 +0100
 config.dat | Bin 65536 -> 65536 bytes
 1 file changed, 0 insertions(+), 0 deletions(-)
````

So there was a change mid march wich resulted in 0Bytes and the next in April in these 64Byte, which is the current state...

TD-er, 2021-04-14:
What is the content of the 64 bytes?
Is it part of the files as how they should be?

clumsy-stefan, 2021-04-14:
Yes, when you do that with a text files (eg. rule files) it's exactly the first 64Bytes of the rules. Did'nt check with a binary file though...

TD-er, 2021-04-14:
I am working on this PR: https://github.com/letscontrolit/ESPEasy/pull/3576
In there I have made some changes in the strictness for checking whether you're logged in.
Maybe you can also test with this PR used in your build?

clumsy-stefan, 2021-04-14:
I quickly did a version with that PR included, unfortunately same issue as before.

Honestly, I don't think it's your code/changes that causes this. I'm quite sure it's due to a change in the esp8266 core which probably need to be adapted then in your code... but as there were so many changes around that time it's quite hard to pinpoint.... and takes quite some time to revert / check after every commit...

However I really appreciate your help and any pointers you can give me... if we find it now, we can already make sure, that once you change to the new core, the problem won't appear....

clumsy-stefan, 2021-04-14:
I think I'm going back to various commits from the core to find which one breaks it... a number of checkout and compiles will hopefully give me more information... If I can pinpoint which core change produces the issue, I'd be happy if you could have a look at it to find out what would need to be changed in ESPEasy to adapt the core changes.

TD-er, 2021-04-14:
... or needs to be reported as an issue with the core. ;)

clumsy-stefan, 2021-04-14:
more likely, yes 😁

clumsy-stefan, 2021-04-14:
quick update: a first try up to and including commit [4cc147](4cc1472821d9f56938dcd5601f3478851f9200ab) makes it ok again... so at least it seems really to be the core... I'll now forward along the commits to find the breaking one...

clumsy-stefan, 2021-04-14:
So, after ~10 builds I found the following:

- Everything works fine up to commit [4cc1472](https://github.com/esp8266/Arduino/commit/4cc1472821d9f56938dcd5601f3478851f9200ab)
- starting from commit [c720c0d](https://github.com/esp8266/Arduino/commit/c720c0d9e8db2a2911ddd797c0876d675f58ddfb) downloads won't start or return 0 bytes
- Commit [1c57b34](https://github.com/esp8266/Arduino/commit/1c57b3408cfdc2956d3c11307b79e2e0055db038) changes it, so the download starts but sends only the first 64 Bytes before the session breaks. Which is like this up to the last commit.

I had a quick look at these commits and code changes, but honestly, that's way above my programming knowledge... so if @TD-er or someone else could have a look at it, it would be great!

TD-er, 2021-04-14:
@clumsy-stefan 
Can you test what is suggested here: https://github.com/esp8266/Arduino/issues/7982#issuecomment-819838025
?

clumsy-stefan, 2021-04-15:
I'll quickly do a test build now and report directly in the core issue, if that's ok with you...

TD-er, 2021-04-15:
Sure, that is the best place to continue I guess

clumsy-stefan, 2021-04-15:
Then I think this one can be closed. It's discussed in the related esp8266 core issue...

Again, thanks for your help though!!

TD-er, 2021-04-15:
You're welcome and good work finding the commit that caused it.

closed 2021-04-15 by TD-er