letscontrolit/ESPEasy #3564: Web admin password not recognised in web login

kuppe234, 2021-03-29:
I use an own build of ESPEasy on ESP32. Etherthing works fine, but I can not login on web site if a password is set. It is also not possible to set another password over serial port.
If I use a "Automated binary releases" from this git, the login is possible.

My build: 
Build:⋄ | 20112  - Mega32
System Libraries:⋄ | ESP32 SDK v3.3.4-432-g7a85334d8
Git Build:⋄ |  
Plugin Count:⋄ | 47 [Normal]
Build Origin: | Self built
Build Time:⋄ | Mar 29 2021 14:41:09
Binary Filename:⋄ | ESP_Easy_mega_20210329_normal_ESP32_4M316k_ETH
Build Platform:⋄ | Linux-5.11.7-100.fc32.x86_64-x86_64-with-glibc2.2.5
Git HEAD:⋄ | mega_2be16c1

Can you help me?

Greetings kuppe234

TD-er, 2021-03-29:
How did you build it?
I assume you built it on Linux using PlatformIO, or else it was not possible to see the "Build Platform" value being set.

In platformio_core_defs.ini you can change `platform                  = espressif32@3.1.0` into `platform                  = espressif32@2.1.0`
That's one of the changes made since the last 'nightly build'.

kuppe234, 2021-03-29:
Hello,
I´m build it on Linux using PlatformIO.
I changed `platform = espressif32@3.1.0` into `platform = espressif32@2.1.0`. Now it´s possible to set the admin password and to login with this password.
Thank you!

TD-er, 2021-03-29:
Thanks for reporting it.
Apparently something has changed in the Espressif32/Arduino framework regarding authorization.

juliokele, 2021-03-29:
tested with platform = espressif32@3.2.0, it seems to be working fine

TD-er, 2021-03-29:
> i tested with `platform = espressif32@3.2.0`, it seems to be working fine

[image]
Well, had not yet tested that version ;)
Glad to know it has already been fixed.

PR #3565 (2021-03-30): [ESP32] Set to PIO espressif32@3.2.0 (#3564)
Fixes: #3564
commit 5ea80b87 (2021-03-30): [ESP32] Set to PIO espressif32@3.2.0 (#3564)

Fixes: #3564
closed 2021-03-30 by TD-er