letscontrolit/ESPEasy #3566: Hi. Could we put the analog plugin to the minimal OTA bin‘s?

fly74, 2021-04-02:


tonhuisman, 2021-04-04:
Adding the ADC plugin (P003) to a build adds ca. 2.7kB to the .bin size, and that defeats the OTA possibility. Only option will be to disable another feature, but most likely you need that as well...
Maybe you can craft a Custom build that has this plugin and other features, and still allows OTA? It will be a challenge to make that fit.

fly74, 2021-04-05:
@tonhuisman I feared that, but can we kick these 2 for the ADC?
[image]

tonhuisman, 2021-04-05:
For a Gosund P1 I'd expect you to need a POW build like ` hard_other_POW_ESP8285_1M` ? As AFAICS, that Gosund has only 1MB flash, and not much possibility for mounting alternative sensors? Not sure if the CSE7766 is actually supported by the POW plugin though.
The ADC plugin won't be able to deliver _any_ power measurement on that device, AFAIK.

fly74, 2021-04-05:
@tonhuisman The thing is not the measurement. The buttons are realized with resistors on adc input. So we only need Switch and ADC for a Test with OTA.

tonhuisman, 2021-04-05:
What is the max. OTA size mentioned next to the Firmware Update button on your device?, With your suggestion of removing Dummy and System info the .bin size still increases with 1354 bytes (622140 total), so I'm not sure it is possible to use OTA there.
I'll dig up an empty device here and flash this build myself to see what I'm getting (it is based on the latest `mega` sources).

fly74, 2021-04-05:
@tonhuisman 
[image]

tonhuisman, 2021-04-05:
Hm, that's going to be problematic, as I'm unable to go that low on the current Sketch size and OTA still won't be possible 😞 
I remember TD-er 'fighting' this battle a couple of weeks/months ago, and he was then able to get it small enough to allow OTA again, but the 'base' ESPEasy code, and probably also the Arduino libraries, obviously have grown since then.

I've been able to bring the .bin size down to 617800 bytes by disabling both the HTTP and ESPEasy P2P Network controllers, but I don't know if you can work without these controllers.

Personally I've sort of given up on trying to get my Sonoff S20's to do OTA, as I have a couple of them with BME280 sensors added, and that is also not in these OTA builds, and isn't going to fit either. Every couple of months I open the S20's and update them with the latest normal build using a serial USB port. Next project is to replace the flash chips with 4MB ones allowing me to do OTA.

fly74, 2021-04-05:
Thank you for your explanation. Serial flash is not an option because the pins are difficult to connect. Soldering is needed directly. So I'm glad to made it once and now is the only option OTA. It's a bad solution in hardware design to make the 3 plug switches with a resistor bridge. 🙄

closed 2021-04-05 by fly74

tonhuisman, 2021-04-05:
> Serial flash is not an option because the pins are difficult to connect.

And if you solder wires on those contacts, leading into a small connector, internally accessible only of course, for connecting the serial to?

With the build I mentioned above I'm still not able to use OTA, so something else has to go too 😞 

[image]

fly74, 2021-04-05:
It's not the maintenance I prefer. Let's put our energy in other esp projects. 🗿😆