letscontrolit/ESPEasy #3567: ESP32 Info page crashes on handle_sysinfo_memory()

tonhuisman, 2021-04-02:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### Summarize of the problem/feature request
While building the [ESP MAX] configurations with the Espressif ESP32 3.2.0 release, the Memory part of the Info page crashes the ESP. I've captured the crash-dump as available from the USB monitoring connection, but disabling possible function calls of that function doesn't get me in pinpointing the culprit.

I'm up to par on the latest `mega` branch source.

I suspect this is someway related to the recent update to the Espressif ESP32 3.2.0 release from 3.1.0, but trying to reverse fails in compiling, I probably need to clear some cache/library folder, but I have no info on how/what to clear.

tonhuisman, 2021-04-02:
If added these `build_flags` parameters:
```
-DBOARD_HAS_PSRAM
-mfix-esp32-psram-cache-issue
```
but this hasn't improved the situation yet.

TD-er, 2021-04-03:
PSRAM is not used yet, but I guess you may need to add some build defines to be able to use it.
Maybe also check the required wrappers on the `ESP.getPsramSize()` and related functions as you can apparently hit such crashes so we're missing some compile time checks.

tonhuisman, 2021-04-03:
I have tried to find what compiler build flags should be added, and I did find this [Espressif page on getPsramSize](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html) but that hasn't revealed any build flags so far. Probably because I've skipped a step in the process 😅 or I missed the relation to all that 'IDF' stuff.

TD-er, 2021-04-03:
Maybe good idea to have a look at https://github.com/cyberman54/ESP32-Paxcounter
That project uses PIO and PSRAM.

tonhuisman, 2021-04-03:
Thanks for the research. Unfortunately it doesn't help me solve this yet.

In the meantime I've found that this is to do with my use of the PlatformIO setting `board=lolin_d32_pro` for the ESP32 MAX configurations, that enables PS-RAM, and already includes the compiler flags I added above. When BOARD_HAS_PSRAM is not set, it probably returns the stub value of 0 from getPsramSize() et al.
For now I've wrapped all PS-RAM related stuff in `#ifdef ESP32_ENABLE_PSRAM ... #endif` so I can use the Info page again 😅 

A new feature request for enabling the use of PS-RAM seems to be the best solution for now. I'll create that and close this one.

closed 2021-04-03 by tonhuisman

TD-er, 2021-04-03:
> it probably returns the stub value of 0 from getPsramSize() et al.

Divide by zero?

tonhuisman, 2021-04-03:
> Divide by zero?

Looks like a Core panic something, I can't recognize the cause, the complete crash log is up, a couple of messages above.