letscontrolit/ESPEasy #3568: [CSS - Device Edit] FormCheckBox and addHtml elements not in the same line

heinemannj, 2021-04-03:
CSS issue related to FormCheckBox and addHtml elements in device edit pages:

```
      addFormSubHeader(F("Measurement settings"));
      addFormCheckBox(F("LEDs"), F("p112_MeasurementWithLEDsOn"), PCONFIG(5));
      addHtml(F(" White, IR and UV On"));
      addFormCheckBox(F("Calibrated Readings"), F("p112_CalibratedReadings"), PCONFIG(6));
      addFormNote(F("Unchecked (Raw Readings): Use only for the adjustment of Device and LED settings"));
```

[image]

[image]

[image]

After changing the display property from block into inline-block the checkbox and the html suffix is displayed in the same line as expected:

[image]

[image]

TD-er, 2021-04-03:
Maybe also check how it is done with the `addUnit` call as that's the only one I can imagine that is used in a similar situation.
Currently discriptions like these are only done via a note as you also have in the screenshot, but I can imagine it may be strange to always have a "Note:" in front of the text.

For this specific use case I would say the text should not be after the checkbox, but in front of it as it is a descriptive label of what the checkbox does.
"LEDs" is more of a category while the checkbox is about "Turn White, IR and UV LEDs on".

heinemannj, 2021-04-03:
This is my actual setup:

[image]

The usage of addHtml feature has simplyfied a lot and has made the output much more readable

TD-er, 2021-04-03:
Still, I think the labels should be more descriptive.
Now it looks like Blue has 2 parameters, a status LED on/off and a current limit.
So why not describe the options in the label?

- Blue Status LED On:
- Blue Status LED Current Limit:
- White LED Current Limit:
- IR LED Current Limit:
- UV LED Current Limit:

- White/IR/UV LEDs On:

Also the [mA] could be placed after the current limit combo box using `addUnit`
Then you don't have them in each option.

heinemannj, 2021-04-03:
The `addUnit` I have already seen - makes really sense to be consistent with other predefined elements.
Your above approach was my first approach - but nearly every description was over two lines - Not really good readable

[image]