letscontrolit/ESPEasy #3573: Feature Request: add %sunset% and %sunrise% to json

TungstenE2, 2021-04-11:
Please add Variables %sunset% and %sunrise% to json

https://espeasy.readthedocs.io/en/latest/Reference/URLs.html?highlight=json#json

http://IP/json

thx a lot

tonhuisman, 2021-04-11:
Any other variables you think are missing from there?

TungstenE2, 2021-04-11:
hm, may be NTP and Location Settings? But these are no varaiables...

TD-er, 2021-04-11:
And `%uptime_ms%` was also requested, while you're at it ;)
N.B. maybe best to output the micros64 / 1000 for that value as that one doesn't wrap after 49.7 days.

tonhuisman, 2021-04-11:
One thing I noticed in the JSON output is that bool values are represented as strings: `"key":"true"` and `"key":"false"`, where JSON actually does support real bools `"key":true` and `"key":false`. I won't change that now, as that would probably cause issues with backward compatibility, and the connection with controllers (or is this a Domoticz-related quirk, perhaps?).

Grovkillen, 2021-04-11:
We might need to add a setting for that, "standard json" vs "backwards
compatible json".

tonhuisman, 2021-04-11:
> We might need to add a setting for that

Good plan, but that'll be a separate PR then 😉

closed 2021-04-11 by TD-er

tonhuisman, 2021-04-11:
> 
> 
> > We might need to add a setting for that
> 
> Good plan, but that'll be a separate PR then 😉

I opened PR #3575 for that 😄

PR #3574 (2021-04-11): [JSON] Add extra variables to JSON output
Add extra variables to JSON output: (`http://<esp-ip>/json`)
- `%sunrise%`
- `%sunset%`
- `%isntp%`
- `%uptime_ms%` (new system variable)
- Location:
  - Timezone Offset
  - Latitude
  - Longitude

Resolves #3573