letscontrolit/ESPEasy #3578: Add support for ESP32S2

fmuntean, 2021-04-15:
Would be nice to start adding support for ESP32S2 boards.
Looks like there is some traction on the platform.IO for the support.
This board is better than ESP8266 and easier to solder than ESP32.

fmuntean, 2021-04-15:
the environment config I am using is:
[env:ESP32S2_dev]
extends                   = esp32_common
lib_ignore                = ${esp32_common.lib_ignore}
lib_deps                  = https://github.com/lbernstone/Tone, MFRC522
platform                 = https://github.com/platformio/platform-espressif32.git
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
board                     = esp32dev
framework              = arduino
board_build.mcu     = esp32s2
board_build.variant = esp32s2

Any pointers of getting the project compiled are appreciated.

TD-er, 2021-04-15:
See my attempt so far...
https://github.com/TD-er/ESPEasy/tree/feature/ESP32_S2_support

fmuntean, 2021-07-26:
I got the firmware to compile and run succesfully on the ESP32-S2-wroom board using the settings below:

; ESP32-S2
[env:custom_ESP32s2_4M316k]
extends                   = esp32_common
platform                  = ${esp32_common.platform}

build_flags               = ${esp32_common.build_flags}   
                            -DPLUGIN_BUILD_CUSTOM
                            -DESP32S2
board                     = esp32s2
extra_scripts             = ${esp32_common.extra_scripts}
                            pre:tools/pio/pre_custom_esp32.py
platform_packages         = framework-arduinoespressif32 @ https://github.com/Jason2866/arduino-esp32/releases/download/esp32-2.0.0-pre/esp32-2.0.0-pre.zip
                            platformio/tool-esptoolpy @ ~1.30000.0

I had to do some code changes and I will work on a pull request to push those over for review.
I saw that some sizes were changed when I compiled and not sure why.

However the moment I include any plugin It stops working and not booting at all.

all I get is a boot loop like this:
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002d56b
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3ffe6100,len:0x4b0
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28e8
SHA-256 comparison failed:
Calculated: 2d04894f4d176b806e7e3761811f63f415223fc9d9980ca46ea6aacd6c079bd2
Expected: 1759f975513717b714ddbd483f4673fa88d9bd97b1991a82b3269740017349d2
Attempting to boot anyway...
entry 0x4004c190
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
....

TD-er, 2021-07-26:
Yep, that's similar to what I noticed in my attempt.
Not sure what happens.
It could be something as simple as trying to read the reset reason of the (not existing) 2nd core.

TD-er, 2021-07-26:
@fmuntean I just checked the mentioned PIO config you posted, but that does seem exactly the same as I used here: https://github.com/letscontrolit/ESPEasy/pull/3579

So what other things did you change compared to that PR?

TD-er, 2021-07-26:
[image]
Have it running too :)

fmuntean, 2021-07-26:
with any plugins ?

TD-er, 2021-07-26:
I did push the changes to the PR for it
Only have to make changes to the ESPEasySerial, but for now you can change those lines addressing `Serial2` into `break;` instead of returning the `Serial2`

Here a [test build ESP32S2](https://www.dropbox.com/s/xsy255qig3bqmyz/ESP_Easy_mega_20210726_custom_ESP32s2_4M316k-factory.zip?dl=0)

fmuntean, 2021-07-26:
yes. Your built firmware works perfect for me. When I compiled I had to do far more changes to the code on my side.
I will refresh the code and try again to build.

TD-er, 2021-07-26:
Yep, I think some structs may now have members starting at 32-bit boundary addresses, so for the NodeStruct that is for sure an issue although it does seem to work.

I do find the free amount of memory to be quite low for an ESP32, but maybe it is caused by the new SSL libraries?

TD-er, 2021-07-26:
Hmm looks like `String` is now using 4 bytes more?
Very strange.

fmuntean, 2021-07-26:
I saw the same for the string and the 4 extra bytes not sure what exactly changed.
Any reason why the CONFIG_LWIP_GARP_TMR_INTERVAL=30 instead of the default 60 ?
It does trigger lots of warnings.

TD-er, 2021-07-26:
OK, that makes sense.
I have also been looking into this redefine issue, but could not yet find where it was defined.

I am now adding the new pin definitions for the S2 as it does have a lot more GPIO pins, more touch pins and only 1 pin that's input-only.
It also has a lot less crytical pins.

fmuntean, 2021-07-26:
It is in the hwconfig for esp32 idf4.4

TD-er, 2021-07-26:
Yep, but it seems it is also included from the same file in `esp32/include/config/sdkconfig.h`
Still trying to find where I may include an ESP32 file instead of esp32s2

fmuntean, 2021-07-27:
based on this response (https://github.com/espressif/arduino-esp32/issues/5461#issuecomment-887591270) it seems that CONFIG_LWIP_GARP_TMR_INTERVAL should not be changed

TD-er, 2021-07-28:
As far as I know we're not changing it.
This warning generated during compile time is typically an indication the `#define` call for it is called more than once.
But so far I was not able to find where I might include an esp32 specific header file where there is an esp32s2 version.

TD-er, 2021-07-28:
Ah found it, I had it defined in the platformio_esp32_envs.ini file.

tonhuisman, 2021-09-25:
This has been resolved by the merge of PR #3579

closed 2021-09-25 by TD-er

fmuntean, 2021-09-25:
Thank you all for making this a reality.

TD-er, 2021-09-25:
Please keep in mind that it has been tested only by me (as far as I know) on 2 nodes.
So please let me know if there are some issues with it.

tonhuisman, 2021-09-25:
I have an S2 on order, should arrive within a few weeks (from my friends at Ali)

fmuntean, 2021-10-01:
got the latest and getting a linker error:
.platformio/packages/toolchain-xtensa32s2/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: final link failed: bad value
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\custom_ESP32s2_4M316k\custom_ESP32s2_4M316k_2021_10_01.elf] Error 1

tonhuisman, 2021-10-01:
Have you tried a clean & build on that PIO environment? I've seen PIO builds fail often after switching from another env. build. A rebuild usually fixes it. And updating PIO may also be of help.

fmuntean, 2021-10-01:
Yes. Have done that. Also added the debug flags and I do not get more info either.
I use VS code PIO Core v5.2.0 Home 3.3.4

TD-er, 2021-10-01:
What platform IO config do you use?
You really should use what I put in our PIO config and like Ton already suggested you may have to thoroughly clean your PIO caches:
[image]

fmuntean, 2021-10-01:
I am using the one provided: 
[env:custom_ESP32s2_4M316k]
extends                   = esp32_common
build_flags               = ${esp32_common.build_flags} 
                            -DPLUGIN_BUILD_CUSTOM
                            -DESP32S2
                            -DBOARD_HAS_PSRAM
                            -DESP32_ENABLE_PSRAM
                            -mfix-esp32-psram-cache-issue
board                     = esp32-s2-saola-1
extra_scripts             = ${esp32_common.extra_scripts}
                            pre:tools/pio/pre_custom_esp32.py
platform                  = ${core_esp32_3_3_2_esp32s2.platform}
platform_packages         = ${core_esp32_3_3_2_esp32s2.platform_packages}

fmuntean, 2021-10-01:
after I deleted the two folders: packages and platforms the build was succesfull.
I have few boards with ESP32S2 that I will be using for testing this build.
Thank you for the hard work.