letscontrolit/ESPEasy #3584: Ser2net error használata /client connect, client disconect led kijelzésre ,Pl. gpio,2

kervill68, 2021-04-17:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

kervill68, 2021-04-17:
Szeretném a p20(serial server) használni a gpio,2 output amennyiben van "ser2net:  Client connected"

TD-er, 2021-04-17:
Can you please formulate it in English?

kervill68, 2021-04-18:
Hi, 
Thanks for your answer, and for your help. So, I would like to solve, that serial server (p20) would use GPIO2 output if "ser2net" is activated, as rules. 
I am waiting for your answer, Many thanks for your supporting.
Have a nice day!
Laszlo

TD-er, 2021-04-18:
You mean GPIO-2 as TX for the serial interface?

Right now the serial server (P020_Ser2Net) does not (yet) use the same interface to configure a serial port as (almost) all other plugins do that support a serial port.
So as of now it is not possible to configure SW-serial for this plugin.

Or are you talking about the "TX enable pin" option of that plugin?

uzi18, 2021-04-18:
Maybe just new event is needed - "client connected"

kervill68, 2021-04-18:
A p20 ban van válasz a client kapcsolodásra. Azt szeretném kijelezni, egy output ként. Lássam mikor kacsolódott a cliensre.

uzi18, 2021-04-18:
@TD-er just look at #3587

kervill68, 2021-04-19:
> @ TD-er csak nézd meg a [# 3587-et](https://github.com/letscontrolit/ESPEasy/pull/3587)
Thank you.
Only ​rules is possible. esp12E module is programmed with espeasy. I see this is not a solution for now. I can't program it any other way. I just started programming this a few weeks ago. I checked the program of the p20, I found that the connection is indicated, but I can't get the indication from there

TD-er, 2021-04-20:
@kervill68 See also this PR: https://github.com/letscontrolit/ESPEasy/pull/3483
Here are some test builds: https://github.com/letscontrolit/ESPEasy/actions/runs/766736216
This includes @uzi18 's PR and should be able to select any serial port

uzi18, 2021-04-27:
@TD-er silly question how to download these new test builds ? ;)

TD-er, 2021-04-27:
@uzi18 Maybe also have a look at my latest PR I just made (not sure if it fixes the issues I'm seeing though, still have to get the tests)

The link to the GH actions has a link to the binaries at the bottom. (which I cannot link to directly)
Although it looks like all "testing" builds no longer seem to fit, but @tonhuisman  is working hard on splitting those into plugin sets that hopefully make sense :)
The code was merged, so you could also build yourself if you need to test it.

uzi18, 2021-04-27:
ok now get it, always can build needed one with gitpod ... as usual and download bin from there ;)

uzi18, 2021-04-27:
@TD-er 300MB bin file ;)

TD-er, 2021-04-27:
Yep I've been busy lately ;)

tonhuisman, 2023-09-02:
This seems to have been resolved, so can be closed.

closed 2023-09-02 by TD-er