letscontrolit/ESPEasy #3586: pn532 RFID Reader - wrong instructions / fault in Code?

kr0815, 2021-04-18:
I connected a RFID Reader according to these Instructions:

[https://www.letscontrolit.com/wiki/index.php/PN532](url)

The reader randomly freezes

I measure on the RSTPDN Line only about 2.2V

Here i found the circuit diagram for the reader:
[http://www.elechouse.com/elechouse/images/product/PN532_module_V3/PN532_shematic_drowing.pdf](url)

we have a voltage divider on the board, that´s why RSTPDN is that low

In the code i found:
`if (resetPin != -1)
  {
    String log = F("PN532: Reset on pin: ");
    log += resetPin;
    addLog(LOG_LEVEL_INFO, log);
    pinMode(resetPin, OUTPUT);
    digitalWrite(resetPin, LOW);
    delay(100);
    digitalWrite(resetPin, HIGH);
    pinMode(resetPin, INPUT_PULLUP);
    delay(10);
  }`

wouldn´t it be better to pull the Reset Pin to High so we have 3.3V on RSTPDN?  

Or would it be better to run the Reader with 5V
I also see in the schamitic there are no Pull-Ups on the I2C Lines

TD-er, 2021-04-18:
I assume you're referring to this voltage divider as mentioned in the schematic?
[image]

That 10k/20k ratio is typical for 5V to 3v3.
So if you have 5V, you could power the 2 pins on the board that are marked for 5V as such. (I only saw 2, both present in my screenshot)
I would not advice to put 5V on the pins marked for 3v3.

If you don't have any other I2C sensor present, then you could add pull-up resistors if it still isn't working.

I do see this part of the schematic you posted:
[image]

Not sure how to interpret it, as I always have to search for how to use those FET transistors, but it looks like a pull-up at first glance.

kr0815, 2021-04-18:
In generall my set-up is working - only some freezes from time to time, that´s why i went into detail

On the whole board, both available  Vcc Pins only go to Pin40 of the IC , no chance to feed 3.3V directly
(The board connections are shown on the right side of the PDF)

The 3.3V are internally generatet by the NXP Chip

NXP says - Pin 40 - VBat - Battery Power in - 2,7 - 5,5V

Because of the 10k / 20k divider i see only 3 possibilities-

- change the ESPEasy Manual - this Board needs 5V
- change the code to have permanently 3.3V on the ESP  Pin that feeds the RSTPDN line
- remove R9 /20K

Any ideas what would be the best way?

TD-er, 2021-04-18:
If you have it powered by 3v3, then removing R9 makes sense.

tonhuisman, 2023-09-01:
Don't think this is still relevant, so probably can be closed.

closed 2023-09-01 by TD-er