letscontrolit/ESPEasy #3602: P012 LCD2004 plug-in only partially works

elektron-bbs, 2021-04-24:
### Summarize of the problem/feature request

I downloaded the latest sources today and compile it with build normal and custom.
The P012 LCD2004 plug-in only partially works. Texts directly in the plugin are displayed. Texts that are transferred using rules or via HTTP are no longer displayed. 

### System configuration

Hardware: ESP32-PICO-KIT_V4.1

ESP Easy version: nighty

### Rules or log data

```
on Clock#Time=All,**:** do // will run once every minute
 if [DummyWarn#Gefrierschrank]=0 and [DummyWarn#Kuehlschrank]=0 and [DummyWarn#Dummy_3]=0 and [DummyWarn#Dummy_4]=0 
  GPIO,33,0 // LED rot aus
  if %sysyear%>2020
   LCD,1,1,"%sysday_0%.%sysmonth_0%.%sysyear% %systm_hm%"
  endif
 endif
 if [DummyInfo#Briefkasten]=0 and [DummyInfo#Gewitter]=0 and [DummyInfo#FritzBox7590_AB]=0 and [DummyInfo#FritzBox7590_Call]=0 
  GPIO,32,0 // LED gelb aus
  if [Dummy#Display]=1
   LCD,2,1,"%ssid%"
  elseif [Dummy#Display]=2
   lcd,2,1,'WiFi-Channel: %wi_ch%'
  elseif [Dummy#Display]=3
   LCD,2,1,"RSSI: %rssi% dB"
  elseif [Dummy#Display]=4
   LCD,2,1,"Load: [SystemInfo#Load] %"
  elseif [Dummy#Display]=5
   LCD,2,1,"Helligkeit: [ADC1_0#brightness]"
  elseif [Dummy#Display]=6
   LCD,2,1,"Spannung 1: [ADC1_1#voltage]"
  elseif [Dummy#Display]=7
   LCD,2,1,"Spannung 2: [ADC1_2#voltage]"
   TaskValueSet,11,2,0
  endif
  TaskValueSet,11,2,[Dummy#Display]+1
 endif
endon
```
Log:
```
85429: EVENT: Clock#Time=Sat,15:24
85533: ACT : GPIO,33,0
85534: GPIO : port#33: set to 0
85547: ACT : LCD,1,1,'24.04.2021 15:24'

```

TD-er, 2021-04-24:
> Release mega-20210223 works.

As stated here: https://github.com/letscontrolit/ESPEasy/pull/3483#issuecomment-826098443

TD-er, 2021-04-24:
@tonhuisman Do you have any idea about recent changes in command handling that may cause this?

tonhuisman, 2021-04-24:
> @tonhuisman Do you have any idea about recent changes in command handling that may cause this?

Hm, nothing that comes to mind, didn't touch that plugin, AFAICS, I'll have a look at the source.

That long `IF` has quite a few conditions, are the values all actually 0? And I don't think the rules processing of `IF` has changed the max. number of conditions (even though the docs used to say 3). As the reset of GPIO 32 isn't in the log.

TD-er, 2021-04-24:
The "LCD" command is showing up in the logs as a command to be executed.
Also since the mentioned plugin was not changed and the last nightly build was working, I guess it is probably something with command handling.
As soon as I have put Lilly to bed, I will also have a look at it.

elektron-bbs, 2021-04-24:
Maybe that will help you.
I couldn't find a way to copy this as text, so a screenshot:

[image]

TD-er, 2021-04-24:
Nope, those are intellisense warnings/errors. But you can compile it, so those are not the cause here.

TD-er, 2021-04-24:
What plugins do you have in your build?

TD-er, 2021-04-24:
I did test with a similar line in the rules for the OLED display.
And that one is working, so I guess it may have something to do with your specific setup, like a badly behaving plugin on your node.
The command handling itself from rules and HTTP is working.

elektron-bbs, 2021-04-24:
I only included a few plugins. I also upload the two files of my configuration. The Custom_rel_20210223.h is from the working build release mega-20210223 and the Custom_nightly_20210423.h from the no longer working today. I had to rename the files because Github doesn't accept * .h.

[Custom_rel_20210223.txt](https://github.com/letscontrolit/ESPEasy/files/6370459/Custom_rel_20210223.txt)
[Custom_nightly_20210423.txt](https://github.com/letscontrolit/ESPEasy/files/6370460/Custom_nightly_20210423.txt)

TD-er, 2021-04-24:
Can you show a screenshot of the devices tab in ESPEasy?

elektron-bbs, 2021-04-24:
of course:
[image]

TD-er, 2021-04-24:
Ah found the culprit....
```c++
 case PLUGIN_WRITE:
    {
      String command  = parseString(string, 1);
      P020_Task *task = static_cast<P020_Task *>(getPluginTaskData(event->TaskIndex));

      if (nullptr == task) { 
        break; 
      }
      success = true;

      if (command == F("serialsend")) {
        const char *tmpBuf = string.substring(11).c_str();
        task->ser2netSerial->write(tmpBuf);
        task->ser2netSerial->flush();
      }

      if ((command == F("ser2netclientsend")) && (task->hasClientConnected())) {
        task->ser2netClient.print(string.substring(18));
        task->ser2netClient.flush();
      }
      break;
    }
```

TD-er, 2021-04-24:
Fixed in https://github.com/letscontrolit/ESPEasy/commit/b4253d6922d66bda40567affa028b8c3fe479d7a

TD-er, 2021-04-24:
@elektron-bbs Can you let us know if it is now working?
I will close this issue as  I am convinced it is now fixed.

closed 2021-04-24 by TD-er

elektron-bbs, 2021-04-25:
Yes, it works again.

TD-er, 2021-04-25:
Great :)