letscontrolit/ESPEasy #3604: Move AHT10 from playground to main repository

TD-er, 2021-04-25:
See https://github.com/letscontrolit/ESPEasyPluginPlayground/pull/176

It should be documented that this sensor cannot (!!!!!) be used along with any other I2C device.
Not tested if it may work along with a multiplexer, but I wonder if it will work as it does trigger on any byte sent on the I2C bus resembling its I2C address.
So if switching the multiplexer does need that specific byte, it will not work.

Wookbert, 2021-04-25:
[SHT30 or SHT31](https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20210425034355&SearchText=sht30) instead?

TD-er, 2021-04-25:
Well I do prefer units that do not claim a shared bus for themselves, so if you can please do use those.
SHT3x of BME280 are both fine sensors.

Wookbert, 2021-04-25:
So if I want to design a Motion-Detection-Lux-Temperature-Humidity-Multi-Sensor with

- Wemos D1 Mini (ESP8266)
- [HFS-DC06 Microwave Sensor](https://www.aliexpress.com/item/4000092663070.html)
- [GY-302 Illumination Sensor](https://www.aliexpress.com/item/1005001586029026.html)

which Temperature/Humidity Sensor can/should I pick (with the near future of ESPEasy releases in mind)? As smaller, as more accurate, as better.

TD-er, 2021-04-25:
The light sensor you linked is an I2C sensor, so given you want to use other I2C devices (as would anyone else) you should NOT use AHT10.
To measure temp/hum you can pick BME280 or SHT3x.
DHT series I would not advice to use as they are not really reliable and also quite large.

tonhuisman, 2021-04-25:
And the BME280 adds barometric pressure over the SHT, DHT and BMP sensors.

For any temperature sensor you might need to take care of correctly locating the sensor, as it is quite easy to have the temperature measurement influenced by close-by heat sources. Example: I have a couple of Sonoff S20's where I added a BME280, after creating ventilation holes in the housing, but still I have to subtract ~6 degrees from the temperature to get close to the actual temperature, and probably the humidity reading is also influenced by that off-beat temperature. For me that's not an issue, but others may need more accurate values.

TD-er, 2021-04-25:
@tonhuisman When setting the temperature offset in the BME280 plugin, the humidity will also be corrected for it :)

Wookbert, 2021-04-25:
So, just to be safe: I can order a couple of the BME280s and be sure that I can use them parallel to the GY-302 and my mentioned HFS-DC06 using ... an existing or a upcoming version of ESPEasy?

TD-er, 2021-04-25:
BME280 only supports 2 I2C addresses, so if you want to use more of them on the same ESP, you may need to use an I2C multiplexer chip.
Or you use BME280's and SHT3x units on the same I2C bus.
But I doubt you'll need humidity on all of them, so you could also look for Dallas sensors if you need a lot of temperature sensors as those can be chained on the same single GPIO pin (1-wire bus for the Dallas sensors) and can also run on a longer cable than with I2C.
Typical I2C busses should not exceed roughly 40 cm (not too critical, just a rule of thumb).

So lots of options to choose from.

Wookbert, 2021-04-25:
I'm aiming only for one of each (BME280, GY-302, HFS-DC06). However just saw that there is a difference between BMP280 (Pressure + Temperature) and BME280 (Pressure + Temperature + Humidity), but if you [search for BME280 on AliExpress](https://www.aliexpress.com/wholesale?trafficChannel=main&d=y&CatId=0&SearchText=bme280&ltype=wholesale&SortType=price_asc&groupsort=1&page=1), you'll get mostly BMP280. Actually the purple PCB is identical for both sensors.

TD-er, 2021-04-25:
Make sure you get the 3v3 boards.
There are also boards being sold meant for 5V.

Wookbert, 2021-04-25:
Voltage: Exactly what I'm just looking at. If I power the Wemos D1 mini with 5V from the HLK-PM01, I can „steal“ the 3.3V from the Wemos 3V3 Pin because of the ME6211 on the Wemos, right?

[image]

TD-er, 2021-04-25:
Well regarding "Wemos D1 mini" I will not give you a definite "yes" or "no", since there have been no real Wemos boards for over 2 years already.
So everything sold now is not the same as the original boards.
Some even have such bad voltage regulators that those cannot handle over 150 mA of current, so even the ESP itself may see some brown-out when calibrating WiFi at boot.
The BME280 doesn't use a lot, so that will probably be fine, but if you try to design your own board using a HiLink mains adapter, please do yourself a favor and add some AMS1117 on the board.
If you have a "Wemos" board with a voltage regulator in a SOT25 footprint, you better watch out.
See also [this Reddit thread](https://www.reddit.com/r/esp8266/comments/9iizx4/warning_clone_wemos_d1_minis_with_only_150ma_33v/)

Wookbert, 2021-04-26:
These ESP8266-EX boards are the ones I bought lately (better Wifi performance with the chip antenna compared to the printed antenna). And as we can see, it looks like it indeed has a voltage regulator in SOP-25. Will take a closer look on the marking when back home.

[image]

On AMS1117: So, you suggest that if I equip [my board](https://www.letscontrolit.com/forum/viewtopic.php?p=51838&sid=ac84d33bffd4ee73165214bfe34d5f6b#p51838) with a 5V Hilink HLK-PM01, that I should use my own VR (AMS1117) to create 3.3V in addition to the 5V, right? Why not simply stick to 5V sensors: The GY-302 is 3-5V, the HFS-DC06 is 5V ... I could also get the BME280 in 5V instead of 3V3!?

(Sorry, for drifting away).

Wookbert, 2021-04-26:
For the record: The 5V BME280 has 4 pin holes (and a 6x2K transistor), while 3V3 BME280 has five.

[image]
[image]

TD-er, 2021-04-26:
Some modules have SDA/SCL pulled up to their supply voltage.
The ESP8266 does seem to be 5V-tolerant according to several sources, but most other I2C devices on the same bus may not be 5V tolerant.
And using an external AMS1117 is not for the BME280, as that one doesn't take a lot of current.
The ESP does and low power voltage regulators may lead to all kinds of strange errors.

Wookbert, 2021-04-26:
Best offer for the [BME280-3V on AliExpress](https://www.aliexpress.com/item/1005002537216150.html). USD 0.71. I've contacted them and asked twice for verification that it is indeed a BME, not a BMP.

TD-er, 2021-04-26:
For such prices I think you're not getting what you're hoping for.

There are boards sold as BME280 in large quantities which have a BMP280 and also soldered 180 degree rotated, so they won't work.
The prices of BME280 have gone up quite a bit lately and I have never been able to pay anything that low.
The regular price was roughly 1,80 euro a piece and now it is more like doubled.

ghtester, 2021-04-26:
I did not understand the reason why AHT10 can't work together with other i2c devices at ESP Easy but at least on RPi 4 it works at the same i2c bus together with LCD 1602 connected through i2c adapter (a display operated by LCDPROC plugin and AHT10 at intervals by Python script).

Wookbert, 2021-04-26:
> For such prices I think you're not getting what you're hoping for.

Again I've asked twice, whether it's the BME or BMP:

[image]

Perhaps they haven't updated the prices yet to the chip shortage and are selling old stock.

Anyhow, I've ordered 6 pcs. and will report.

TD-er, 2021-04-26:
> I did not understand the reason why AHT10 can't work together with other i2c devices on ESP Easy but at least on RPi 4 it works at the same i2c bus together with LCD 1602 connected through i2c adapter (a display operated by LCDPROC plugin and AHT10 at intervals by Python script). So far I did not encounter any issue.... So why it's impossible on ESP Easy please?

There are lots of reports to be found online about this sensor.
[this explanation](https://www.raspberrypi.org/forums/viewtopic.php?t=265734#p1624629) was one of the most elaborate ones I could find describing _why_ it does not work well along with others.
So I guess as soon as you will write that specific byte to your LCD it will also cause issues, given it is connected to the same I2C bus.

TD-er, 2021-04-26:
> Perhaps they haven't updated the prices yet to the chip shortage and are selling old stock.
> 
> Anyhow, I've ordered 6 pcs. and will report.

Well keep us posted.
I expect it will be not what you expect.
A friend of mine tried to order larger batches of BME280 boards that were even double that price and all of those orders were cancelled by the sellers.
Larger orders of defective or bad units are more likely to be disputed.

Let's hope I'm wrong, but for 6*60ct you can try of course. :)

ghtester, 2021-04-26:
OK, thanks for the details. So far I can't confirm any issue with the described combination on the same i2c bus (1) on RPi 4B.
 #i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- 38 -- -- -- -- -- -- 3f
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

The LCD is in VERY active use (24 by 7) and now I put the AHT10 reading in endless loop so I'll see...  So far so good...
Perhaps there're chip fake clones with that issue?

TD-er, 2021-04-26:
> The LCD is in VERY active use (24 by 7) and now I put the AHT10 reading in endless loop so I'll see... So far so good...
> Perhaps there're chip fake clones with that issue?

Well given that it's a Chinese manufacturer in the first place and they also mention it in their database, I guess it might be a new revision of the chip.

Wookbert, 2021-04-28:
> Anyhow, I've ordered 6 pcs. [BME280 advertised for USD 0.76/each] and will report.

Surprise, surprise ... prior to shipping the seller checked again, and suddenly it’s a BMP280. As mentioned above: BME280 shortage and skyrocketing prices [also see here](https://forum.sensor.community/t/bme280-global-shortage/413). (Interesting as well: [BME280 under the microscope](https://forum.sensor.community/t/bme-bmp-280-bme680-sensors-under-microscope/520)).

Guess I'll order one expensive BME280 for designing/testing, and wait for the price to come back to normal again.

Wookbert, 2021-04-28:
Question: If I understand the [BME280 datasheet](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bme280-ds002.pdf) correctly, the sensor requires an output calibration/compensation (see section 4.2) and an initial calibration if the chip has exceeded its normal operating range, e.g. because of reflow soldering (section 7.9).

Is this or will this be taken care of in the ESPEasy implementation? Experiences or hints how to?

TD-er, 2021-04-28:
I never changed anything on the BME's.
The measured pressure is always spot on.
The temperature may be off sometimes, but often this is due to improper mounting inside an enclosure where it may heat up.
I do have added a temperature compensation in that plugin, to apply some temp offset, which on BME280 also corrects the humidity accordingly.

This temperature offset can only be used on a narrow temp range as it assumes a constant heat surrounding the unit (e.g. an ESP running in the same enclosure).

Wookbert, 2021-04-28:
**Mounting inside enclosure vs. temperature compensation:** Critical IMHO. I mean if a PSU is embedded in that enclosure, the heat generated by that power supply may vary on the load, depending on what’s connected and how busy the attachments (sensors, motors, you name it ...) are.

Either one has to separate the sensor from the heat source or has to shield it from radiating heat (the latter is what I have in mind ... thinking of a double walled cage, so that the sensor is facing entirely away from the rest).

timof123, 2021-07-06:
Is it possible to use AHT10 on Espeasy? Maybe something new has appeared?

TD-er, 2021-07-06:
Hmm apparently it was not yet moved from the playground to the main repository

timof123, 2021-07-06:
There are no options for using AHT10 with espeasy at all, do I understand correctly?

TD-er, 2021-07-06:
Not at this moment, unless you would like to compile it yourself from the playground
See: https://github.com/letscontrolit/ESPEasyPluginPlayground/pull/176

tonhuisman, 2021-07-30:
I've created a PR to include (migrate from Playground) this plugin in ESPEasy.

Any takers for testing? (as I don't own such sensor)

PR #3724 (2021-09-05): [P105] Move AHT10/20/21 plugin from Playground
Move plugin for AHT10/20/21 I2C environment sensor from the Playground.

- Some adjustments to the code, reformatted using Uncrustify
- Re-used Plugin ID P105
- Documentation

TODO:
- [x] Complete documentation
- [ ] Testing (to be reported)
  - [x] AHT10 (confirmed by ghtester, and dragonfly69)
  - [ ] AHT20
  - [x] AHT21

Resolves #3604 
Resolves #3466
Resolves #3436 
Resolves #3550
closed 2021-09-05 by TD-er