letscontrolit/ESPEasy #3606: NTP cause periodic reset/reboot

toast1234, 2021-04-26:
Hi
Maybe this is known already. Just want to report in case it gives you a lead in troubleshooting some issues. So you can just read and close :-)
I use my router as NTP server.
Configure ip address for ntp in espeasy.
After some router configuration changes my router was lacking a good ntp source. It was still possible for nodes to get/set time from the router but it also generated an error. Probably something like unknown offset. This resulted in reboot of nodes every ~20 min.
After fixing my router ntp server referens settings the issue went away.

TD-er, 2021-04-26:
I do know UDP communication is flaky sometimes.
For TCP connections, you can set a timeout, but for UDP you can't.

Do you know what was happening when your router was not configured correctly?
Was it returning an invalid packet? Or not replying at all?

toast1234, 2021-04-26:
I know that the router itself as client did not report any valid peer. I hava a nagios check that monitors the router and it reported offset value unknown. But as the esp nodes came up with setting correct time i assume that the initial set time worked. I think esp node reported some ntp error in log.
It can maybe be recreated by removing sync source from ntp server and restart the ntp server.

TD-er, 2021-04-26:
ESPEasy does keep track of the last known time and stores it in RTC memory.
At boot it does restore it, so you will have some idea of time when the node reboots/crashes and cannot reach a NTP server.
Depending on the reason of the reboot, this may introduce some drift, but it is better that no idea of time. (e.g. a watchdog timer reboot may fire after about 6 seconds, so each reboot caused by WDT may cause a drift of > 6 seconds)

NTP may give a number of messages in the log, like the 'wander' in msec/sec which can be useful to detect really bad crystals in the ESP board.
Would be helpful if you have an example of such an error.

toast1234, 2021-04-26:
I will see if I can recreate the issue later.

Maybe last reboot cause is still valid:

Unit Number: | 11
-- | --
Local Time: | 2021-04-26 19:00:38
Uptime: | 2 days 0 hours 45 minutes
Load: | 4.66% (LC=466)
CPU Eco Mode: | true
Boot: | Manual reboot (20)
Reset Reason: | Exception
Last Action before Reboot: | 1 timer, id: 0
SW WD count: | 0

The error message repeating in the log might have been something like (but I'm not sure):
NTP : NTP host (192.168.5.1) unsynchronized

Sorry if debug is not that helpful. If I recreate the issue again  I will take better logs and notes.

TD-er, 2021-04-26:
Hmm an exception could also indicate that we're trying to read where no byte was allocated.

tonhuisman, 2023-09-02:
Not sure if this is still relevant. If not it should best be closed.

closed 2023-09-02 by TD-er