letscontrolit/ESPEasy #3609: MQTT - missing ACK from ESPEasy

kr0815, 2021-04-27:
I try to run ESPEasy / MQTT in a bigger company network - Port 1883 is open and all such things

we get a continous MQTT connected - disconnected....

Admin says it´s due to a problem in the TCP Stack of ESP-Easy - maybe even esp8266wifi.h -

Company has a Firewall that does statefull Paket inspection,  Admin says TCP Pakets are inconsistent, that´s why the firewall blocks them

MQTT Broker sends a Connect Ack - wants a response from ESPEasy, which is not send

I,m not a TCP expert - but did some investigations - see attached  screenshots:

[image]
[image]
[image]

Both - ESPEasy and MQTTLens under Chrome:

Broker --> Client: Connect Ack 
Client --> Broker:  Subscribe Request

But in the Android APP:

Broker --> Client: Connect Ack
**Client --> Broker: ACK** 
Client --> Broker:  Subscribe Request

Any suggestions what could be done or where the problem comes from?

TD-er, 2021-04-27:
You're the first to mention this.
I know Tasmota also uses the ESPEasy version of the PubSub MQTT client library, so that should be the same and thus it is somewhat plausible it may even be an issue in the LWIP stack itself.

kr0815, 2021-04-27:
I´m always the one that gets in such unusual problems :-)

Big question - is this ACK really needed - is it mandatory in the MQTT / TCP protocol?
Or is it a fault in the Firewall?

TD-er, 2021-04-27:
By asking a question with both opposites in it, you make my work a lot simpler :)
The answer is "yes" :)

Honestly I have no clue yet if it is mandatory, but given it has never been reported in the past few years I am slightly more prone to blame the firewall rules.
Not saying the rules are interpreting the specifications incorrect, but more likely that they are too strict.

But to be sure, it must be investigated, although right now I don't have a clue where to start looking. (may also be caused by the beverages consumed at "Kings day" here in the Netherlands, even though I never left my street....)
So better have a look at it when my head is fresh again.

kr0815, 2021-04-27:
Drink another one - i´m doing also :-) 

But it´s a really strange problem - let´s see what the experts say :-)

uzi18, 2021-04-27:
If server needs some data, so could you reveal internals of packet sent by server with ACK?

kr0815, 2021-04-27:
sorry, i didnt´t really get what you need - but will do everything to solve this problem

Do you need more detailed logs?

uzi18, 2021-04-27:
please save/export packets file from wireshark and attach it here
curious why all connections are with same IPs where you say they are from different apps, esp easy should have different ip

kr0815, 2021-04-27:
they are with same IP because i made an Hotspot on my Laptop

First try was to log on my Ubiquity Access Points - but they were to slow , lost messages

But in my screenshots .- i always had only client attached

uzi18, 2021-04-27:
@kr0815 ok now understand, please provide file to import localy into wireshark to investigate :)

TD-er, 2021-04-28:
Just some links that may be of any help in looking into this...
[MQTT protocol messages overview](http://www.steves-internet-guide.com/mqtt-protocol-messages-overview/)
At roughly 1/3rd of the page is a section called "Wireshark Network Analysis".
Maybe that's useful here?

kr0815, 2021-04-28:
Istudied the link you postet:

About in the middle, they say "The MQTT Connect ACK response gets a TCP ACK response."

That´s in my understanding exactly what the Android App does, but neither ESPEasy nor MQTTLens?

TD-er, 2021-04-28:
Here another [MQTT protocol handshake](https://bytesofgigabytes.com/mqtt/mqtt-protocol-handshake/)
[image]

kr0815, 2021-04-28:
According to the picture, there is no ACK - but in the text under it / in the Link, it says :

> After the reception of   SYN Flag=1 & ACK=1 in acknowledgment packet from server , client will send  acknowledgment packet with SYN Flag=0 & ACK=1 to inform the server that client has  received the acknowledgment of connection request and ready to communicate with you. Transmission of data only  happens after the successful TCP handshake. <

I found another example here: [https://www.msxfaq.de/code/powershell/ps_mit_mqtt.htm](url)

Although it´s in german - in the wireshark screenshot there is always a TCP ACK after each MQTT Message

I´m not really into TCP details, but isn´t it generally in TCP Protocol that each message has to be responded with an ACK? (TCP three way handshake)

In my Case, because of the statefull paket inspeection, Firewall sees a fault and rejects connection

kr0815, 2021-04-28:
please stop Investigation in that Problem for the moment  - i think it´s a problem of the Firewall

The subsribe request - if i understand it right, involves the ACK?

TD-er, 2021-04-28:
Just put the diagram and your Wireshark along each other:
[image]

As far as I can see, they do seem to be following the same steps in initiating the connection.

kr0815, 2021-04-28:
I don´t know - is an ACK needed after the Connect Ack or not?

Android App sends it - ESPEasy not

But i found out that in the Subscribe Request - there is also an ACK
Is this ACK sufficient according to the TCP Protocol?

Please see my examples in the first Post - there is a difference between ESPEasy and the Adroid App

Android App sends an additional ACK

Fact is - in the "firewalled" company network, Windows and Android  Apps like MQTTLens and others are working - ESPEasy doesn´t...

TD-er, 2021-08-12:
@kr0815 Have you done some more research on this issue?

tonhuisman, 2023-09-02:
This doesn't seem to be an ESPEasy issue, so can be closed.

closed 2023-09-02 by TD-er