letscontrolit/ESPEasy #3617: [Build] GitHub Actions building releases

TD-er, 2021-05-03:
This morning we had the first 'release' build using GitHub Actions (implemented by @mcspr )

Things that need to be improved:
- Build failed as one of the binaries apparently already existed?? See: https://github.com/letscontrolit/ESPEasy/actions/runs/806287809
- Each env is generating its own zip, which is great for faking download statistics, but highly impractical for users.
- Failing env builds are reported in the annotations, but do not fail the complete build

tonhuisman, 2021-05-03:
And not all zip files have been uploaded, apparently caused by 'authentication errors', whatever that means in this context (I'd expect time-out issues because of the 'large' number of files (100+2)).

TD-er, 2021-05-03:
At least one shows an exclamation mark, only to be seen when you edit the release:
[image]

TD-er, 2021-05-03:
@tonhuisman You had 5 commits made under another name, so in the text I merged them with the (impressive) 65 other commits.
The commits under a different name were:
```
      [Task* commands] Add support for TASKname/VARname in addition to TASKnr/VARnr + documentation
      [Documentation] Add TaskValueSetAndRun
      [Task* commands] Fix a few silly typos
      [Task* commands] Some code cleanup
      [Task* commands] Code cleanup and some corrections
```
So maybe you did those from another computer which has a different name in the Git config.

tonhuisman, 2021-05-03:
Well, I have switched laptop in December last year, but all commits for this release come from exactly the same laptop and account. So I can't explain why that happens, I've seen the same in previous releases. I don't mind, one is using my handle, the other the full name for that handle. Probably a 'Github thing'.

fly74, 2021-05-03:
Puh, why so many separate zip's in releases? Can the list plz be sorted?

tonhuisman, 2021-05-03:
> Puh, why so many separate zip's in releases? Can the list plz be sorted?

This?

>     * Each env is generating its own zip, which is great for faking download statistics, but highly impractical for users.

TD-er, 2021-05-03:
> I've also seen TD-er and this 'Gijs Noorlander' fellow in the releasenotes 😉, but that sounds more like 2 different accounts, right?

Yep me too, That's why I know where the setting differene may be located.

> Puh, why so many separate zip's in releases? Can the list plz be sorted?

I added the "old school" ZIPs also to the list and even linked from it in the release description.

FxAlex, 2021-05-03:
different sizes of bin files, i.e. _ESP_Easy_mega_20210503_normal_ESP8266_4M1M.bin_ in builds:
ESPEasy_normal_ESP8266_4M1M.zip - 963,488 bytes
ESPEasy_ESP82xx_mega-20210503.zip - 961,152 bytes

TD-er, 2021-05-03:
The small ZIP files were built on the build environment of GitHub Actions.
The large ZIP files (also linked from the text on the release page) are built on my own Linux build server.
So there can be small differences in build size due to the used path where the build is taking place.

mcspr, 2021-05-03:
> Build failed as one of the binaries apparently already existed?? See: https://github.com/letscontrolit/ESPEasy/actions/runs/806287809

Perhaps related to retries in the library. Docs suggest that the release asset name is 'locked', and needs manual removal before uploading the same name even when API returns some error. I think you should be able to see the failed upload though, check out the release asset list when 'Editing' the release

> Each env is generating its own zip, which is great for faking download statistics, but highly impractical for users.

Release script could be modified to merge everything in one .zip though, last lines simply push every artifact which could be another zip(...) call and only one upload. Not sure why statistics faking is considered an issue?

> Failing env builds are reported in the annotations, but do not fail the complete build

This goes back to the original problem - how to chain jobs together in a way that only the last job is failing. Right now release builds are in the middle, thus the last step won't trigger unless all jobs are successful.

mcspr, 2021-05-14:
> This goes back to the original problem - how to chain jobs together in a way that only the last job is failing. Right now release builds are in the middle, thus the last step won't trigger unless all jobs are successful.

Only got around to look at this, some WIP at the https://github.com/mcspr/ESPEasy/actions/workflows/release.yml and the ghactions-order branch
Will PR some time today, tomorrow... I want to look at script again and remove the temporary stuff.

Resulting .zip is 'squashed' from all of .zip artifact files that were produced by the previous steps. continue-on-error removed, assuming that everything is able to build. There could be a separate list for releases though, different from the one for PRs & default branch.

Also looking at the original actions/upload-release-asset, suggested replacement https://github.com/softprops/action-gh-release and https://github.com/google/github-release-retry - there still needs to be some kind of manual upload retry mechanism that will try upload and remove asset by name if that fails, wait a bit and try again a couple of times.
(with either of available API tools, octokit.js or the pygithub)

tonhuisman, 2023-09-01:
This has been fixed in the mean time, AFAICS, so issue can be closed.

closed 2023-09-01 by TD-er