letscontrolit/ESPEasy #3618: Mysterious sign

fly74, 2021-05-03:
Little Question what means this sign, everything works fine.

[image]

TD-er, 2021-05-03:
Some controllers can use an IDX, like the Domoticz controllers.
However the value `0` as an IDX is an invalid value for Domoticz.

Are you using Domoticz on that controller?
Or is it another controller supporting IDX values?

If it is another controller, then I guess it may be a warning which may not be needed.

fly74, 2021-05-03:
It's a Generic HTTP, set to 1 fixed it. thx.

TD-er, 2021-05-03:
OK, still makes it a bug as an IDX of 0 on HTTP controller is not really an issue.

fly74, 2021-05-03:
@TD-er  It depends on the backend, maybe some HTTP controller will affect it. Mine is not :)

tonhuisman, 2022-05-07:
AFAIR, this has been fixed, right?

fly74, 2022-05-07:
Not seen since a long time.

closed 2022-05-07 by TD-er

fly74, 2022-09-16:
@TD-er  The thing is back in ESP_Easy_mega_20220809_normal_ESP8266_4M1M

[image]

fly74, 2022-09-16:
It's the HTTP generic controller.

tonhuisman, 2022-09-16:
> It's a Generic HTTP, set to 1 fixed it. thx.

Wasn't this the 'fix'?

fly74, 2022-09-16:
Dann. You're right. Thx.