letscontrolit/ESPEasy #3626: MS5611 (GY-63) Sensor shows very high Values when Temperature is low

DrMucki, 2021-05-10:
Hi out there

I have a Problem with the mentioned Sensor (MS5611 (GY-63).
 My ESP Version is ESP_Easy_mega_20210223_normal_ESP8266_4M1M_VCC

The Environment - MS5611 (GY-63) Device was Set up correctly and its shows normal Values for temperature and pressure, as long as they are above 20°C. When you go lower than you get Values like this

Temperature: 145167.53
Pressure: 369257.69

Normal would be

Temperature: 23.31 °C and Pressure 1006 (I have a value divider by 100) but then not applied the value of cause is 100 times higher... so that works... but when having temperatures below 20°C I have the strange values.

Any Ideas?

TD-er, 2021-05-10:
One of the most recent changes to this plugin is this one: https://github.com/letscontrolit/ESPEasy/commit/f2f71c35400e18a6209f6bed0719fe4dc46f7075
Can you try a build from before Nov 22, 2020 ?
That's the day the commit I linked was merged.

DrMucki, 2021-05-10:
Thank you! That solved the problem!! Used a version before the mentioned date and the temperature and pressure now are shown correctly!
Great work!

TD-er, 2021-05-10:
OK, then I will revert that commit.
So no need to close this issue, as it isn't fixed (yet).

What version are you now using?

DrMucki, 2021-05-10:
I am using ESPEasy_ESP82xx_mega-20201102

TD-er, 2021-05-10:
Thanks.
I will look into it and may ask you to test a fix.

TD-er, 2021-05-10:
Can you pick a build from [here](https://github.com/letscontrolit/ESPEasy/actions/runs/829346938) (click on "Binaries" as I cannot give you a direct link to the bin)

DrMucki, 2021-05-11:
I tested the corresponding binary you supplied with success. Of cause I totally erased the ESP before flashing to get it everything from scratch.

Temperature and Pressure data are being displayed correctly as expected .

Thank you for your great help and work

TD-er, 2021-05-11:
Thanks!
I will merge the PR.

PR #3628 (2021-05-11): MS5611 (GY-63) unrealistic values for <20C (#3626)
Fixes: #3626
commit d5fa55ca (2021-05-11): MS5611 (GY-63) unrealistic values for <20C (#3626)

Fixes: #3626
closed 2021-05-11 by TD-er