letscontrolit/ESPEasy #3635: TSL2561 i2c error

anno006, 2021-05-14:
- [ ] TSL2561
- [ ] Do nothing, just wait for pir sensor out china than after that the TSL2561 will give this error, i2c error and won't work anymore even after updating to new firmware, 20210223 normal esp8266 and the new 20210503, ESP_Easy_mega_20210503_normal_ESP8266_4M1M

### Steps already tried...
- [x ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x ] Tested previous/other build (mention which one already tested)
- [x] Tested on other node to make sure hardware isn't defective.
- [x ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

What need to happen? the need to be a lux value?!

### Actual behavior

i2c error in log

### Steps to reproduce

1. make the esp and connect the TSL2561
2.  order pir sensors
3.  wait a month, in the meantime the esp will be not powered on the shelf also waiting for the pir sensors.
4.  connect the pri sensors and power the esp.
5. Now you get a i2c error from the TSL2561. On step 1 the TSL2561 did work properly.

TD-er, 2021-05-14:
What pins do you use for the PIR sensor?
What PIR sensor do you use?
If you disconnect this sensor, is the I2C sensor working again?
Can you find the I2C sensor via an I2C scan?

I do suspect a wiring issue, or maybe the other sensor does cause issues (e.g. putting 5V on some pin)

tonhuisman, 2021-05-14:
This would have been a great [Forum](https://www.letscontrolit.com/forum) question... these Github issues are meant to register ESPEasy software issues, feature requests etc.

anno006, 2021-05-14:
> What pins do you use for the PIR sensor?
> What PIR sensor do you use?
> If you disconnect this sensor, is the I2C sensor working again?
> Can you find the I2C sensor via an I2C scan?
> 
> I do suspect a wiring issue, or maybe the other sensor does cause issues (e.g. putting 5V on some pin)

io don't, did already check that. 

The ic2 is working fine, the are other sensors that do work properly. After being on the shelf for a month the sensor itself won't work anymore. The pir sensors are not connected to the esp, but to a MCP23017, and yes this was already when the tsl2561 did work a month ago.

Yes, the tsl2561 is showing on the scan.

anno006, 2021-05-14:
> This would have been a great [Forum](https://www.letscontrolit.com/forum) question... these Github issues are meant to register ESPEasy software issues, feature requests etc.

I think it is a software issue, so?

TD-er, 2021-05-14:
What is the exact error in the logs?

anno006, 2021-05-14:
> What is the exact error in the logs?

TSL2561: i2c error

TD-er, 2021-05-14:
What else is connected to that I2C bus?
And are you still using the build that was working, or does this error occur right after updating the firmware?

TD-er, 2021-05-14:
The reason I'm asking is because this (not really descriptive) error only shows up if the I2C communication was unable to complete.
So either the I2C bus is pulled into an unknown state by some other device, or the used speed is too much for your setup. 
What is too much can be caused by wiring (length), whether or not pull-up resistors are being used, or some I2C device that may need to have I2C clock stretch setting changed to work.

anno006, 2021-05-14:
> What else is connected to that I2C bus?
> And are you still using the build that was working, or does this error occur right after updating the firmware?

"even after updating to new firmware, 20210223 normal esp8266 and the new 20210503"

anno006, 2021-05-14:
> The reason I'm asking is because this (not really descriptive) error only shows up if the I2C communication was unable to complete.
> So either the I2C bus is pulled into an unknown state by some [...]

Never did I! have any problems with wire length by i2c bus. I even have a esp with i2c bus running and about 3 meters of wire between a temp sensor and the esp.

This is the only one that has this problem. And i run a lot of i2c sensors in my home. I did spend another 3 euros for nothing, the sensor itself is not broken, also no wrong wiring, because a temp sensor and the MCP are running great. And the TCL is showing in the i2c scan tab. Also on the internet the TSL2561 is a fast i2c bus, 400kHz but when i try to force 400kHz, the TSl is gone in the i2c scan tab. So i really think a plugin or firmware error somewhere.

tonhuisman, 2021-05-14:
Most likely your TSL2561 is damaged in some way, unless you have it working in another setup. I've used 3 TSL2561's connected on the same bus (on the 3 different addresses available, about 30 cm wire in total) that has been working fine ever since I bought them 3 years ago. If other devices on your ESP I2C bus are working as expected, then I'd suspect the TSL2561 to be faulty.
Sure the SCL/SDA connections to the TSL aren't switched? With _some_ devices, mixing up these wires the I2C scan still seems to work, but the device itself doesn't operate as expected...

anno006, 2021-05-14:
> Most likely your TSL2561 is damaged in some way, unless you have it working in another setup. I've used 3 TSL2561's connected on the same bus (on the 3 different addresses available, about 30 cm [...]

I did try that and you know what?????............................... The sensor is now completely NOT!! working. Because i did had the wires in the good order, as i did say in my first topic!!! and it is't damaged, as you can reed in my first post, you do not reed other peoples post do you?

TD-er, 2021-05-14:
Just to summarize the information here as it tends to get quite messy over a number of comments:

- ESP + sesnors was working fine when shelved
- ESP + sensors taken off shelve, nothing changed => Stopped working
- Tried new FW version => Not working

Just when looking at those 'steps' I am not thinking about a firmware bug, but something damaged of gotten loose when you took it off the shelve, right?

The sensor itself may work fine at 400 kHz, so that's not the problem of the sensor, but like I said it may very well be that loose wiring or lots of other factors may limit the max. achievable I2C frequency.
This seems to be supported by the fact that scanning at 400 kHz does not show the sensor anymore.
This could be caused by other sensors too, so it does not mean the TSL is damaged by the way.

Can you try to test the sensor without anything else connected to it?

tonhuisman, 2021-05-14:
> This seems to be supported by the fact that scanning at 400 kHz does not show the sensor anymore.

We changed that to scan at the configured low I2C speed, in the second half of last year.

> you do not reed other peoples post do you?

I did before I replied, and all facts I can find there lead me to thinking there is some damage to your sensor, most likely cause: ESD (electrostatic discharge)
Swapping SCL/SDA on an I2C sensor does _not_ damage the device electrically, it just won't work.

TD-er, 2021-05-14:
Just had a call with him and when setting both high and low I2C clock freq. to 100 kHz, the sensor worked again.
He told me the MCP23017 was having some issues in the past where the sensor every now and then stopped working, so after putting it to 400 kHz then solved all issues since with that chip.
Not sure why the TSL2561 does have issues with running at 400 kHz, as it should work fine on that frequency according to the datasheet.
Maybe having multiple sensor modules (!!) on the same bus does put the pull-up resistors parallel which may in the end cause a too low resistance as pull-up which might cause instabilities for some.

So he will also look into the use of multiplexers as those separate the I2C bus per chip, making it hopefully more flexible.

For now I guess we can close this issue as it is working now.

closed 2021-05-14 by TD-er