letscontrolit/ESPEasy #3638: Wake up Problem after DS, Wemos D1, with ESP_Easy_mega_20210503_normal_ESP8266_4M1M_VCC.bin

roschme, 2021-05-14:
Hi diligent guys,

i've a Problem with ESP_Easy_mega_20200829_normal_ESP8266_4M1M_VCC.bin on Wemos D1 mini.

System goes to deep-sleep but don't wake up (RST-GPIO16=D0 is jumpered).
I've checked the problem with a second Wemos running ESP_Easy_mega_20200829_normal_ESP8266_4M1M_VCC.bin everything is fine and wakeup works as it should. 

After downgrading the first Wemos to ESP_Easy_mega_20200829_normal_ESP8266_4M1M_VCC.bin it worked also as it should.

I'm wondering about the GPIO16 entry in the hardware GPIO boot states (ESP_Easy_mega_20200829_normal_ESP8266_4M1M_VCC.bin). 
I tried the different configs for the GPIO but the strange behaviour remains awake was'nt possible.

Thank you for your really great work!!!
B.

TD-er, 2021-05-14:
That's quite a big jump in version numbers.
Can you try to find a shorter build interval to see where things may have gotten broken?
I don't think it is that far back as I did not see lots of reports on this.

tonhuisman, 2021-05-15:
@TD-er didn't you recently fix something in that area? Don't know if it's already merged though, was a rather small fix involving inverting a comparison somewhere, IIRC 🤔

TD-er, 2021-05-15:
Not entirely sure, also thought I have done something, but adding GPIO 16 to the boot state selector was done quite a long time ago.

tonhuisman, 2021-05-15:
I think the fix had to do with going into really low power state on sleep, not with wake-up, so it could be unrelated.

TD-er, 2021-05-15:
Ah the "binary blob nobody really knows what it does..." ?
https://github.com/letscontrolit/ESPEasy/blob/7e7c6020721da9ceabb9d6d03bddb84e0751f2bd/src/src/Helpers/DeepSleep.cpp#L134-L163

tonhuisman, 2021-05-15:
No, there was something, much more recent, but I haven't been able to find it yet, even looked in the forum... 😞 maybe have to look better.

TD-er, 2021-05-15:
Maybe give me some more word suggestions?

tonhuisman, 2021-05-15:
Ah, found it, eventually, a new message in this [older forum thread](https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=7389&p=51370)

TD-er, 2021-05-15:
Hmm strange, I thought I did revert it back then... Apparently not.
Maybe we should make it configurable?
After all, it is claimed to work on some ESP modules...
Probably got to do with chip silicon revisions.

roschme, 2021-05-15:
Hi,

sry for the big jump in revisions (and my missmatch in versionnumbering above). The second Wemos was already flashed and fully working with the 20200829 Version so i used this version for the crosscheck.

addionally i mentioned the strange current consumption of the wemos with 20210503 Version: 
- configured: 60 s sleep, 120 s wake
- go to deepsleep (~30 µA)
- after ~20 s current goes up to 65 mA for about 5-10 s
- then deep sleep ~30 µA again
- all time: no access, no login to my W-LAN, just a hard power cycle helps for fixing status

With Version 20200829:
- go to ds ~20 µA
- wake up, everything working 75 mA 
- got to ds ~20 µA
timing and everything else ok!

TD-er, 2021-05-15:
If you could also test a version before Dec 8 and version mega-20201227 (the first one with that commit merged), then we can be sure it has something to do with this commit.

blb4github, 2021-05-16:
I'm experiencing issues with some of my 7 modules, all worked fine with version ESP_Easy_mega_20210223_normal_ESP8266_4M1M. Maybe this helps.

TD-er, 2021-05-16:
> I'm experiencing issues with some of my 7 modules, all worked fine with version ESP_Easy_mega_20210223_normal_ESP8266_4M1M. Maybe this helps.

The same issues as described here, that the units may consume more power in deep sleep and some may not even wake up from deep sleep?

blb4github, 2021-05-17:
Sorry, It could be something (completely) different. I'm not using deep sleep but just see some modules not responding anymore after some time with the latest version and can only reset to get them working again.

TD-er, 2021-05-17:
OK, yep that's also what I (now) understand from other reports as well... Seems to be something funky in the latest changes related to reconnecting. But that's different from this issue.

tomstrom, 2021-05-24:
Here is the same. He don't wake up. The Problem is, I think it dont reconnect to WLAN. If I do a RESET by the switch, the WLAN also don't come up.

TD-er, 2021-06-07:
Another user reported similar issues on [the forum](https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=8593&p=53071#p53071)

PR #3677 (2021-06-07): [Deep Sleep] Revert default deep sleep call & make it optional
Reverting this commit for default behaviour: https://github.com/letscontrolit/ESPEasy/commit/c70e54c03e99776dedc705a452066bc63cc60374

Fixes: #3638
commit 59837c48 (2021-06-07): [Deep Sleep] Revert default deep sleep call & make it optional

Reverting this commit for default behaviour: https://github.com/letscontrolit/ESPEasy/commit/c70e54c03e99776dedc705a452066bc63cc60374

Fixes: #3638
closed 2021-06-07 by TD-er