letscontrolit/ESPEasy #3651: RTTTL module not present in test_a build?

algorni, 2021-05-21:
### Checklist
- [X] The title of this issue is "to the point" and descriptive.
- [X] This issue describes what is happening.
- [X] This issue describes what components are affected (e.g. name of plugin/controller)
- [X] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [X] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [X] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [X] Tested on other node to make sure hardware isn't defective.
- [X] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I'm trying to leverage HTTP call to run RTTTL on TESTA build but seems that module is not there in the TEST build.
Would be possible to include the RTTTL in this build?

### Expected behavior

RTTTL playback

### Actual behavior

This is working:
6972143: HTTP: tone,15,1300,250

This is not working:
6994345: HTTP: rtttl,15:d=4,o=5,b=112:8a,8a,a,8a,8a,a,8a,8c6,8f.,16g,2a,8a-,8a-,8a-.,16a-,8a-,8a,8a.,16a,8a,8g,8g,8a,g,c6
6994347: RTTTL : command not included in build

### Steps to reproduce

1.  Just make an HTTP call with a Command to run a RTTTL schema

### System configuration

Hardware: Node MCU 

ESP Easy version: ESP_Easy_mega_20210503_test_A_ESP8266_4M1M.bin

tonhuisman, 2021-05-21:
It does actually fit in that build, so please try this test-build I just made:
[ESP_Easy_mega_20210521_test_A_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6523593/ESP_Easy_mega_20210521_test_A_ESP8266_4M1M.zip)
(Didn't have a test-device at hand, so not tested on real hardware)

algorni, 2021-05-21:
BINGO!!  now i've my esp8266 board with RTTTL capabilities + my BlueDot with BME280+ TSL2591 + 2 relay +  HC RS04 to make my IKEA standup task more smart! 

I can acquire temperature humidity and luminosity and control the height of my standup desk and with the relay i can even command it!  

Thanks @tonhuisman!!

algorni, 2021-05-21:
How does it works?  from now on you will add this module in the test_a build?  shall we close the issue?

uzi18, 2021-05-21:
@algorni please attach some more info/photos

TD-er, 2021-05-21:
> @algorni please attach some more info/photos

I guess @algorni wants to know if the RTTTL will be included in the next (nightly)  build :)

tonhuisman, 2021-05-22:
Adding such feature to a single build is a bit complicated, but I'll see if I can get a more common way working. Requires to make a lot of builds, so it's quite time-consuming. Currently the `USE_RTTTL` feature is disabled by the `LIMIT_BUILD_SIZE` check, and the TESTING builds all fall in that category.

tonhuisman, 2021-05-22:
>  shall we close the issue?

Please leave it open as a reminder, it'll be closed through a pull request once that gets merged.

algorni, 2021-05-22:
@TD-er  you are right, that's what i was looking to understand, IF this RTTTL feature will be enabled on those TEST build from now on or not. 

BTW is there a way to understand when a test feature will be promoted to NORMAL ?   e.g. the TLS  luminosity sensor i'm using is on TEST A but...   honestly seems working fine...   how this can be promoted to a NORMAL status?

tonhuisman, 2021-05-22:
Most of the stuff that has TESTING status is working (very) fine, else it wouldn't have made it into the mega repository 😄 

NORMAL, DEVELOPMENT and TESTING used to be statuses that sort of indicated in what binary it was included, but as the number of plugins has grow over time, resulting in the bin size to hold them all no longer fits in the ESP8266 max. sketch size (just below 1 MB), so these tags no longer fit the bill.

One solution would be to have everybody build their own CUSTOM build, but that would take the Easy out of ESPEasy for many users, and another solution, that is in the design phase, would be to have an online build server where a user can check a couple of options (plugins, controllers and other features), click a button, and presto!, a few minutes later the personalized .bin file will be available for download.
As that second solution isn't available yet, we split up the, already quite large, TESTING build into separate sets (A..D), hoping that the combinations work for most users.

A fourth solution, added a short while ago, is support for ESP32 with 16 MB Flash, named MAX, where all plugins, controllers and features, that actually work on ESP32, are included, as ESP32 allows for up to 4 MB sketch size (currently the MAX sketch is about 2.5 MB in size, but ESP32 code compiles into larger binaries as ESP8266 does). Downside is that this ESP32 with 16 MB Flash board is larger and more costly than the common ESP8266 boards (mostly NodeMCU and Wemos D1 mini clones). Upside is that the ESP32 is much faster than an ESP8266 and has more RAM and GPIO pins available, making it much more capable. The bluetooth feature of the ESP32 chip isn't enabled in ESPEasy yet, though.

algorni, 2021-05-23:
I remember for a LUA firmware there was a kind of online tools to build your own firmware with selected features available in your built!

Would be wonderful to keep the EASY part of ESP Easy while having the opportunity to have custom-built with the sensors you need!

ghtester, 2021-05-23:
It's not an online solution but if one could afford an old PC (or a virtual machine ) capable of running Ubuntu Linux & Vagrant environment, then making a Custom ESP Easy builds is easy: https://github.com/letscontrolit/ESPEasy/issues/2594

uzi18, 2021-05-23:
@algorni @ghtester you can use gitpod for this easily

ghtester, 2021-05-23:
Yeah thanks for a hint, it looks interesting but it would also need a more learning to use that. Also I am afraid of continual changes in such online platform as I am aware of a compilation sensitivity to many conditions...
 Currently I am happy with local Vagrant environment which I have fully under my control and customized for all my needs, it works really fine.

uzi18, 2021-05-23:
actually i used it for almost all changes or debug stuff in ESPEasy

ghtester, 2021-05-23:
OK, that sounds great but as there's not (or I did not find) any HowTo for building ESP Easy through that platform, starting from zero is not so easy (I think). On the other side, Vagrant was documented here and it helped a lot.

uzi18, 2021-05-24:
@ghtester add extension https://addons.mozilla.org/firefox/addon/gitpod/ or https://chrome.google.com/webstore/detail/gitpod-dev-environments-i/dodmmooeoklaejobgleioelladacbeki <- it adds gitpod button on all github projects in right top corner - so you can test PRs etc.
or if you want just to compile clean project just click here: 
https://gitpod.io/#https://github.com/letscontrolit/ESPEasy 
choose login by github

wait a moment to prepare the workspace
edit tools/pio/pre_custom_esp82xx.py or upload Custom.h into src dir (check/copy Custom-sample.h as draft)
on terminal - bottom right paste line like this: `pio run -e custom_ESP8266_4M1M` and hit enter (if you want different env choose other one)

After succesfull build your bin is located in build_output/bin click right mouse button and choose download on bin file
Remember to download Custom.h file for next time because your workspace will be deleted after 1-2 weeks

ghtester, 2021-05-24:
Wow, it's a great How To, thanks a lot! I'll try for sure.

PR #3654 (2021-05-31): [Build] Re-enable RTTTL in TESTING where it fits
Re-enables the `rtttl` command in the TESTING builds where it fits.
Mostly TEST_B ESP8266 builds don't fit, and also not in `alt_wifi` and `beta` versions of the Arduino libraries.

Resolves #3651
closed 2021-05-31 by TD-er