letscontrolit/ESPEasy #3652: TSOP4838 Plugin: Long Hex Codes gets cutted

tbnobody, 2021-05-22:
I am using ESP_Easy_mega_20210503_minimal_IRext_ESP8266_1M

When I enter the received hex code in the TSOP4838 plugin the code gets cutted after a specific length.
E.g. I enter:
[image]

Then I press submit and get:
[image]

It seems to work with shorter Hex codes (e.g. 10001C3 is working)

tonhuisman, 2021-05-22:
The variable that stores the value is only 32 bits wide, so anything above that isn't going to fit there.
Seems like a bug to me, unless the specifications for what's allowed there says otherwise.
As the plugin currently uses `uint32_t` for these variables, I guess some rework may be required.

tonhuisman, 2021-05-22:
Aha, the actual error is in the parsing of the (hex) input to `uint32_t`, the used conversion `strtol` results in a signed value, so values between 0x7FFFFFFF and 0xFFFFFFFF won't fit. I've replaced it with `strtoul`, that should fix it.

Can you try with this attached build?

[ESP_Easy_mega_20210522_minimal_IRext_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6526715/ESP_Easy_mega_20210522_minimal_IRext_ESP8266_4M1M.zip)

tbnobody, 2021-05-22:
Is the attached build really a 1M build? (controller is a 8285 with only 1M memory)

tonhuisman, 2021-05-22:
Ah, sorry, didn't look close enough, here's a 1M build

[ESP_Easy_mega_20210522_minimal_IRext_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/6526821/ESP_Easy_mega_20210522_minimal_IRext_ESP8266_1M.zip)

tbnobody, 2021-05-22:
Thank you! The Hex code is saved correctly now. I am just no sure wether it's an usage mistake or program error. 
I configured the plugin in the following way:
[image]

I am getting the following output in the console
```
888514 : Info   : IRSEND,'{"protocol":"NEC","data":"0x807F708F","bits":32}'
888524 : Info   : EVENT: IRReceive#IR=2155835535
```

But it seems that the command is not executed.

tonhuisman, 2021-05-22:
Which of the 3?
1) the rule `on IRReceive#IR do` is not executed?
2) the rule `on IRReceive#IR=2155835535 do` is not executed?
3) the `IRSEND` command is not executed?

tbnobody, 2021-05-22:
I would say there is a 4th method.
If I understood the documentation correctly, the command `publish %sysname%/irkey,1` should be executed when the IR Code 807F708F is received. (see screenshot above)

tonhuisman, 2021-05-22:
I did make another similar change, that is also in the PR I made to fix this, and that is probably the fix you need, new attachment (same name though, I'll remove the other attachment)

[ESP_Easy_mega_20210522_minimal_IRext_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/6526983/ESP_Easy_mega_20210522_minimal_IRext_ESP8266_1M.zip)

tbnobody, 2021-05-22:
Hm I tried this version as well. But the commands specified in the commands textbox are not getting executed. 
[image]

Thats the output of the serial console when I press the buttons on the remote control:
```
361660 : Info   : IRSEND,'{"protocol":"NEC","data":"0x807F708F","bits":32}'
361669 : Info   : EVENT: IRReceive#IR=2155835535
362359 : Info   : IRSEND,'{"protocol":"NEC","data":"0x807F609F","bits":32}'
362368 : Info   : EVENT: IRReceive#IR=2155831455
362959 : Info   : IRSEND,'{"protocol":"NEC","data":"0x807FF00F","bits":32}'
362968 : Info   : EVENT: IRReceive#IR=2155868175
```

I don't see any publish events nor do I see published events on the mqtt broker. 
The IRReceive topic btw is receiving updates:
[image]

tonhuisman, 2021-05-23:
Can you remove, f.e., the last code, enable the 'Add new received code to command lines' option, and send that code again, to see how/what is added?

tbnobody, 2021-05-23:
Ok, thats interesting....

```
52133601 : Info   : EVENT: TaskInit#IRReceive=2,1
52138804 : Info   : IRSEND,'{"protocol":"NEC","data":"0x807FF00F","bits":32}'
52138806 : Info   : [P36] AddCode: 0x837FF00F to index 2
52138812 : Info   : EVENT: IRReceive#IR=2155868175
52139124 : Info   : P016_PLUGIN_ONCE_A_SECOND CustomTaskSettings Saved
```
(The Code in `IRSEND` and `AddCode` is different)

But when I now add a command it get's executed:
[image]

tonhuisman, 2021-05-23:
So this means the issue is resolved then? Great 👍 

Don't close this manually please, it will be closed automatically once the PR that I made is merged, and Github will send the corresponding notifications.

tbnobody, 2021-05-23:
hm not sure if resolved? I mean IRSEND shows code 0x807FF00F and AddCode shows 0x837FF00F... not sure why this happens or if its correct in this way? In my opinion both numbers should be the same?

tonhuisman, 2021-05-23:
All I did here was parse the value you can input to the correct result, the rest of the inner workings I didn't touch. And I've never used such IR plugin either, so that could be another, maybe not even related, bug. The left byte (first 2 positions) of the code is directly derived from what is received.

I got a few improvement suggestions in the PR, I'll implement those and upload a new test build here.

TD-er, 2021-05-23:
Looks like 2 bits difference.
Could it be a reception error?
How reliable is the IR reception for such long codes?

tonhuisman, 2021-05-23:
That code is actually assembled, the first byte (MSB) is the decode_type code, then it gets a (1) bit for 'repeat' and then the value received is added. Currently the plugin only supports 32 bits for the total code, but it is possible (read: very likely) that the value is > 24 bits, so it overflows into the command byte (bits 31-24). Also, the `value` field in the receiver data structure is an `uint64_t`. If that is the case, the plugin needs a big overhaul, as currently the stored codes are max. 32 bits.
In the log, the 'IRSEND' line shows the protocol, and 'NEC' is probably the 3 there.
I've just added an extra part to that log line, stating the numeric value of that type.

[ESP_Easy_mega_20210523_minimal_IRext_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/6528380/ESP_Easy_mega_20210523_minimal_IRext_ESP8266_1M.zip)

tbnobody, 2021-05-23:
Hmm thats the output now:

```
281774 : Info   : IRSEND,'{"protocol":"NEC","data":"0x807FF00F","bits":32}' type: 0x3
281776 : Info   : [P36] AddCode: 0x3807FF0 to index 2
281784 : Info   : EVENT: IRReceive#IR=2155868175
282833 : Info   : P016_PLUGIN_ONCE_A_SECOND CustomTaskSettings Saved
```

tonhuisman, 2021-05-23:
Hm, It looks like I broke something, as that `AddCode` isn't correct any longer, I'll look at that change.

tonhuisman, 2021-05-23:
The actual issue here is that in the current implementation, the `decode_type` byte value is stored in the highest 8 bits of the 32 bits available for the code that probably never exceeded 23 bits, as the 24th bit is used for `repeat` status, but your remote is actually sending 32 bit values.

There are a few possible solutions to this, but I'd like to involve @uwekaditz, the original author of this plugin, before 'messing it all up' 😉 

Uwe, can you please share your thoughts on this subject?

@tbnobody attached a correction for the AddCode issue
[ESP_Easy_mega_20210523_minimal_IRext_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/6528543/ESP_Easy_mega_20210523_minimal_IRext_ESP8266_1M.zip)

ghtester, 2021-05-23:
I was also struggling with TSOP4838 Plugin's Execute commands option and it did not work for me as my Remote Control also send 32 bits, also I have another RC with 36 bit length so even though the issues mentioned above are fixed, there'll be still a limitation...

tonhuisman, 2021-05-23:
I've been thinking of 'upgrading' this structure to use 64 bit (replace `uint32_t` by `uint64_t`, and while that seems simple, it requires a conversion and a (con)version flag in the settings, to see if conversion is still needed.
Maybe that should be a new issue & PR.

tbnobody, 2021-05-23:
I've just tested the latest attached version of the precompiled binary again:
```
22414 : Info   : IRSEND,'{"protocol":"NEC","data":"0x807FF00F","bits":32}' type: 0x3
22415 : Info   : [P36] AddCode: 0x837FF00F to index 2
22422 : Info   : EVENT: IRReceive#IR=2155868175
23045 : Info   : P016_PLUGIN_ONCE_A_SECOND CustomTaskSettings Saved
```

uwekaditz, 2021-05-23:
The structure for adding and saving the IR commands can be 'upgraded' to 64 bits.
As @tonhuisman mentioned there is the need for conversion and a conversion flag because we must support the users using the 32bit structure.
That means 54bits can be used with the new structure for the IR code.
The problem I see are there other RCs which can send codes even longer as 54bits?

tonhuisman, 2021-05-23:
> The problem I see are there other RCs which can send codes even longer as 54bits?

Looking at the `IRrecv.cpp` source and comments, there are at least a couple that support 56 bits, and I even saw mentions of ~424 bits (size constants), but I'm not sure how to interpret that.
Maybe the `decode_type` and `repeat` bits should be stored in separate struct members. (as we are about to change stuff, best do it as extensive as might be needed).

tonhuisman, 2021-05-24:
Looked a bit closer at those bit definitions, and indeed the worse case that is supported is 424 bits for some Hitachi AC equipment. But as those long codes aren't supported by this plugin for sending commands, we don't need to store such large codes.

I've changed an initialization, to be able to use `hexToUL()` instead of `strtoul()` (as requested by TD-er), so @tbnobody can you please repeat the above test with AddCode enabled again, to see if it now handles code detection correctly?

[ESP_Easy_mega_20210524_minimal_IRext_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/6530850/ESP_Easy_mega_20210524_minimal_IRext_ESP8266_1M.zip)

uwekaditz, 2021-05-24:
The `IRrecv.cpp` provides a hash functionality.
May be we should store the IR code as hash value if it is longer than 64 bits (requires another `hash` bit in the `P016_data_struct`)

tbnobody, 2021-05-24:
This is how it looks now:
```
102214 : Info   : IRSEND,'{"protocol":"NEC","data":"0x807FF00F","bits":32}' type: 0x3
102216 : Info   : [P36] AddCode: 0x3807FF0 to index 2
102222 : Info   : EVENT: IRReceive#IR=2155868175
102905 : Info   : P016_PLUGIN_ONCE_A_SECOND CustomTaskSettings Saved
```

tonhuisman, 2021-05-24:
> (requires another `hash` bit in the `P016_data_struct`)

I was thinking of adding a `uint16_t` to store flags like this in the struct, and maybe use an `uint16_t` to store the `decode_type`, so there is enough room to expand.

> Seems that the last 0F is missing and the 3 was added...

That 3 was 'hidden' in the 0x83... where 0x80... was expected.

I think we should 'upgrade' this issue & PR to be able to handle codes up to 64 bits properly, it may require work in the IRSEND (P035) plugin as well, but that is to be determined.

tonhuisman, 2021-05-24:
> Seems that the last 0F is missing and the 3 was added...

This was caused by the 0x prefix not being accounted for in the `hexToUL()` converter. Should be fixed now, can you please test again?

[ESP_Easy_mega_20210524_minimal_IRext_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/6531548/ESP_Easy_mega_20210524_minimal_IRext_ESP8266_1M.zip)

uwekaditz, 2021-05-24:
@tonhuisman can you provide the code you have changed that I can modify the P016 plug in.

tbnobody, 2021-05-24:
This is the output of the lateest binary...
```
457308 : Info   : IRSEND,'{"protocol":"NEC","data":"0x807FF00F","bits":32}' type: 0x3
457310 : Info   : [P36] AddCode: 0x837FF00F to index 2
457316 : Info   : EVENT: IRReceive#IR=2155868175
458191 : Info   : P016_PLUGIN_ONCE_A_SECOND CustomTaskSettings Saved
```

tonhuisman, 2021-05-24:
> @tonhuisman can you provide the code you have changed that I can modify the P016 plug in.

Please view PR #3653

tonhuisman, 2021-05-24:
> @tonhuisman can you provide the code you have changed that I can modify the P016 plug in.

@uwekaditz are you going to pick up the task of converting the plugin to 64 bit values? I'd hate to start modifying the code and find out it is already/also being done by someone else 😉

uwekaditz, 2021-05-24:
@tonhuisman I havn't started yet.
If you want to do the modification, no problem

tonhuisman, 2021-05-29:
I have a first version of the P016 IR receiver plugin that supports 64 bit codes. It automatically converts the current settings **not backward compatible**, (but only after they are saved, so if you want to revert a previous version after saving settings, or adding a code, you will need to re-enter all codes!)

Attached are both a 1M build and a 4M1M build, for those that want to test this (and please do!)

[ESP_Easy_mega_20210529_minimal_IRext_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6565185/ESP_Easy_mega_20210529_minimal_IRext_ESP8266_4M1M.zip)
[ESP_Easy_mega_20210529_minimal_IRext_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/6565186/ESP_Easy_mega_20210529_minimal_IRext_ESP8266_1M.zip)

I've ordered a couple TSOP4838's, but these will take some time to arrive, so I can't currently test it myself, nor do I own many IR senders (only TV/audio remotes), so any help is really appreciated.

uwekaditz, 2021-05-30:
@tonhuisman can you provide the changed source code? I can test the changes with my project, but I also have no RCs sending a code longer the 24bits.

TD-er, 2021-05-30:
@uwekaditz I think it is based on the changes in this PR: https://github.com/letscontrolit/ESPEasy/pull/3653

tbnobody, 2021-05-30:
It seems to crash when clicking on the existing device in the web gui:

```
83185 : Info   : P016_PLUGIN_WEBFORM_LOAD ...
83198 : Error  : P016 IR: Settings conversion, save task settings to store in new format.
83198 : Info   : P016: struct size: 880/88/72 enum: 2
83202 : Info   : P016: converting 0: type UNKNOWN code 0x807F708F
83203 : Info   : P016: converting 1: type UNKNOWN cod
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3fff1e90 end: 3fffffb0 offset: 01a0
3fff2030:  00000000 0194882b 40274bd8 40273b15  
3fff2040:  40274bf1 00000001 3fff3a3c 40274be6  
3fff2050:  00000001 000005fd 3fff567c 40204259  
3fff2060:  00000190 00000000 3fff567c 402042e1  
3fff2070:  00000000 3fff20d2 40100a4c 3fff1404  
3fff2080:  3fff1a20 00000190 3fff13e0 40243b2d  
3fff2090:  0d303931 0000000a 3fff1a20 40105e35  
3fff20a0:  40270420 3fff3a3c 3fff2158 3fff1b64  
3fff20b0:  3fff1a20 000020d8 3fff19f8 40243bb5  
3fff20c0:  00000190 3fff2130 00014535 00000000  
3fff20d0:  0000006c 3fff2130 3fff19f8 00000003  
3fff20e0:  3fff1a20 3fff2130 3fff19f8 40243cb5  
3fff20f0:  3fff2164 00000005 00000001 402728ac  
3fff2100:  0000003f 402c0398 3fff2164 00000000  
3fff2110:  0000003f 402c0398 3fff2130 40243d34  
3fff2120:  00000000 402c0398 00000020 402636ae  
3fff2130:  00000000 00003ee0 00002628 00002628  
3fff2140:  00003440 00000000 00000000 00001900  
3fff2150:  00000000 00000000 3fff8b00 018d018f  
3fff2160:  80003440 756c6176 00000065 85001900  
3fff2170:  00000000 00000000 3fff8b00 40263bc9  
3fff2180:  80000020 74706f00 206e6f69 80270000  
3fff2190:  00000000 3fff25a4 3fff21e0 40264600  
3fff21a0:  0000003e 00003ee0 00002ec0 000030c8  
3fff21b0:  0000003f 00000000 3fff21e0 402721e5  
3fff21c0:  00000000 00000000 00000000 0000003f  
3fff21d0:  0000003f 00000000 3fff2740 402646ac  
3fff21e0:  00000000 3fff28f0 80ff03e8 40264415  
3fff21f0:  00000061 3fff22b0 00000080 40272938  
3fff2200:  3fff28c4 3fff28f0 00000001 00000061  
3fff2210:  3fff22b0 3fff2740 00000000 4026488f  
3fff2220:  00000000 402c7da8 3fff28c4 402728ac  
3fff2230:  00000080 3fff8594 3fff28c4 402728dc  
3fff2240:  402c7da8 3fff8594 3fff28f0 00000000  
3fff2250:  00000000 3fff8594 3fff28c4 40238de8  
3fff2260:  00000000 00000001 3fff28c4 3fff086e  
3fff2270:  40102a67 3ffee240 00000002 00000000  
3fff2280:  402bfd70 00000020 3fff2a50 40272292  
3fff2290:  3ffeb03c 00000000 00000000 40272448  
3fff22a0:  0000000b 04acd1fc 4010310a 00000100  
3fff22b0:  53554e55 7f004445 86002200 00354352  
3fff22c0:  402bfd70 83000018 00364352 40272383  
3fff22d0:  83000018 0043454e 0000001c 83ff1585  
3fff22e0:  594e4f53 00000000 84ff2a50 414e4150  
3fff22f0:  494e4f53 892b0043 0043564a 402728ac  
3fff2300:  83238970 534d4153 00474e55 87238a35  
3fff2310:  4e594857 00524554 87238984 3fff64f4  
3fff2320:  000c000f 00000000 0000474c fffffffe  
3fff2330:  82000700 594e4153 0000004f 85ff2b60  
3fff2340:  5354494d 53494255 8a004948 48534944  
3fff2350:  00000000 84000000 52414853 3fff0050  
3fff2360:  85ff4332 4c4f4f43 3f005849 86272a54  
3fff2370:  4b494144 40004e49 86ff23b0 4f4e4544  
3fff2380:  0000004e 85ff431c 564c454b 54414e49  
3fff2390:  8a00524f 52454853 444f4f57 88000800  
[... 1081 lines cut ...]
3fff6730:  00000000 00000000 00000000 00000000  
3fff6740:  00000000 00000000 00000000 00000000  
3fff6750:  00000000 00000000 00000000 00000000  
3fff6760:  00000000 00000000 00000000 00000000  
3fff6770:  00000000 00000000 00000000 00000000  
3fff6780:  00000000 00000000 00000000 00000000  
3fff6790:  00000000 00000000 00000000 00000000  
3fff67a0:  00000000 00000000 00000000 00000000  
3fff67b0:  00000000 00000000 00000000 00000000  
3fff67c0:  00000000 00000000 00000000 00000000  
3fff67d0:  00000000 00000000 00000000 00000000  
3fff67e0:  00000000 00000000 00000000 00000000  
3fff67f0:  00000000 00000000 00000000 00000000  
3fff6800:  00000000 00000000 00000000 00000000  
3fff6810:  00000000 00000000 00000000 00000000  
3fff6820:  00000000 00000000 00000000 00000000  
3fff6830:  00000000 00000000 00000000 00000000  
3fff6840:  00000000 00000000 00000000 00000000  
3fff685þ
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3584, room 16 

```

TD-er, 2021-05-30:
Can you also give the amount of free memory (before it crashes) ?

tbnobody, 2021-05-30:
Not sure if there is a more elegant way but I just waited to get the WiFi Status and clicked on the device:

```
31569 : Info   : WD   : Uptime 1 ConnectFailures 0 FreeMem 16344 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
34014 : Info   : EVENT: Clock#Time=Sun,11:53
49163 : Info   : P016_PLUGIN_WEBFORM_LOAD ...
49176 : Error  : P016 IR: Settings conversion, save task settings to store in new format.
49177 : Info   : P016: struct size: 880/88/72 enum: 2
49180 : Info   : P016: converting 0: type UNKNOWN code 0x807F708F
49181 : Info   : P016: converting 1: type UNKNOWN code 0
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset
```

TD-er, 2021-05-30:
That's good enough. 
16k free, so it is unlikely a memory allocation issue.

tonhuisman, 2021-05-30:
I have seen similar crashes, at PLUGIN_WEBFORM_LOAD, PLUGIN_WEBFORM_SAVE but most often at PLUGIN_EXIT _after_ that PLUGIN_WEBFORM_SAVE.
The crash at WEBFORM_LOAD is way less frequent if I give it a physical reset (not power-cycle) after the OTA update I usually do.
I have no real stack dump from this crash, but I did manage to get a stack dump after that save a few times (if I manage to reset it fast enough to avoid the hexdump to scroll the messages out of the PlatformIO buffer), that shows stuff like this:
(sorry for the long post)
It seems to suggest it crashes while sending via serial.
Free memory is in the range of 13 to 16k (another log is below)
```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40276368 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00002209 depc=0x00000000

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x40276368 in uart_tx_free at ??:?

>>>stack>>>

ctx: sys
sp: 3fff1ff0 end: 3fffffb0 offset: 0190
3fff2180:  3fff6204 00000001 3fff21d0 402724ce
...
3fff4430:  fffffff8 04df0352 00de03ab 02a100e1
3fff4440:  0

0x402724ce in String::String(unsigned long, unsigned char) at ??:?
0x4024c8e5 in process_serialWriteBuffer() at ??:?
0x4024cb67 in addToSerialBuffer(String const&) at ??:?
0x4024a825 in addToLog(unsigned char, char const*) at ??:?
0x4024a96a in addToLog(unsigned char, String const&) at ??:?
0x402b0032 in wifi_station_ap_number_set at ??:?
0x4024a980 in addLog(unsigned char, String const&) at ??:?
0x4023943e in Plugin_016(unsigned char, EventStruct*, String&) at ??:?
0x40271ec9 in String::invalidate() at ??:?
0x40271ee8 in String::~String() at ??:?
0x40239b78 in Plugin_016(unsigned char, EventStruct*, String&) at ??:?
0x40271f76 in String::changeBuffer(unsigned int) at ??:?
0x402bfa44 in chip_v6_unset_chanfreq at ??:?
0x4027212c in String::copy(__FlashStringHelper const*, unsigned int) at ??:?
0x402bfa44 in chip_v6_unset_chanfreq at ??:?
0x40272590 in String::operator=(__FlashStringHelper const*) at ??:?
0x402bfa44 in chip_v6_unset_chanfreq at ??:?
0x40272067 in String::reserve(unsigned int) at ??:?
0x40272590 in String::operator=(__FlashStringHelper const*) at ??:?
0x4010407b in lmacProcessTXStartData at ??:?
0x40104078 in lmacProcessTXStartData at ??:?
0x40102fdb in wDev_ProcessFiq at ??:?
0x401003e0 in ets_post at ??:?
0x40102e2c in wDev_ProcessFiq at ??:?
0x401003e0 in ets_post at ??:?
0x401065a0 in spi_flash_write at ??:?
0x4027049c in EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at ??:?
0x401065a0 in spi_flash_write at ??:?
0x40276da4 in flash_hal_write(unsigned int, unsigned int, unsigned char const*) at ??:?
0x4027049c in EspClass::flashWrite(unsigned int, unsigned int*, unsigned int) at ??:?
0x40276da4 in flash_hal_write(unsigned int, unsigned int, unsigned char const*) at ??:?
0x401065a0 in spi_flash_write at ??:?
0x401003e0 in ets_post at ??:?
0x40102301 in rcUpdateTxDone at ??:?
0x40101ed0 in pp_post at ??:?
0x4010516b in lmacTxFrame at ??:?
0x40104346 in lmacRecycleMPDU at ??:?
0x40273846 in yield at ??:?
0x401047b7 in lmacRecycleMPDU at ??:?
0x40276da4 in flash_hal_write(unsigned int, unsigned int, unsigned char const*) at ??:?
0x4010427f in lmacProcessTxSuccess at ??:?
0x40103144 in wDev_ProcessFiq at ??:?
0x40276da4 in flash_hal_write(unsigned int, unsigned int, unsigned char const*) at ??:?
0x40102e2c in wDev_ProcessFiq at ??:?
0x4027885a in spiffs_obj_lu_find_entry_visitor at ??:?
0x40278738 in spiffs_obj_lu_find_entry_visitor at ??:?
[... 82 lines cut ...]
0x40280190 in std::_Function_handler<void (WiFiEventStationModeGotIP const&), void (*)(WiFiEventStationModeGotIP const&)>::_M_invoke(std::_Any_data const&, WiFiEventStationModeGotIP const&) at ??:?    
0x40202794 in std::_Function_base::_Base_manager<ESP8266WiFiGenericClass::onStationModeDHCPTimeout(std::function<void ()>)::{lambda(_esp_event*)#1}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<ESP8266WiFiGenericClass::onStationModeDHCPTimeout(std::function<void ()>)::{lambda(_esp_event*)#1}> const&, std::_Manager_operation) at ESP8266WiFiGeneric.cpp:?
0x40201bac in std::_Function_handler<void (_esp_event*), ESP8266WiFiGenericClass::onStationModeDHCPTimeout(std::function<void ()>)::{lambda(_esp_event*)#1}>::_M_invoke(std::_Any_data const&, _esp_event*) at ESP8266WiFiGeneric.cpp:?

```

Another log after a crash after clicking Submit:
This time it sort of completes the PLUGIN_EXIT (I added that done message, just to be sure it wasn't accidentally trying to de-initialize the IR object, as that's never initialized here for lack of matching hardware)
```
121675 : Info   : WD   : Uptime 2 ConnectFailures 0 FreeMem 19488 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
130709 : Info   : P016_PLUGIN_WEBFORM_SAVE ...
130806 : Info   : P016_PLUGIN_WEBFORM_SAVE Done
130949 : Info   : P016_PLUGIN_EXIT ...
130950 : Info   : P016_PLUGIN_EXIT done
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3fff2580 end: 3fffffb0 offset: 01a0
3fff2720:  3ffefac0 3fff2f48 3fff2858 00000000
3fff2730:  00000000 02796b86 40274bb8 00000001
```

tonhuisman, 2021-05-30:
I also want to add that similar crashes already appeared, with similar frequency, _before_ I merged the String/flashhelper optimizations that where merged (very) recently.

TD-er, 2021-05-30:
Yep, there is something really strange going on in the webserver part now that we're sending lots of flash strings directly to it.

tonhuisman, 2021-08-05:
I have (finally) done some more investigation into these crashes, and there are a few things that seem to stand out:
- The IR library 'juggles' the ESP timers (calling stuff like `ets_timer_disarm` and `ets_timer_setfn`, that might interfere with Arduino timer setup and ESPEasy timer setup
- It feels like the `delay()` is redefined somewhere (though I can't find it, it _may_ be caused by the first point mentioned), causing the default wdtFeed() not to be called, causing the `Soft WDT reset` as seen in many of the log dumps shown above. (Have found redefinitions of delay in the FastLED and Blynk libraries, but these are both not included in the IR builds)
- The stack usage of the library seems to be quite high, possibly causing crashes during WEBFORM_LOAD and WEBFORM_SAVE plugin events.

I realize there's a lot of "possibly"'s and "maybe"'s here, but that's just my inexperience with a) developing for Arduino, and b) the use of this IR library.
Have tried 'appending' all `delay(0)` calls with `FeedSW_watchdog()` calls in this plugin, but that hasn't improved anything yet.

I'll search for an older version of the IR library, to see if that'll improve anything.

tonhuisman, 2021-08-05:
Well, I was obviously on the wrong track, after replacing 2 huge `String[]` and `int` arrays by a `std::vector` implementation, the crashes have now vanished. 😄

TD-er, 2021-08-05:
Ah so it was a stack overflow?

tonhuisman, 2021-08-06:
Now that I've resolved the undesirable crashes, some testing should be applied:

[ESPEasy_minimal_IRext_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/6947481/ESPEasy_minimal_IRext_ESP8266_1M.zip)
[ESPEasy_minimal_IRext_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6947482/ESPEasy_minimal_IRext_ESP8266_4M1M.zip)

Please report your findings here (success and (hopefully not) failures 😄)

tonhuisman, 2021-08-13:
@tbnobody Can you confirm this issue is now resolved, with the build made available last week? I will update to the latest branch, as there have been a few interesting improvements recently, just a short wait for the GH Actions build to complete.

tonhuisman, 2021-08-13:
The build has completed, so here are fresh builds, based on current `mega` branch:

[ESPEasy_minimal_IRext_ESP8266_1M.zip](https://github.com/letscontrolit/ESPEasy/files/6980776/ESPEasy_minimal_IRext_ESP8266_1M.zip)
[ESPEasy_minimal_IRext_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6980777/ESPEasy_minimal_IRext_ESP8266_4M1M.zip)

PR #3653 (2021-08-13): [P016] Correct hex value conversions to unsigned and support 64 bit IR codes
~~Conversion of hex input values to `uint32_t` was done using 'strtol`, but requires `strtoul`.~~

Correct conversion of IR codes using the supplied full 64 bit (`uint64_t`/`unsigned long long`) and implement full 64 bit support (upgraded from original 23 bits).

Resolves #3652 

TODO:
- [x] Update settings to support 64 bit codes and flags
- [x] Add conversion of old settings to new format
- [x] Documentation
- [x] Solve stack overflow during save (and sometimes on load) of settings
- [ ] Testing
  - [x] Local testin [... cut]
closed 2021-08-13 by TD-er