letscontrolit/ESPEasy #3657: ESP_Easy_energy_ESP8266_4M1M Can't select Rules Event name.

PrutzerNL, 2021-05-25:
Hi ALl,

I tried all kind of differend releases of the Easy_energy_ESP8266_4M1M.bin files.
From 2021 to november 2020. They all have the same problem, you can not select a Even name if you want to create a rule.
And the latest release may 2021 looks very unstable.

Is it possible to fix this?

TD-er, 2021-05-25:
May 2021 WiFi issue is a known issue, so please revert to the Feb build.

I don't know what you mean by the Rules event name.
Do you have the "Old Engine" selected?
[image]

If not, please do. 
I will remove this "new engine" check from the code as it is not working as it was intended and only makes things confusing.

tonhuisman, 2022-04-17:
The 'new' rules support has been discontinued from latest releases, so this issue should no longer exist.

TD-er, 2022-04-17:
The code for the "new rules" will also be removed soon as the optimization in rules processing makes the "new rules" obsolete anyway.

closed 2022-04-17 by TD-er