letscontrolit/ESPEasy #3662: P108_DDS238 Does not interpret -ve number

aneip, 2021-05-31:
Just tested P108 plugin. The DDS238 is connected to house main incoming line. My house have Solar PV. As I monitor the Power variable.. The value steadily coming down as solar is generating power. When the value goes below 0, the value read from the meter goes up to 65500 instead of -ve. Seems, like the value considered as unsigned int16 instead of signed 1. 

The wrong variable is at the following line.

https://github.com/letscontrolit/ESPEasy/blob/75e46ea2fad251bdcf662cedaff3624dba4e855b/src/_P108_DDS238.ino#L395-L398

I tried to change to `read_float_HoldingRegister` but got 0 value. 

Currently I just correct the value after read from modbus as below 
`if (value > 32767) value = 0 - (65535 - value);`

TD-er, 2021-05-31:
Ah, two-complement values.
Do you make a pull request, or should we change it?

uzi18, 2021-05-31:
@aneip nice catch

aneip, 2021-05-31:
@TD-er Better you guys change it. I just see the source just few hours today. Not really familiar with the code. PHP devs here. Also my solutions is hack one. Should have better way to handle the value. But I'm open to test new code once you guys make it available.

tonhuisman, 2021-08-05:
@aneip Looking at the [specs of the device](http://domoticx.com/modbus-kwh-meter-dds238-1-zn/), the same issue applies to the `P108_QUERY_W` result (is the only other register that also returns a signed value), can you confirm that?
For now I'll assume it _does_, and create a pull-request to solve this issue.

PR #3740 (2021-08-05): [P108] DDS238 Fix 2-complement values (#3662)
- Fix a reported bug that VA (and probably also W) measurements return wrong values when negative (2s complement)
- Minor code improvements, using `f` postfix for float constants

Resolved #3662
closed 2021-08-05 by TD-er