letscontrolit/ESPEasy #3664: Detect and clear I2C freeze

TD-er, 2021-06-01:
It has been reported indirectly on the forum and in some GitHub issues a few times, but I've seen it happen too now on my own boards.

What happens:
- Communication to any I2C device is no longer possible
- I2C scan results in errors on all I2C addresses
- Change pin settings for I2C and change it back (or power cycle) makes I2C available again.

This also happens every now and then on OLED display, which is the only one connected.
This plugin used to be one of the most tested and stable ones for years, so I guess something has changed in the core, or we may trigger some corner case now that we allow to switch frequencies or maybe load a lot more from flash instead of RAM.

It has been suggested on a number of forum posts on other forums that the clock pin or data pin may remain pulled down for unknown reasons.
One of the reported fixes seems to be to toggle the pins for some time and re-set their pull-up resistors, as described [here](http://www.forward.com.au/pfod/ArduinoProgramming/I2C_ClearBus/index.html)

rka-germany, 2021-06-11:
Hello, I also often have stuck when displaying the OLED SSD1306 / SH1106 Framed.
I am using the status from 8/8/2020: (ESP_Easy_mega_20200608_dev_ESP8266_4M1M.bin), there are no stuck here yet, the error must have crept in later.
Best regards, Rainer

marstum, 2021-06-22:
> Hello, I also often have stuck when displaying the OLED SSD1306 / SH1106 Framed.
> I am using the status from 8/8/2020: (ESP_Easy_mega_20200608_dev_ESP8266_4M1M.bin), there are no stuck here yet, the error must have crept in later.
> Best regards, Rainer

I am using the same OLED and had only this issue with the one which has difficulties in connecting to WIFI.
All ESPs running on ESP_Easy_mega_20210615_normal_ESP8266_4M1M.

KR marstum

TD-er, 2021-06-22:
Does that unit also crash, or is difficulties with WiFi only resulting in slow responses on the web UI and/or staying connected to WiFi?

marstum, 2021-06-22:
> Does that unit also crash, or is difficulties with WiFi only resulting in slow responses on the web UI and/or staying connected to WiFi?

As usual, it depends, and I don't know on what.
Sometimes it crashes completely, and sometimes the other ESP can see this one, but it can't be connected via web UI.

TD-er, 2021-06-22:
The reason I ask is because I have the feeling this stuck I2C may happen due to crashes in the middle of an I2C transfer.

marstum, 2021-06-22:
> The reason I ask is because I have the feeling this stuck I2C may happen due to crashes in the middle of an I2C transfer.

I am happy to help, so what do I need to do, in case the OLED is freezing next time, and I would have access to the ESP? I2C scan, or what else?

TD-er, 2021-06-22:
I2C scan is a good first attempt to see if the I2C bus is stuck.
A simple way to get it to work again is to swap the pin configuration in the settings for I2C, so swap SCL and SDA pins and save.
Then swap them back and save and I2C should be working again.

marstum, 2021-06-22:
Ok, I let you know, if this was working.

marstum, 2021-06-23:
The I2C scan doesn't show anything different to yesterday, during the OLED was working as expected, but you are absolutely correct the swap solved the freeze.

tonhuisman, 2023-09-02:
This seems to be solved, and since then, fixing I2C connection trouble has been added to the code, so this can be closed.

closed 2023-09-02 by TD-er